OE4T/meta-tegra #344: flashing Xavier NX

outbackdingo, 2020-06-09:
So i got through the nano and flashed it, with the proper zip, no can i ask how to flash the Xavier NX as i dont see a generated zip file.

madisongh, 2020-06-09:
Starting with the L4T R32.4.2-based branches, the default format for tegraflash packages has changed to `tar.gz`.

outbackdingo, 2020-06-09:
ok still not seeing it, using this conf for the Xaviaer NX dev kit... whats missing or wrong ??
MACHINE = "jetson-xavier-8gb"
LICENSE_FLAGS_WHITELIST = "commercial"
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash sdimg"
LICENSE_FLAGS_WHITELIST = "commercial_ffmpeg commercial_x264 commercial_gstreamer1.0-plugins-ugly"
LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx-tegra"
DISTRO_FEATURES_append = " l4t-usb-device-mode virtualization k8s"
#NVIDIA_DEVNET_MIRROR
INITRAMFS_MAXSIZE = " 40960000"

DISTRO_FEATURES_append = " systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
INITRAMFS_IMAGE_BUNDLE=" 1"
IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}"

ls -a ../xavier/tmp/deploy/images/jetson-xavier-8gb/
core-image-minimal-jetson-xavier-8gb-20200606125039.rootfs.cpio.gz
core-image-minimal-jetson-xavier-8gb-20200606125039.rootfs.manifest
core-image-minimal-jetson-xavier-8gb-20200606125039.testdata.json
core-image-minimal-jetson-xavier-8gb.cpio.gz
core-image-minimal-jetson-xavier-8gb.manifest
core-image-minimal-jetson-xavier-8gb.testdata.json
Image
Image--4.9.140+git0+47e7e1cb0b-r0-jetson-xavier-8gb-20200606125039.bin
Image-jetson-xavier-8gb.bin
modules--4.9.140+git0+47e7e1cb0b-r0-jetson-xavier-8gb-20200606125039.tgz
modules-jetson-xavier-8gb.tgz
tegra194-p2888-0006-p2822-0000--4.9.140+git0+47e7e1cb0b-r0-jetson-xavier-8gb-20200606125039.dtb
tegra194-p2888-0006-p2822-0000.dtb
tegra194-p2888-0006-p2822-0000-jetson-xavier-8gb.dtb
tegra-minimal-initramfs-jetson-xavier-8gb-20200606125039.rootfs.cpio.gz
tegra-minimal-initramfs-jetson-xavier-8gb-20200606125039.rootfs.manifest
tegra-minimal-initramfs-jetson-xavier-8gb-20200606125039.testdata.json
tegra-minimal-initramfs-jetson-xavier-8gb.cpio.gz
tegra-minimal-initramfs-jetson-xavier-8gb.manifest
tegra-minimal-initramfs-jetson-xavier-8gb.testdata.json

madisongh, 2020-06-09:
>IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}"

If this line is in your local.conf, it's the culprit.

outbackdingo, 2020-06-09:
so i removed IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}" and ran bitbake core-image-minimal and the result is ...

| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_image_tegraflash
| cp: cannot stat '/home/dingo/nvidia/xavier/tmp/deploy/images/jetson-xavier-8gb/Image-initramfs-jetson-xavier-8gb.cboot': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/home/dingo/nvidia/xavier/tmp/work/jetson_xavier_8gb-poky-linux/core-image-minimal/1.0-r0/temp/run.do_image_tegraflash.10629' failed with exit code 1:
| cp: cannot stat '/home/dingo/nvidia/xavier/tmp/deploy/images/jetson-xavier-8gb/Image-initramfs-jetson-xavier-8gb.cboot': No such file or directory
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/dingo/nvidia/poky/meta/recipes-core/images/core-image-minimal.bb:do_image_tegraflash) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3328 tasks of which 3327 didn't need to be rerun and 1 failed.

madisongh, 2020-06-09:
The kernel build should have generated the `.cboot` image after bundling the initramfs image.  Did your kernel build go through the bundle_initramfs step?

BTW, I just noticed that the MACHINE setting you have there isn't the one for the Xavier NX dev kit - you want `jetson-xavier-nx-devkit`.

outbackdingo, 2020-06-09:
oh he!! .... thanks for that :)
trying again!

outbackdingo, 2020-06-09:
I take it it requires the devkit ......
ERROR: No recipes available for:
  /home/dingo/nvidia/meta-tegra/recipes-graphics/vulkan/vulkan-loader_1.2.%.bbappend
  /home/dingo/nvidia/meta-tegra/recipes-graphics/vulkan/vulkan-tools_1.2.%.bbappend

madisongh, 2020-06-09:
Make sure you've got the right branches checked out.  Looks like you might be on master in meta-tegra, instead of dunfell-l4t-r32.4.2 ?

outbackdingo, 2020-06-09:
building now... thanks for the push forward
git checkout dunfell-l4t-r32.4.2
Branch 'dunfell-l4t-r32.4.2' set up to track remote branch 'dunfell-l4t-r32.4.2' from 'origin'.
Switched to a new branch 'dunfell-l4t-r32.4.2'
dingo@gitlab:~/nvidia/meta-tegra$ cd ../xavier/
dingo@gitlab:~/nvidia/xavier$ bitbake core-image-minimal

outbackdingo, 2020-06-09:
No joy, same error!

ERROR: core-image-minimal-1.0-r0 do_image_tegraflash: Execution of '/home/dingo/nvidia/xavier/tmp/work/jetson_xavier_nx_devkit-poky-linux/core-image-minimal/1.0-r0/temp/run.do_image_tegraflash.41312' failed with exit code 1:
cp: cannot stat '/home/dingo/nvidia/xavier/tmp/deploy/images/jetson-xavier-nx-devkit/Image-initramfs-jetson-xavier-nx-devkit.cboot': No such file or directory
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/dingo/nvidia/xavier/tmp/work/jetson_xavier_nx_devkit-poky-linux/core-image-minimal/1.0-r0/temp/log.do_image_tegraflash.41312
Log data follows:
| DEBUG: Executing python function set_image_size
| DEBUG: 106631.200000 = 82024 * 1.300000
| DEBUG: 14680064.000000 = max(106631.200000, 14675968)[14675968.000000] + 4096
| DEBUG: 14680064.000000 = int(14680064.000000)
| DEBUG: 14680064 = aligned(14680064)
| DEBUG: returning 14680064
| DEBUG: Python function set_image_size finished
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['virtual:native:/home/dingo/nvidia/poky/meta/recipes-extended/pigz/pigz_2.4.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-core/update-rc.d/update-rc.d_0.8.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-core/glibc/ldconfig-native_2.12.1.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-core/glibc/cross-localedef-native_2.31.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-kernel/kmod/kmod-native_git.bb:do_populate_sysroot', '/home/dingo/nvidia/meta-tegra/recipes-bsp/tegra-binaries/tegra186-flashtools-native_32.4.2.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/opkg-utils/opkg-utils_0.4.2.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-devtools/gcc/libgcc_9.3.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/e2fsprogs/e2fsprogs_1.45.4.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-devtools/gcc/gcc-cross_9.3.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-support/lzop/lzop_1.04.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-devtools/qemu/qemuwrapper-cross_1.0.bb:do_populate_sysroot', '/home/dingo/nvidia/meta-tegra/recipes-bsp/tegra-binaries/tegra-bootfiles_32.4.2.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/perl/perl_5.30.1.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-kernel/kmod/depmodwrapper-cross_1.0.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-devtools/binutils/binutils-cross_2.34.bb:do_populate_sysroot', '/home/dingo/nvidia/meta-tegra/recipes-kernel/linux/linux-tegra_4.9.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-extended/xz/xz_5.2.4.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-kernel/dtc/dtc_1.5.1.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/python/python3_3.8.2.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/makedevs/makedevs_1.0.1.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/bison/bison_3.5.3.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-extended/bc/bc_1.07.1.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/createrepo-c/createrepo-c_0.15.7.bb:do_populate_sysroot', '/home/dingo/nvidia/meta-tegra/recipes-bsp/tegra-binaries/tegra-redundant-boot-base_32.4.2.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/pseudo/pseudo_git.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/rpm/rpm_4.14.2.1.bb:do_populate_sysroot', '/home/dingo/nvidia/poky/meta/recipes-devtools/mklibs/mklibs-native_0.1.44.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/opkg/opkg_0.4.2.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/prelink/prelink_git.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/fdisk/gptfdisk_1.0.4.bb:do_populate_sysroot', 'virtual:native:/home/dingo/nvidia/poky/meta/recipes-devtools/dnf/dnf_4.2.2.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['pigz-native', 'kern-tools-native', 'update-rc.d-native', 'ldconfig-native', 'cross-localedef-native', 'kmod-native', 'tegra186-flashtools-native', 'opkg-utils-native', 'libgcc', 'e2fsprogs-native', 'patch-native', 'gcc-cross-aarch64', 'lzop-native', 'qemuwrapper-cross', 'tegra-bootfiles', 'perl-native', 'depmodwrapper-cross', 'binutils-cross-aarch64', 'linux-tegra', 'xz-native', 'dtc-native', 'python3-native', 'makedevs-native', 'bison-native', 'bc-native', 'createrepo-c-native', 'tegra-redundant-boot-base', 'pseudo-native', 'quilt-native', 'rpm-native', 'mklibs-native', 'opkg-native', 'prelink-native', 'gptfdisk-native', 'dnf-native', 'zlib-native', 'systemd-systemctl-native', 'qemu-native', 'shadow-native', 'automake-native',  [... cut ...]
| cp: cannot stat '/home/dingo/nvidia/xavier/tmp/deploy/images/jetson-xavier-nx-devkit/Image-initramfs-jetson-xavier-nx-devkit.cboot': No such file or directory
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/home/dingo/nvidia/poky/meta/recipes-core/images/core-image-minimal.bb:do_image_tegraflash) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3317 tasks of which 3316 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/dingo/nvidia/poky/meta/recipes-core/images/core-image-minimal.bb:do_image_tegraflash
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

madisongh, 2020-06-09:
That file should have been created during the do_deploy task of the kernel build.  Take a look under `$BUILDDIR/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+21848bb857-r0/` - check the log files in the `temp` subdirectory for any errors that might have somehow got missed, and look in the `deploy-linux-tegra` subdirectory to see if the `.cboot` file is there.

outbackdingo, 2020-06-10:
still errors.... no .cboot in deploy-linux-tegra... nothing odd in the log file

ls tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+7519746f16-r0/deploy-linux-tegra/
Image
Image--4.9.140+git0+7519746f16-r0-jetson-xavier-nx-devkit-20200609224110.bin
Image-jetson-xavier-nx-devkit.bin
modules--4.9.140+git0+7519746f16-r0-jetson-xavier-nx-devkit-20200609224110.tgz
modules-jetson-xavier-nx-devkit.tgz
tegra194-p3668-all-p3509-0000--4.9.140+git0+7519746f16-r0-jetson-xavier-nx-devkit-20200609224110.dtb
tegra194-p3668-all-p3509-0000.dtb
tegra194-p3668-all-p3509-0000-jetson-xavier-nx-devkit.dtb

outbackdingo, 2020-06-10:
fixed... first i removed #IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}"
as suggested... no joy ... removing both however resolved the issue, its working now
#INITRAMFS_IMAGE_BUNDLE=" 1"
#IMAGE_FSTYPES = "${INITRAMFS_FSTYPES}"

closed 2020-06-10 by outbackdingo