OE4T/meta-tegra #347: nice to have: be able to build meta-tegra on Arm64

ericvh, 2020-06-11:
So I just fired up a graviton2 instance to try and build meta-tegra.  There was an issue with golang in poky that I worked around, but not sure how to work around the nvidia x86 binaries we use in tegra-bootfiles

Upside to fixing this is graviton2's are really fast building meta-tegra natively up to the point where things break.

madisongh, 2020-06-11:
I ran a test build the other day (while investigating #339) and got it working on an m6g instance running Ubuntu 20.04. I installed `binfmt-support` to get the x86-64 binaries running automatically.

I too ran into some go issues.  Setting `ASSUME_PROVIDED = "go-native"` got me most of the way, but I had to hack a couple of scripts in the docker-ce build to use `${GO}` instead of just `go`, since the Ubuntu go version is 1.13 while the one in OE-Core is 1.14.  If you don't need to build the nvidia-docker stuff, that shouldn't be an issue.

I also had a minor issue with the CUDA toolchain recipes, but I have already pushed fixes for that to master.  I've just now cherry-picked them to dunfell-l4t-r32.4.2 as well.

ericvh, 2020-06-12:
can confirm this works for me as well - was able to finish building the image.

closed 2020-06-12 by ericvh