OE4T/meta-tegra #350: Virtualization kernel extensions / KVM on Jetson / Xavier

outbackdingo, 2020-06-14:
would these patches be a decent addition to the build tree as bb appends for virtualization / KVM

https://github.com/b-man/Xavier-KVM

kekiefer, 2020-06-14:
Did you find the first two patches to be necessary? I see the third is just to fix a warning. It's been a couple releases since I tried it, but I was able to run KVM with Debian and linux-yocto guests with only the GIC device tree changes and a custom kernel config, so I'd be curious where the first two changes are applicable.

In the latest l4t, note that the default device tree includes the GIC device tree changes.

outbackdingo, 2020-06-14:
according to the documentation on the nvdia forum the rest were added for winders 10 arm64
https://forums.developer.nvidia.com/t/guide-to-enable-kvm-on-the-xavier/119777
and the recently merged windows patch. Be nice to see it added as a bitbake recipe so we can Yoctoize the Xaviers, since i have also meta-virtualization with k8s and docker-nvidia, going for kubevirt / k8s on Xavier..... requires a working libvirt / kvm .... I also previously inquired about XEN which is whats preferred, or making xvisor work for the Jetsons

madisongh, 2020-06-14:
The kernel patches are backports from upstream, so I'd be OK with submitting them as a PR in the linux-tegra-4.9 repo. 

If you can isolate the specific config settings that need to be enabled, the config changes could be applied in a `.cfg` file as part of a linux-tegra bbappend under `external/meta-virtualization`, conditional on DISTRO_FEATURES containing `kvm`.  That would be the least intrusive.