OE4T/meta-tegra #354: dunfell-l4t-r32.4.2 branch do_populate_sdk error

mozcelikors, 2020-06-15:
I am trying to generate SDK using a recipe that does do_populate_sdk, however I get following issue:

`ERROR: gcc-cross-canadian-aarch64-8.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.1/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/8.3.0/cc1 contained in package gcc-cross-canadian-aarch64 requires libdl.so.2()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]`

Is this error happen to be meta-tegra related?, has anyone verified generating SDK for this branch?
Complete list of layers included are as follows:

```
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:6436d102f15a129384d980f65b02b86fc52ae77a"
meta-tegra           = "dunfell-l4t-r32.4.2:d5c0bedf56ad58e9a9b046aefe2c023134726bdf"
contrib              = "dunfell-l4t-r32.4.2:d5c0bedf56ad58e9a9b046aefe2c023134726bdf"
meta-oe              
meta-networking      
meta-multimedia      
meta-gnome           
meta-python          = "dunfell:cc155826b0b4f202a8fa4ac103b7c38368d32734"
meta-qt5             = "dunfell:fdd19517e17240b0b61765bd02fc483a1bde986f"
meta-qt5-extra       = "master:f154761f10096aaed27277f0d4d211564022b58c"
```

MJLHThomassen-Sorama, 2020-06-16:
@mozcelikors I have succesfully build an SDK image for the dunfell-l4t-r32.4.2 branch (regular sdk, not ext_sdk). 

Did you set your GCCVERSION = "8.%" in your local.conf correctly?

madisongh, 2020-06-16:
I'm not sure what changed to start triggering this, but I tracked down what I think is the right fix for this issue and have pulled it into the relevant branches.  Give it another try after updating meta-tegra to the tip of dunfell-l4t-r32.4.2 and see if it works for you.

closed 2020-06-16 by madisongh

mozcelikors, 2020-06-16:
@MJLHThomassen-Sorama  Of course.
@madisongh  I will try that, thanks.

PR #356 (2020-06-16): contrib: gcc-7, gcc-8: drop obsolete shlibs exclusion
from gcc-cross-canadian.inc.  This change was made in
commit 096fa15efbcb704451b2f38ceab36508ef64f07e in OE-Core,
and fixes missing dependencies issues during package QA
for the cross-canadian compiler package.

Fixes #354