OE4T/meta-tegra #355: could not initialize GPT on sorama-image-tegra-devkit.sdcard

MJLHThomassen-Sorama, 2020-06-16:
I am getting the following error while running ./dosdcard.sh:

could not initialize GPT on sorama-image-tegra-devkit.sdcard

For some strange reason, this only occurs when running the command unattended via our Teamcity buildagent. If i manually run the command on my own pc, everything succeeds.
Has anyone seen this or knows why this is happening?

madisongh, 2020-06-16:
Does your build server have the `sgdisk` command installed on it?

MJLHThomassen-Sorama, 2020-06-16:
> Does your build server have the `sgdisk` command installed on it?

I will check.

madisongh, 2020-06-22:
Try using `bash -x ./dosdcard.sh` to run it, to get some more information about why it failed.  It's pretty simple up to the point you're seeing the error message for, though - it's just creating the output file and then applying the partition table.  Maybe your build server is running short on disk space or something?

madisongh, 2020-11-02:
Nothing new on this for while, so I'll close this under the assumption that the problem got resolved.

closed 2020-11-02 by madisongh