OE4T/meta-tegra #357: Advice on using Chromium?

dreambeyondorange, 2020-06-16:
Has anyone had success with building chromium on a jetson image?  Right now my build is complaining about DRI but from what I understand NVIDIA doesn't support that.  Anyone know how to get Chromium working with meta-tegra?

madisongh, 2020-06-18:
Looks like the recipe is assuming that depending on `virtual/egl` and/or `virtual/libgl` is equivalent to depending on `mesa`, which for our platforms isn't the case.  `mesa` provides the DRI support that the chromium build is looking for, so adding it to DEPENDS for that recipe takes care of the issue.  From a brief glance of where it gets used in the code base, it looks like it's only really used at runtime for AMD GPU support.

I ran a test build and was able to get a working chromium-x11 running in my sato image.  I added `mesa` to DEPENDS and `use_system_libdrm=true` to GN_ARGS.  I'm not sure that additional setting is really needed, but it appeared to work OK, and looking at `chrome://gpu` in the browser running on my Nano showed hardware acceleration enabled.

madisongh, 2020-06-18:
BTW, this was on a build off `zeus-l4t-r32.3.1`.

dreambeyondorange, 2020-06-18:
Thanks for the feedback!  I did a similar thing on the NX and saw HW acc for several items in chrome://gpu as well.  

What does `use_system_libdrm=true` do in this case?  I thought adding mesa to the depends was a hack that might bite me later.  Do you think it's safe to build it this way?

Thanks!
Ryan

madisongh, 2020-06-19:
> What does use_system_libdrm=true do in this case?

It looked like Chromium includes its own copy of libdrm, which it normally builds against - there was a comment about how Ubuntu's libdrm is too old or missing some functionality in the GN file I looked at - but I didn't track down where it's actually used in the code, so I don't know whether it would work or not.  Sounds like it doesn't matter either way, at least for the X11 case.

> I thought adding mesa to the depends was a hack that might bite me later.

I'm not 100% sure, but I think it's the right solution.  The `dri.pc` file that the Chromium build files are looking for is mesa-specific, and doesn't have anything to do with OpenGL.

closed 2020-06-22 by dreambeyondorange