OE4T/meta-tegra #358: Add yocto kernel-cache

paroque28, 2020-06-17:
Continuing this commit:
https://github.com/madisongh/meta-tegra/commit/29ee143b26af5e1057fb4533a9376ca43512f603#diff-9b732ab86ef3483bb0e54684c28fd84b

We can leverage features from yocto, as an example I am using `systemd.scc` and `docker.scc` and I encounter the following error:
```
 DEBUG: Executing shell function do_kernel_metadata
| ERROR. input file "cfg/systemd.scc" does not exist
| ERROR: could not process input files: ...linux/linux-tegra-4.9/tegra210/defconfig /wr/sandbox/prodrigu/projects/gpu-operator/layers/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/localversion_auto.cfg cfg/systemd.scc features/lxc/lxc-enable.scc features/netfilter/netfilter.scc
|        See /tmp/tmp.5e6VqiQdz2 for details
| ERROR: Could not generate configuration queue for jetson-nano-qspi-sd.
| WARNING: ..../linux-tegra/4.9.140+gitAUTOINC+47e7e1cb0b-r0/temp/run.do_kernel_metadata.112329:1 exit 1 from 'exit 1'
| ERROR: Execution of '..../linux-tegra/4.9.140+gitAUTOINC+47e7e1cb0b-r0/temp/run.do_kernel_metadata.112329' failed with exit code 1:
| ERROR. input file "cfg/systemd.scc" does not exist
| ERROR: could not process input files: /..../linux/linux-tegra-4.9/tegra210/defconfig /wr/sandbox/prodrigu/projects/gpu-operator/layers/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/localversion_auto.cfg cfg/systemd.scc features/lxc/lxc-enable.scc features/netfilter/netfilter.scc
|        See /tmp/tmp.5e6VqiQdz2 for details
```

This can be fixed with the following patch:

```
From d6d11b9921083c41ac8b701db645e5b992f865f9 Mon Sep 17 00:00:00 2001
From: Pablo Rodriguez <pablo.rodriguez-quesada@windriver.com>
Date: Tue, 16 Jun 2020 23:15:28 -0600
Subject: [PATCH] linux-tegra: Add yocto-kernel-cache sources

Making use of the current kernel-cache is possible for linux-tegra.
Features and configurations are compatible.

---
 recipes-kernel/linux/linux-tegra_4.9.bb | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/recipes-kernel/linux/linux-tegra_4.9.bb b/recipes-kernel/linux/linux-tegra_4.9.bb
index ffa1811..78eb89f 100644
--- a/recipes-kernel/linux/linux-tegra_4.9.bb
+++ b/recipes-kernel/linux/linux-tegra_4.9.bb
@@ -16,10 +16,13 @@ SCMVERSION ??= "y"
 
 SRCBRANCH = "patches${LINUX_VERSION_EXTENSION}"
 SRCREV = "47e7e1cb0b492487faa6258a4f3efe91676568b7"
+SRCREV_meta = "c0c614c597b20fe0070b44eebd6e3265d006827d"                                                                                                                                           
 KBRANCH = "${SRCBRANCH}"
+KMETA = "kernel-meta"
 SRC_REPO = "github.com/madisongh/linux-tegra-4.9"
 KERNEL_REPO = "${SRC_REPO}"
 SRC_URI = "git://${KERNEL_REPO};name=machine;branch=${KBRANCH} \
+       git://git.yoctoproject.org/yocto-kernel-cache;type=kmeta;name=meta;branch=yocto-4.9;destsuffix=${KMETA} \
           file://defconfig \
           ${@'file://localversion_auto.cfg' if d.getVar('SCMVERSION') == 'y' else ''} \
 "
-- 
2.7.4

```
Reference: https://git.yoctoproject.org/cgit.cgi/poky/plain/meta/recipes-kernel/linux/linux-yocto_5.4.bb

Build tested in zeus-r32.3.1 for jetson-nanor

Thanks,
Pablo

madisongh, 2020-06-18:
Yeah, I held off going this far because I don't think this will generalize to any feature that's defined in the linux-yocto metadata... in particular, features that include patches to be applied.  Our kernel diverges from upstream significantly in some areas. And I'm not sure how to provide test coverage.

Maybe at some point in the future we could add this to the base recipe, either using the linux-yocto metadata directly, or a fork of our own. In the meantime, you're free to add a bbappend in your distro layer to make use of the metadata if you'd like to.

ichergui, 2021-09-07:
Hello guys  @paroque28 @madisongh 

Any update concerning this feature ? 
Should we close this ticket ?

paroque28, 2021-09-07:
I think we should close it for now. When Nvidia releases a newer version of the kernel, this will be much easier.

ichergui, 2021-09-07:
Thanks @paroque28 for your quick feedback.
CLosing this ticket

closed 2021-09-07 by ichergui