OE4T/meta-tegra #359: nvidia camera source blacklisted

jlucke86, 2020-06-17:
I'm doing a board bring up and the camera input source nvarguscamerasrc isn't available.

I'm using a XavierNX Production Version (with EMMC) installed into a Jetson Nano Developer Kit carrier. I've supplied a minimum example of a local.conf file as well as the error on the terminal. I'm using the current version of 'dunfell-l4t-r32.4.2'. Everything else seems to be working OK so far. I copied the receipes for gstreamer 1.14.4 from the test-distro

`root@jetson-xavier-nx-devkit-emmc:/# uname -a
Linux jetson-xavier-nx-devkit-emmc 4.9.140-l4t-r32.4.2+ge3c8d3e9030a #1 SMP PREEMPT Thu Jun 11 13:28:30 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux`

`root@jetson-xavier-nx-devkit-emmc:~# gst-inspect-1.0 -b
Blacklisted files:
  libgstnvarguscamerasrc.so

Total count: 1 blacklisted file`

`NVIDIA_DEVNET_MIRROR = "file:///nvidia_sdk/nvidia/sdkm_downloads/"
CUDA_BINARIES_NATIVE = "cuda-binaries-ubuntu1804-native"
MACHINE = "jetson-xavier-nx-devkit-emmc"
IMAGE_CLASSES += "image_types_tegra"               
IMAGE_FSTYPES = "tegraflash"
GCCVERSION_aarch64 = "8.%"
SDKGCCVERSION = "9.%"
INIT_MANAGER = "systemd"
LICENSE_FLAGS_WHITELIST="commercial"
LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx-tegra"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_deb"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks tools-debug"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"

BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"

BB_NUMBER_THREADS = "70"
PARALLEL_MAKE = "-j 70"

IMAGE_INSTALL_append = " example"
CORE_IMAGE_EXTRA_INSTALL += "openssh"
CORE_IMAGE_EXTRA_INSTALL += "htop"
CORE_IMAGE_EXTRA_INSTALL += "vim"

PREFERRED_VERSION_gstreamer1.0 = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-plugins-base = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-plugins-good = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-plugins-bad = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-plugins-ugly = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-libav = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-vaapi = "1.14.4"
PREFERRED_VERSION_gstreamer1.0-omx-tegra = "1.0.0-r32.4.2"
PREFERRED_VERSION_gstreamer1.0-plugins-nvvideo4linux2 = "1.14.0-r32.4.2"
#PREFERRED_VERSION_gstreamer1.0-omx = "1.14.4"

CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-plugins-base"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-plugins-good"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-plugins-bad"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-plugins-ugly"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-libav"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-vaapi"
#CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-omx"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-omx-tegra"
CORE_IMAGE_EXTRA_INSTALL += "gstreamer1.0-plugins-nvvideo4linux2"
CORE_IMAGE_EXTRA_INSTALL += "v4l-utils"

CONF_VERSION = "1"
TOOLCHAIN_HOST_TASK_append = " nativesdk-cmake"
TOOLCHAIN_TARGET_TASK_append = " gstreamer1.0-dev gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-libav gstreamer1.0-vaapi gstreamer1.0-omx-tegra gstreamer1.0-plugins-nvvideo4linux2"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-ugly = "x264 x265"
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-good = "libv4l2"
`

madisongh, 2020-06-17:
You'll get that if the Argus libraries or daemon is missing.  Make sure you're including `tegra-libraries-argus` in your image.  That should pull in the Argus support.

If you do have them in your image and the plugin is still blacklisted, make sure the Argus daemon is up and running.

If all that checks out, remove the gstreamer cache and re-run `gst-inspect-1.0` with the debug log level cranked up to get additional info about why the plugin is failing to initialize.

jlucke86, 2020-06-17:
Adding `CORE_IMAGE_EXTRA_INSTALL += "tegra-libraries-argus"` fixed the issue with the blacklisted gstreamer tegra v4l2 component.

Have you tested the camera inputs? If you have could you please share details on which nx and also which carrier you used.

"dmesg | fgrep imx219" shows a i2c bus probe error.

jlucke86, 2020-06-17:
I have 2 imx219 pi cameras neither work. Both were working on last test in a PI4.

madisongh, 2020-06-17:
I have not used any MIPI cameras on the Xavier NX, only USB.

In general, though, it usually takes some device tree/driver changes, and/or settings configuration files for the Argus daemon, to make MIPI cameras work on any of the Jetsons.  There's a [chapter in the L4T docs](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fcamera_sensor_prog.html%23) on the subject, and there are [specific NVIDIA partners](https://developer.nvidia.com/embedded/jetson-partner-supported-cameras) that sell cameras and provide (more or less) the necessary software.

jlucke86, 2020-06-19:
Thanks for your help.

Are you interested in collaboratively extending the support for say the imx219 (pi camera) or some other readily available csi camera to increase the usability for the general users. I would be happy to donate some sensors to the project if its something you are interested in?

madisongh, 2020-06-20:
Thanks for the offer.  It would help to be able to test with a MIPI-CSI camera setup, on both the Nano and NX.  I'm just not sure which would be the right choice.

jlucke86, 2020-08-10:
@madisongh this turned out to be a furphy. There is no issue with the imx219 camera sensor working. I had the jetson nx in a different carrier board and the camera section of the device tree didn't match up (probably i2c)

However I have found the image quality was better on the jetson nano with yocto zeus than the jetson nx with yocto dunfel.

I'm still happy to supply a couple of cameras if you can give me some contact details

TIS-Arne, 2020-08-19:
@jlucke86 

Just saw this issue. The degraded image quality comes from a missing configuration file for the camera. On the NVIDIA image this file is called "camera_overrides.isp" and resides in the "/var/nvidia/nvcam" folder. 
Make sure to retain the permissions of this file when copying it over (ie. owner root.root and 0660 permissions), otherwise the argusdaemon will not load the file.

jlucke86, 2020-08-28:
Hi @TIS-Arne,

I've ssh'd into my NX, and downloaded and mounting the nvidia production image, there is no camera_overrides.isp file in either /var/nvidia/nvcam folder? Could you please steer me to the documentation for the .isp files?

jlucke86, 2020-08-28:
@TIS-Arne 

I found a camera_overrides.isp posted online and put it in the filesystem. Didn't seem the fix the issue.

`root@jetson-xavier-nx-devkit-emmc:~# GST_DEBUG=3 gst-launch-1.0 -e -v nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=3264, height=1848, format=NV12, framerate=28/1' ! nvv4l2h264enc bitrate=6000000 ! h264parse ! rtph264pay config-interval=1 pt=96 ! udpsink host=10.1.1.40 port=5600
Setting pipeline to PAUSED ...
Failed to query video capabilities: Inappropriate ioctl for device
Opening in BLOCKING MODE 
0:00:00.108357056 137645   0x55a1c2b130 WARN                    v4l2 gstv4l2object.c:2372:gst_v4l2_object_add_interlace_mode:0x55a1bf64b0 Failed to determine interlace mode
0:00:00.108450784 137645   0x55a1c2b130 WARN                    v4l2 gstv4l2object.c:2372:gst_v4l2_object_add_interlace_mode:0x55a1bf64b0 Failed to determine interlace mode
0:00:00.108508384 137645   0x55a1c2b130 WARN                    v4l2 gstv4l2object.c:2372:gst_v4l2_object_add_interlace_mode:0x55a1bf64b0 Failed to determine interlace mode
0:00:00.108913536 137645   0x55a1c2b130 WARN                    v4l2 gstv4l2object.c:4435:gst_v4l2_object_probe_caps:<nvv4l2h264enc0:src> Failed to probe pixel aspect ratio with VIDIOC_CROPCAP: Unknown error -1
Pipeline is live and does not need PREROLL ...
0:00:00.110277600 137645   0x55a1bef370 FIXME                default gstutils.c:3981:gst_pad_create_stream_id_internal:<nvarguscamerasrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstNvArgusCameraSrc:nvarguscamerasrc0.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)3264, height=(int)1848, format=(string)NV12, framerate=(fraction)28/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw(memory:NVMM), width=(int)3264, height=(int)1848, format=(string)NV12, framerate=(fraction)28/1
/GstPipeline:pipeline0/nvv4l2h264enc:nvv4l2h264enc0.GstPad:src: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)NULL, level=(string)NULL, width=(int)3264, height=(int)1848, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)28/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-site=(string)mpeg2
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:sink: caps = video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, profile=(string)NULL, level=(string)NULL, width=(int)3264, height=(int)1848, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)28/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-site=(string)mpeg2
Redistribute latency...
NvMMLiteOpen : Block : BlockType = 4 
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4 
0:00:00.120620288 137645   0x55a1bef370 WARN          v4l2bufferpool gstv4l2bufferpool.c:1057:gst_v4l2_buffer_pool_start:<nvv4l2h264enc0:pool:src> Uncertain or not enough buffers, enabling copy threshold
/GstPipeline:pipeline0/nvv4l2h264enc:nvv4l2h264enc0.GstPad:sink: caps = video/x-raw(memory:NVMM), width=(int)3264, height=(int)1848, format=(string)NV12, framerate=(fraction)28/1
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:sink: caps = video/x-raw(memory:NVMM), width=(int)3264, height=(int)1848, format=(string)NV12, framerate=(fraction)28/1
H264: Profile = 66, Level = 0 
0:00:00.755074528 137645   0x55a1bef1e0 WARN          v4l2bufferpool gstv4l2bufferpool.c:1503:gst_v4l2_buffer_pool_dqbuf:<nvv4l2h264enc0:pool:src> Driver should never set v4l2_buffer.field to ANY
/GstPipeline:pipeline0/GstH264Parse:h264parse0.GstPad:src: caps = video/x-h264, stream-format=(string)avc, alignment=(string)au, profile=(string)constrained-baseline, level=(string)5.1, width=(int)3264, height=(int)1848, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)28/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-site=(string)mpeg2, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, codec_data=(buffer)01424033ffe1000c6742403396540198074fca8001000468ce3c80
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:src: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)424033, sprop-parameter-sets=(string)"Z0JAM5ZUAZgHT8qA\,aM48gA\=\=", payload=(int)96, ssrc=(uint)2239888606, timestamp-offset=(uint)2442996354, seqnum-offset=(uint)15460, a-framerate=(string)28
/GstPipeline:pipeline0/GstUDPSink:udpsink0.GstPad:sink: caps = application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, packetization-mode=(string)1, profile-level-id=(string)424033, sprop-parameter-sets=(string)"Z0JAM5ZUAZgHT8qA\,aM48gA\=\=", payload=(int)96, ssrc=(uint)2239888606, timestamp-offset=(uint)2442996354, seqnum-offset=(uint)15460, a-framerate=(string)28
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0.GstPad:sink: caps = video/x-h264, stream-format=(string)avc, alignment=(string)au, profile=(string)constrained-baseline, level=(string)5.1, width=(int)3264, height=(int)1848, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)28/1, interlace-mode=(string)progressive, colorimetry=(string)bt709, chroma-site=(string)mpeg2, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, codec_data=(buffer)01424033ffe1000c6742403396540198074fca8001000468ce3c80
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: timestamp = 2443051875
/GstPipeline:pipeline0/GstRtpH264Pay:rtph264pay0: seqnum = 15460`

I've attached the log file, smells like something incorrect in the device tree to me.

`root@jetson-xavier-nx-devkit-emmc:/sys/kernel/debug/pinctrl/2430000.pinmux# cat pinmux-functions  | fgrep i2c3
function: i2c3, groups = [ cam_i2c_sda_pp3 cam_i2c_scl_pp2 ]
root@jetson-xavier-nx-devkit-emmc:/sys/kernel/debug/pinctrl/2430000.pinmux#`

jlucke86, 2020-08-28:
In the above gst-launch the camera works but the quality is highest at 1280x720, if i set the mode to something high resolution the image looks worse

this is still using a imx219 pi camera

jlucke86, 2020-08-29:
Looks to me like its an upstream issue

https://forums.developer.nvidia.com/t/imx219-board-setup-error-jetson-xavier-nx-jetpack4-4/140311

jlucke86, 2021-06-10:
Closing this issue. If someone else faces similar problems, the Sony imx477 sensor works well (Raspberry Pi HQ camera). Ridgerun have open source kernel patches which can be made to work on L4T 32.5. Image quality and latency is very good. I never got the imx219 working with adequate quality on L4T.

closed 2021-06-10 by jlucke86