OE4T/meta-tegra #360: QA Issue: libvisionworks-tracking- cudnn- do_package_qa

paroque28, 2020-06-17:
Branch: zeus-l4t-r32.3.1
```
ERROR: libvisionworks-tracking-0.88.2-r0 do_package_qa: QA Issue: libvisionworks-tracking rdepends on cuda-cudart-dev [dev-deps]
ERROR: libvisionworks-1.6.0.500n-r0 do_package_qa: QA Issue: libvisionworks rdepends on cuda-cudart-dev [dev-deps]
ERROR: libvisionworks-1.6.0.500n-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: <project>/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/libvisionworks/1.6.0.500n-r0/temp/log.do_package_qa.3759
ERROR: libvisionworks-tracking-0.88.2-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Task (<project>/layers/meta-tegra/recipes-devtools/visionworks/libvisionworks_1.6.0.500n.bb:do_package_qa) failed with exit code '1'
ERROR: Logfile of failure stored in: <project>/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/libvisionworks-tracking/0.88.2-r0/temp/log.do_package_qa.3754
ERROR: libvisionworks-sfm-0.90.4-r0 do_package_qa: QA Issue: libvisionworks-sfm rdepends on cuda-cudart-dev [dev-deps]
ERROR: Task (<project>/layers/meta-tegra/recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb:do_package_qa) failed with exit code '1'
ERROR: libvisionworks-sfm-0.90.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: <project>/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/libvisionworks-sfm/0.90.4-r0/temp/log.do_package_qa.3756
ERROR: Task (<project>/layers/meta-tegra/recipes-devtools/visionworks/libvisionworks-sfm_0.90.4.bb:do_package_qa) failed with exit code '1'
ERROR: cudnn-7.6.3.28-1-r0 do_package_qa: QA Issue: /usr/src/cudnn_samples_v7/multiHeadAttention/multiHeadAttention contained in package cudnn-samples requires libcudart.so.10.0(libcudart.so.10.0)(64bit), but no providers found in RDEPENDS_cudnn-samples? [file-rdeps]
ERROR: cudnn-7.6.3.28-1-r0 do_package_qa: QA Issue: /usr/src/cudnn_samples_v7/multiHeadAttention/multiHeadAttention contained in package cudnn-samples requires libcudart.so.10.0()(64bit), but no providers found in RDEPENDS_cudnn-samples? [file-rdeps]
ERROR: cudnn-7.6.3.28-1-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: <project>/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/cudnn/7.6.3.28-1-r0/temp/log.do_package_qa.3763
```

For the cudnn:
```
find ./tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/packages-split/ -name libcudart.so.10.0
./tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/packages-split/cuda-cudart-dev/usr/local/cuda-10.0/lib64/libcudart.so.10.0
```

Possible patches:
```
From 7f5299c69521e0d726fca7e8d9b1961c0736fea8 Mon Sep 17 00:00:00 2001
From: Pablo Rodriguez <pablo.rodriguez-quesada@windriver.com>
Date: Wed, 17 Jun 2020 10:01:47 -0600
Subject: [PATCH 1/2] libvisionworks-tracking: Add RDEPENDS cudart

---
 recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb b/recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb
index d23b655..3501c73 100644
--- a/recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb
+++ b/recipes-devtools/visionworks/libvisionworks-tracking_0.88.2.bb
@@ -38,5 +38,5 @@ PACKAGES += "${PN}-samples"
 FILES_${PN}-dev += "${libdir}/pkgconfig ${datadir}/visionworks-tracking/cmake"
 FILES_${PN}-doc += "${datadir}/visionworks-tracking/docs"
 FILES_${PN}-samples += "${datadir}/visionworks-tracking/sources"
-RDEPENDS_${PN} = "libstdc++"
+RDEPENDS_${PN} = "libstdc++ cuda-cudart-dev"
 PACKAGE_ARCH = "${SOC_FAMILY_PKGARCH}"
-- 
2.7.4
```

```
From c249a20e99e93161ccedf6842930bffbf3855499 Mon Sep 17 00:00:00 2001
From: Pablo Rodriguez <pablo.rodriguez-quesada@windriver.com>
Date: Wed, 17 Jun 2020 10:15:02 -0600
Subject: [PATCH 2/2] cudnn: Add RDEPENDS for samples cudart-dev

CUDNN samples depends on libcudart.so.10.0 file.

---
 recipes-devtools/cudnn/cudnn_7.6.3.28-1.bb | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)

diff --git a/recipes-devtools/cudnn/cudnn_7.6.3.28-1.bb b/recipes-devtools/cudnn/cudnn_7.6.3.28-1.bb
index 7d2abce..8fcf573 100644
--- a/recipes-devtools/cudnn/cudnn_7.6.3.28-1.bb
+++ b/recipes-devtools/cudnn/cudnn_7.6.3.28-1.bb
@@ -55,6 +55,6 @@ INHIBIT_PACKAGE_STRIP = "1"
 INHIBIT_PACKAGE_DEBUG_SPLIT = "1"
 INHIBIT_SYSROOT_STRIP = "1"
 
-RDEPENDS_${PN}-samples = "cuda-cublas cuda-cublas-dev cuda-cudart"
+RDEPENDS_${PN}-samples = "cuda-cublas cuda-cublas-dev cuda-cudart cuda-cudart-dev"
 RPROVIDES_${PN}-samples = "${PN}-examples"
 INSANE_SKIP_${PN}-samples = "build-deps dev-deps ldflags staticdev"
-- 
2.7.4
```

paroque28, 2020-06-17:
Add all libvisionworks packages to the same problem
```
ERROR: libvisionworks-1.6.0.500n-r0 do_package_qa: QA Issue: libvisionworks rdepends on cuda-cudart-dev [dev-deps]
ERROR: libvisionworks-1.6.0.500n-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: <project>/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/libvisionworks/1.6.0.500n-r0/temp/log.do_package_qa.88754
ERROR: Task (<project>/layers/meta-tegra/recipes-devtools/visionworks/libvisionworks_1.6.0.500n.bb:do_package_qa) failed with exit code '1'
ERROR: libvisionworks-sfm-0.90.4-r0 do_package_qa: QA Issue: libvisionworks-sfm rdepends on cuda-cudart-dev [dev-deps]
ERROR: libvisionworks-sfm-0.90.4-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: <project>/build/tmp-glibc/work/armv8a_tegra210-wrs-linux/libvisionworks-sfm/0.90.4-r0/temp/log.do_package_qa.88752
ERROR: Task (<project>/layers/meta-tegra/recipes-devtools/visionworks/libvisionworks-sfm_0.90.4.bb:do_package_qa) failed with exit code '1'
```

paroque28, 2020-06-17:
For the libvisionworks I am not sure if an INSANE_SKIP should be the fix:
https://www.yoctoproject.org/pipermail/yocto/2012-June/007198.html

madisongh, 2020-06-17:
I think you need to track down why `libcudart.so.10.0` is ending up in the `cuda-cudart-dev` package in your builds, rather than `cuda-cudart`, where it belongs.  I don't seen any of these packaging QA issues in my builds, and that library symlink is correctly packaged  into `cuda-cudart`.

paroque28, 2020-06-17:
Thanks @madisongh ,
I will look into it

closed 2020-06-17 by paroque28

paroque28, 2020-06-17:
I figured out it's because of this:
```
${@' ${prefix}/local/cuda-${CUDA_VERSION}/lib64' if d.getVar('baselib') != '64' and d.getVar('SITEINFO_BITS') == '64' else ''}"     
```

```
build$ find tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/etc
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/etc/nvidia-container-runtime
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/etc/nvidia-container-runtime/host-files-for-container.d
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/etc/nvidia-container-runtime/host-files-for-container.d/cuda-cudart.csv
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/etc/ld.so.conf.d
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/etc/ld.so.conf.d/cuda-10-0.conf
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64/libcudart_static.a
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64/libculibos.a
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64/libcudart.so
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64/libcudadevrt.a
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64/libcudart.so.10.0.326
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/lib64/libcudart.so.10.0
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/include
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/include/cuda_runtime.h
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/include/cudart_platform.h
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/include/cuda_device_runtime_api.h
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/local/cuda-10.0/include/cuda_runtime_api.h
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/lib64
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/lib64/pkgconfig
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/lib64/pkgconfig/cudart-10.0.pc
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc/cuda-cudart-10-0
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc/cuda-cudart-10-0/changelog.Debian.gz
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc/cuda-cudart-10-0/copyright
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc/cuda-cudart-dev-10-0
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc/cuda-cudart-dev-10-0/changelog.Debian.gz
tmp-glibc/work/armv8a_tegra210-wrs-linux/cuda-cudart/10.0.326-1-r1/image/usr/share/doc/cuda-cudart-dev-10-0/copyright

```

madisongh, 2020-06-17:
That `'64`' should be `'lib64'`, right?

paroque28, 2020-06-17:
Yes I believe so but let me try first

paroque28, 2020-06-17:
@madisongh Yes, changing `64` to `lib64` solved all the issues :)
Thanks for the help

madisongh, 2020-06-17:
Obvious typo, sorry I missed that when these changes first went in.  I'll push a fix.

madisongh, 2020-06-17:
OK, fix should be there in zeus-l4t-r32.3.1, dunfell, and later branches.