OE4T/meta-tegra #363: zeus-l4t-r32.3.1

lfdmn, 2020-06-18:
Hi,

I'm moving to zeus branch and have kernel compilation issues with both jetson-xavier and a custom machine (exact same error). Kernel builds ok, but kernel modules fail with:

```
...
|   CC [M]  drivers/net/wireless/realtek/rtl8821au/platform/platform_ops.o
|   CC [M]  drivers/net/wireless/realtek/rtl8821au/core/rtw_mp.o
|   LD [M]  drivers/net/wireless/realtek/rtl8814au/rtl8814au.o
|   LD [M]  drivers/net/wireless/realtek/rtl8821au/rtl8821au.o
|   Building modules, stage 2.
|   MODPOST 618 modules
| WARNING: could not open drivers/misc/mods/mods.dtb.S: No such file or directory
| ERROR: "snd_pcm_period_elapsed" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_card_disconnect" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_pcm_set_ops" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_card_free_when_closed" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_pcm_lib_ioctl" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_card_new" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_card_free" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_card_register" [drivers/hid/hid-jarvis-remote.ko] undefined!
| ERROR: "snd_pcm_new" [drivers/hid/hid-jarvis-remote.ko] undefined!
| /data/yocto/procbox-zeus/build-jetson-xavier/tmp/work-shared/jetson-xavier/kernel-source/scripts/Makefile.modpost:107: recipe for target '__modpost' failed
| make[3]: *** [__modpost] Error 1
| /data/yocto/procbox-zeus/build-jetson-xavier/tmp/work-shared/jetson-xavier/kernel-source/Makefile:1335: recipe for target 'modules' failed
| make[2]: *** [modules] Error 2
| Makefile:169: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:24: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
```

I use the defconfig from meta-tegra

Does it ring a bell to anyone?

madisongh, 2020-06-18:
If you have a customized defconfig that you're bringing forward from a prior L4T release, make sure you update it with whatever changes you need against the base configuration for the new version of the BSP.  NVIDIA made several changes between R32.2.x and R32.3.1 and the base defconfig has several significant differences.

Using the config-fragment support that was recently added to the kernel recipe should make it easier to do this going forward.

lfdmn, 2020-06-22:
`Using the config-fragment support that was recently added to the kernel recipe should make it easier to do this going forward.`
Good know! I had implemented my own fragment support but now I can cleanup.

I've narrowed down the issue. I get the compilation error (jetson-xavier MACHINE) when using this fragment to disable sound support
`# CONFIG_SOUND is not set`

Building with sound as module succeeds.

madisongh, 2020-11-02:
@lfdmn were you able to figure out the compilation error?  I have a custom TX2-based machine for which I've disabled CONFIG_SOUND, and I don't think I've had to patch anything in the kernel to get that to work.

closed 2020-12-11 by madisongh