OE4T/meta-tegra #365: TEGRA_SIGNING_ARGS add sed removal

lfdmn, 2020-06-22:
Hi,

How can I pass extra sed command to TEGRA_SIGNING_ARGS?

I need the extra 
```
-e/persistent.img/d -e/overlay.img/d -e/ota.img/d
```

otherwise the BUP signing will fail due to <filename> tag not being stripped in the partition layout.

I've tried many variant of this
```
TEGRA_SIGNING_ARGS += "\\\"-e/persistent.img/d\ -e/overlay.img/d -e/ota.img/d\\\"
```
But -e always get re-interpreted as command line option to tegra194-flash-helper.sh

madisongh, 2020-06-22:
That's right, TEGRA_SIGNING_ARGS is only used to pass extra arguments to the flash helper script for adding the `-u`/`-v` options to specify the signing keys.

Modifying the partition layout XML file on the fly is done by appending commands to one of the scriptlets.   One example is [here](https://github.com/madisongh/meta-mender-community/blob/patches-zeus-test/meta-mender-tegra/classes/image_types_mender_tegra.bbclass), in `tegraflash_generate_bupgen_script_append()` - it uses `sed` to append an additional `sed` command to the BUP generation script for modifying the XML file.

lfdmn, 2020-06-22:
Understood. I won't push my patch then :D 

Thanks!

closed 2020-06-22 by lfdmn