OE4T/meta-tegra #366: AppArmor on the NX with dunfell

dreambeyondorange, 2020-06-22:
Hi,

I tried to add the AppArmor distro feature and it fails looking for a .scc file

ndry@u46a4bc29d48753:/portable/jetson-nx 11:22:54 $ bitbake core-image-sato
Parsing recipes: 100% |##################################################################################################| Time: 0:00:56
Parsing of 2977 .bb files complete (0 cached, 2977 parsed). 4496 targets, 131 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "jetson-xavier-nx-devkit"
DISTRO               = "poky"
DISTRO_VERSION       = "3.1"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:9c8049068406532c3dd5d8906c595218b0fefd40"
meta-tegra           
contrib              = "dunfell-l4t-r32.4.2:58e4ffb2cc6b50cf10b2c2584e9e053f4fb02e1d"
meta-virtualization  = "dunfell:5c2de3daedd3f65171b9debf938683ce6e7746ea"
meta-browser         = "master:0acb73cd19d7ae3d105f2ae5ef2aa02e8d6c2a35"
meta-rust            = "common-rust-native:3b783652cc9e0e0ff95fdf151eab3e178b5f1667"
meta-clang           = "master:0c9da2c0646005d35b9845b4f91c5b25fb3d43d1"
meta-python2         = "dunfell:e2ef0dd8fa13d6b96e44773b09d07e4817d0a44d"
meta-oe              
meta-perl            
meta-networking      
meta-filesystems     
meta-python          
meta-webserver       = "dunfell:cc155826b0b4f202a8fa4ac103b7c38368d32734"
meta-selinux         = "master:f677a09992413f32e6b2c4c607ae44fb42de72c2"
meta-security        = "dunfell:217fadb03da925757b236cb91ba9807e3b2bbd6e"

Initialising tasks: 100% |###############################################################################################| Time: 0:00:05
Sstate summary: Wanted 3539 Found 3479 Missed 60 Current 0 (98% match, 0% complete)
NOTE: Executing Tasks
ERROR: linux-tegra-4.9.140+gitAUTOINC+a26c322c04-r0 do_kernel_metadata: Could not generate configuration queue for jetson-xavier-nx-devkit.
ERROR: linux-tegra-4.9.140+gitAUTOINC+a26c322c04-r0 do_kernel_metadata: Execution of '/portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/temp/run.do_kernel_metadata.8898' failed with exit code 1:
ERROR. input file "features/apparmor/apparmor.scc" does not exist
ERROR: could not process input files: /portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/defconfig /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/localversion_auto.cfg features/apparmor/apparmor.scc
       See /tmp/tmp.wkCu9e5ciL for details
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/temp/log.do_kernel_metadata.8898
Log data follows:
| DEBUG: Executing python function extend_recipe_sysroot
| NOTE: Direct dependencies are ['/portable/jetson-nx/meta/recipes-kernel/kern-tools/kern-tools-native_git.bb:do_populate_sysroot']
| NOTE: Installed into sysroot: []
| NOTE: Skipping as already exists in sysroot: ['kern-tools-native', 'quilt-native']
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Searching for localversion_auto.cfg in paths:
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/poky
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/poky
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/poky
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/poky
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/jetson-xavier-nx-devkit
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/jetson-xavier-nx-devkit
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/jetson-xavier-nx-devkit
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/jetson-xavier-nx-devkit
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/xavier-nx
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/xavier-nx
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/xavier-nx
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/xavier-nx
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/tegra194
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/tegra194
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/tegra194
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/tegra194
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/armv8a
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/armv8a
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/armv8a
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/armv8a
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/tegra
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/tegra
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/tegra
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/tegra
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/cuda
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/cuda
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/cuda
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/cuda
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/aarch64
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9.140+gitAUTOINC+a26c322c04/
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra/
|     /portable/jetson-nx/meta-tegra/recipes-kernel/linux/files/
| DEBUG: Executing shell function do_kernel_metadata
| ERROR. input file "features/apparmor/apparmor.scc" does not exist
| ERROR: could not process input files: /portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/defconfig /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/localversion_auto.cfg features/apparmor/apparmor.scc
|        See /tmp/tmp.wkCu9e5ciL for details
| ERROR: Could not generate configuration queue for jetson-xavier-nx-devkit.
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/temp/run.do_kernel_metadata.8898' failed with exit code 1:
| ERROR. input file "features/apparmor/apparmor.scc" does not exist
| ERROR: could not process input files: /portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/defconfig /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/localversion_auto.cfg features/apparmor/apparmor.scc
|        See /tmp/tmp.wkCu9e5ciL for details
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra_4.9.bb:do_kernel_metadata) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2761 tasks of which 2732 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra_4.9.bb:do_kernel_metadata
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

Have you tried adding appArmor to an NVIDIA build with success?

Thanks!
Ryan

lfdmn, 2020-06-23:
@dreambeyondorange I have not tried personally.

Just doing a quick grep on zeus branch I can see the defconfig for Xavier AGX have the kernel feature disabled by default

`# CONFIG_SECURITY_APPARMOR is not set`

Did you enable it in your build?

madisongh, 2020-06-23:
> ```ERROR: could not process input files: /portable/jetson-nx/build/tmp/work/jetson_xavier_nx_devkit-poky-linux/linux-tegra/4.9.140+gitAUTOINC+a26c322c04-r0/defconfig /portable/jetson-nx/meta-tegra/recipes-kernel/linux/linux-tegra-4.9/localversion_auto.cfg features/apparmor/apparmor.scc```

Start with tracking down where `features/apparmor/apparmor.scc` is getting added to the SRC_URI (using `bitbake -e virtual/kernel` should help with this).  I took a quick look at the kernel bbappends in meta-security, and it looks to me like they should only apply for version 5.x kernels, which ours isn't. Even if it were, we don't pull in the Yocto Linux kernel metadata  (see discussion on #358), so you would still get this failure. And even if we did, that `scc` file doesn't appear to be present in the branch for 4.9 kernels, so you would *still* get this failure.

If the addition is happening in a bbappend you don't have control over, adding
```
KERNEL_FEATURES_remove = "features/apparmor/apparmor.scc"
```
to your own bbappend file for the kernel should at least work around this failure.

To actually enable apparmor in the kernel, you should add a config fragment that enables it, as @ldfmn mentions above.  Looking at the cfg file [in yocto-kernel-cache](https://git.yoctoproject.org/cgit/cgit.cgi/yocto-kernel-cache/tree/features/apparmor/apparmor.cfg), there are some other configs that may also need to be turned on, although the ones that are mentioned in that file may not apply to a kernel as old as ours.

dreambeyondorange, 2020-06-23:
I thought adding 'DISTRO_FEATURES_append += " apparmor"' would do it, but that doesn't seem to help.  I'm not 100% sure how I would enable apparmor config.  Is it a different variable or is it a bb apend file to somehow change the defconfig?

Thanks!
Ryah

madisongh, 2020-06-23:
Add a `.cfg` file to the SRC_URI in a bbappend file for the kernel recipe.   You could try starting with the one in yocto-kernel-cache that I pointed at earlier.  The recipe will see the `.cfg` suffix and append it to the kernel configuration.

madisongh, 2020-12-11:
Closing as it's been a while, and I assume you got your kernel configuration sorted.

closed 2020-12-11 by madisongh