OE4T/meta-tegra #368: zeus SDK build issue

lfdmn, 2020-06-23:
Hey,

Anyone having issue building the SDK image for Zeus branch?

I get some missing runtime dependency issues at QA for gcc-cross-canadian
- libstdc++.so.6
- libm.so.6
- libgcc_s.so.1
- libc.so.6
- libdl.so.2
- libgmp.so.10
- libz.so.1

Here's one example:
`ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/procbox/1.2.0/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(GLIBCXX_3.4.20)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]`

I switched to the contrib layer, blaming my own, but same problem.

lfdmn, 2020-06-23:
I managed to fix it. Here's a patch

```diff
diff --git a/contrib/recipes-devtools/gcc/gcc-cross-canadian.inc b/contrib/recipes-devtools/gcc/gcc-cross-canadian.inc
index e7c08d3..57dd55d 100644
--- a/contrib/recipes-devtools/gcc/gcc-cross-canadian.inc
+++ b/contrib/recipes-devtools/gcc/gcc-cross-canadian.inc
@@ -4,7 +4,7 @@ SUMMARY = "GNU cc and gcc C compilers (cross-canadian for ${TARGET_ARCH} target)
 PN = "gcc-cross-canadian-${TRANSLATED_TARGET_ARCH}"
 
 DEPENDS = "virtual/${TARGET_PREFIX}gcc virtual/${HOST_PREFIX}gcc-crosssdk virtual/${HOST_PREFIX}binutils-crosssdk virtual/nativesdk-libc nativesdk-gettext flex-native virtual/libc"
-
+RDEPENDS_${PN} += "nativesdk-libstdc++ nativesdk-libgcc nativesdk-glibc nativesdk-zlib nativesdk-gmp"
 GCCMULTILIB = "--enable-multilib"
 
 require gcc-configure-common.inc

```

madisongh, 2020-06-23:
Strange.  I would have thought that the shared-library processing performed in the packaging step would have automatically picked up on those runtime dependencies.

madisongh, 2020-06-23:
Cherry-picking bb2a00e27947a21958809a4fbfed26886f6fba03 from the dunfell branch might take care of this.

lfdmn, 2020-06-26:
That works

madisongh, 2020-06-26:
I've cherry-picked that change to the zeus branch.

closed 2020-06-26 by madisongh