OE4T/meta-tegra #371: [Question] Is it possible to use the 'sumo' branch for Jetson AGX Xavier?

levioctl, 2020-06-25:
Hi
I have a Jetson Xavier machine, and I would like to build the meta-tegra layer on the 'sumo' release, however it seems that support for Jetson Xavier was only added in the thud release.
I was wondering if it's possible, and if so, how much effort it would take, as I'm fairly new to the Yocto project. That is, is it possible to achieve with some guidance over a period of a few days/weeks.

Thanks!

madisongh, 2020-06-25:
It's possible, but it would be a fair amount of work.  The `sumo` branch is based off the old L4T R28.2 release, as is `thud`.  Xavier support first arrived with L4T R32.1 on the `warrior` branch.

Depending on what features you want to use - CUDA, CuDNN, TensorRT, or any of the other JetPack packages - you'd be better off using L4T R32.3.1/JetPack 4.3.  The `zeus-l4t-r32.3.1` and `dunfell` branches both support that version and are actively maintained.

@leon-anavi backported the R32.3.1 recipes to a `thud` base - that's the `thud-l4t-r32.3.1` branch.  You could compare the contents of that branch against `sumo` as a starting point.  That should at least reduce the effort required.

madisongh, 2020-12-11:
Closing as the question is answered about as well as it can be.

closed 2020-12-11 by madisongh