OE4T/meta-tegra #372: Using MMAPI on zeus branch crashes the camera subsystem

klakaiser, 2020-06-26:
Hi,
as i moved from warrior to zeus branch i encounter trouble using the MMAPI. The camera system is up and running which is proved by using v4l2-ctl and a simple gstreamer pipeline..

`v4l2-ctl -d /dev/video2 -w --set-fmt-video=width=1948,height=1096,pixelformat=BA12 --stream-mmap --set-ctrl bypass_mode=0 --stream-count=1000000`

`gst-launch-1.0 -e -v nvarguscamerasrc sensor-id=2  ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=(fraction)30/1' ! nvvidconv ! xvimagesink`

As soon as i make use of the MMAPI (for example the argus_camera) the camera subsystem crashes.

Do you have an idea what is going on or how i can debug down this issue?

Thanks,
Klaus

nvargus-daemon with enabled logs:
-----------------------------------------------
=== NVIDIA Libargus Camera Service (0.97.3)=== Listening for connections...=== argus_camera[5494]: Connection established (7F8B3901C0)OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module0
OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module1
OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module2
OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module3
OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module4
OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module5
NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
NvPclHwGetModuleList: No module data found
NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
NvPclHwGetModuleList: No module data found
NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
NvPclHwGetModuleList: No module data found
NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
NvPclHwGetModuleList: No module data found
NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
NvPclHwGetModuleList: No module data found
NvPclHwGetModuleList: WARNING: Could not map module to ISP config string
NvPclHwGetModuleList: No module data found
OFParserGetVirtualDevice: NVIDIA Camera virtual enumerator not found in proc device-tree
---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
---- imager: Found override file [/var/nvidia/nvcam/settings/camera_overrides.isp]. ----
=== argus_camera[5494]: CameraProvider initialized (0x7f84b60940)SCF: Error BadValue: NvPHSSendThroughputHints (in src/common/CameraPowerHint.cpp, function sendCameraPowerHint(), line 56)

dmesg:
-----------
<--- snip of the log --->
Argus Version: 0.97.3 (multi-process)
[  354.788360] imx290 2-0022: imx290_power_on: power on
[  356.483079] fence timeout on [ffffffc1e6fd5cc0] after 1500ms
[  356.483083] fence timeout on [ffffffc1e6fd5b40] after 1500ms
[  356.483093] name=[nvhost_sync:23], current value=0 waiting value=1
[  356.483097] ---- mlocks ----
[  356.483110] 8: locked by channel 7
[  356.483126] 
[  356.483129] ---- syncpts ----
[  356.483136] id 2 (disp_a) min 4 max 4 refs 1 (previous client : )
[  356.483140] id 3 (disp_b) min 2 max 2 refs 1 (previous client : )
[  356.483143] id 4 (disp_c) min 2 max 2 refs 1 (previous client : )
[  356.483149] id 8 (vblank0) min 179 max -6 refs 1 (previous client : )
[  356.483164] id 20 (gp10b_507) min 3954 max 3954 refs 1 (previous client : gp10b_507)
[  356.483166] id 21 (gp10b_506) min 20 max 20 refs 1 (previous client : gp10b_506)
[  356.483170] id 23 (15600000.isp_nvargus-daemon_0) min 0 max 4 refs 4 (previous client : )
[  356.483173] id 24 (15600000.isp_nvargus-daemon_1) min 0 max 2 refs 4 (previous client : )
[  356.483175] id 25 (15600000.isp_nvargus-daemon_2) min 0 max 2 refs 4 (previous client : )
[  356.483180] id 28 (15600000.isp_nvargus-daemon_3) min 0 max 2 refs 4 (previous client : )
[  356.483182] id 29 (15600000.isp_nvargus-daemon_4) min 15 max 26 refs 9 (previous client : )
[  356.483185] id 30 (15600000.isp_nvargus-daemon_5) min 0 max 2 refs 4 (previous client : )
[  356.483189] id 32 (gp10b_504) min 10 max 10 refs 1 (previous client : )
[  356.483195] id 36 (15700000.vi_0) min 14 max 14 refs 2 (previous client : 15700000.vi_0)
[  356.483198] id 37 (15700000.vi_0) min 1 max 2 refs 4 (previous client : 15700000.vi_0)
[  356.483201] id 38 (15700000.vi_1) min 0 max 2 refs 4 (previous client : 15700000.vi_1)
[  356.483203] id 39 (15700000.vi_2) min 0 max 2 refs 4 (previous client : 15700000.vi_2)
[  356.483206] id 40 (150c0000.nvcsi_0) min 26 max 26 refs 2 (previous client : 150c0000.nvcsi_0)
[  356.483211] id 43 (gp10b_500) min 8 max 8 refs 1 (previous client : )
[  356.483213] id 44 (gp10b_499) min 6 max 6 refs 1 (previous client : )
[  356.483216] id 45 (gp10b_498) min 6 max 6 refs 1 (previous client : )
[  356.483219] id 46 (gp10b_497) min 6 max 6 refs 1 (previous client : )
[  356.483221] id 47 (gp10b_496) min 6 max 6 refs 1 (previous client : )
[  356.483224] id 48 (gp10b_495) min 6 max 6 refs 1 (previous client : )
[  356.483227] id 49 (gp10b_494) min 6 max 6 refs 1 (previous client : )
[  356.483736] 
[  356.483737] ---- channels ----
[  356.483745] 
[  356.483745] channel 2 - 15820000.se
[  356.483745] 
[  356.483746] NvHost basic channel registers:
[  356.483749] CMDFIFO_STAT_0:  00002040
[  356.483751] CMDFIFO_RDATA_0: 00a93ec9
[  356.483754] CMDP_OFFSET_0:   00000000
[  356.483756] CMDP_CLASS_0:    00000000
[  356.483758] CHANNELSTAT_0:   00000000
[  356.483760] The CDMA sync queue is empty.

madisongh, 2020-06-28:
Debugging this kind of issue is definitely a pain. You may be able to get assistance from your camera vendor; otherwise, I'd suggest searching the NVIDIA dev forums for some pointers, since it looks like others have had similar problems.
A couple of links I found:
 * https://forums.developer.nvidia.com/t/nvcamera-daemon-breakage/58658/19
 * https://elinux.org/Jetson_TX2_Camera_BringUp  (has info on increasing log levels in nvargus-daemon)
 
Also make sure that your vendor has supplied you with a driver, device tree changes, and camera_overrides suitable for the version of the L4T BSP you're using - on the zeus branch, that should be R32.2.3.  I know from past experience that NVIDIA can make changes from version to version that can break previously-working drivers.

klakaiser, 2020-06-29:
You are right, the complete nvidia argus stuff does weird things and is a pain from the beginning on. To bad that this is all closed source.
I just wonder as the 'nvarguscamerasrc' is working fine (i get a nice video stream out of my cameras), so i assume that driver and device tree must be correct. Do you know of any config difference between the usage of:

nvargus-daemon <--> nvarguscamerasrc      (used by gstreamer)
nvargus-daemon <--> nvargus_socketclient  (used by the mmapi)

madisongh, 2020-06-29:
IIRC, the gstreamer plugin might not be enumerating all of the possible supported features that the argus_camera application does, so it may not be exercising all the same code paths. Enabling (or adding) debug logs in the camera driver and comparing the output of the working vs. non-working cases might help.

klakaiser, 2020-06-29:
So i am currently able to get my application up and running but only if i start the /usr/sbin/nvargus-daemon by hand in a shell.
When i start it via the systemd it seems to be running much to slow so that i get timeouts from the driver. I saw that you reworked the systemd stuff. Do you have an idea why starting binaries via systemd can result in such a behaviour?

madisongh, 2020-06-29:
That's certainly odd.  After closing down your successful manual run, can you try starting the nvargus-daemon service again and then see if your application runs OK? It's possible that there's some kind of dependency between the argus daemon startup and the other NVIDIA services such that it runs suboptimally if started too early.

klakaiser, 2020-06-30:
Tried that on but still no success. I already fully disabled the nvargus-daemon and only start it manually during the session
`systemctl start nvargus-daemon` which makes no difference. As soon as i do so the camera subsystem fails. When i start it by hand `/usr/sbin/nvargus-daemon` everything is fine and i receive images from my four connected cameras. Then i stop it again, start it via systemctl start and only get timeout from the iConsumer_->acquireFrame() function.

I already tried disabling the journalctl or instructing the service to not output anything, with no success.

So you don't have any problems on the zeus-l4t-r32.3.1 branch  with the nvargus-daemon?

madisongh, 2020-06-30:
Hmm.  No, I haven't had any issues with nvargus-daemon crashing like this, but I haven't had any projects with multiple MIPI cameras - all single-camera projects.

Maybe systemd is setting some process limits or capabilities that are interfering here?

klakaiser, 2020-06-30:
Ok, i will try to find out if thee are any limits.
I just found out that it works perfectly with a single camera. 
So nvargus-daemon started by systemd (during startup or manual) works as long as i only use one camera.
As soon as i start my app with using more it runs in a timeout and returns no images at all.

But when i start it by hand, up to six cameras work perfectly (but much slower than on a warrior system).

Seems like some' is broken in argus daemon or fully has changed.

madisongh, 2020-06-30:
Searching around, I do see other reports of multi-camera setups causing nvargus-daemon crashes, with no solutions posted.  And I found this multi-camera demo at [RidgeRun](https://developer.ridgerun.com/wiki/index.php?title=RidgeRun_D3_NVIDIA_Partner_Showcase_Jetson_Xavier_Multi-Camera_AI_Demo) where they explicitly stop the systemd service and run it from the shell.

klakaiser, 2020-07-02:
Thank you for your help in understanding this issue. 
I am going to close this as it seems that this has nothing to do with your layer.

closed 2020-07-02 by klakaiser