OE4T/meta-tegra #373: Faster flashing and smaller SD card size

mozcelikors, 2020-06-27:
For 16GB image size, SD card flashing takes a lot of time with dd ( For me I am flashing with 11 MB/s USB 2.0 SD card reader, takes around 25 minutes to flash) . Most of the 16GB is unused space. 
```
Filesystem      Size  Used Avail Use% Mounted on                                    
/dev/mmcblk0p1   14G  1.3G   12G  11% /   
```                                          
Is there a way to make default SD card size smaller, e.g. from both machine.conf and tegraflash package?
I tried to change `TEGRAFLASH_SDCARD_SIZE ?= "16G"` to 4G but I got other errors.
I'd appreciate if you could help regarding this.

mozcelikors, 2020-06-27:
I am going to try:
TEGRAFLASH_SDCARD_SIZE = "6G"
ROOTFSPART_SIZE = "4294967296"

mozcelikors, 2020-06-28:
Also setting ROOTFSPART_SIZE worked (in gibibytes).

closed 2020-06-28 by mozcelikors

madisongh, 2020-06-28:
If you're using the `dunfell` branch (or `master), you can also speed up SDcard writing by installing `bmaptool` on your development host.  The `make-sdcard` script will automatically use that tool, if it's available in your PATH. For this optimization to be effective you have to run `dosdcard.sh` to directly write to an SDcard device by specifying the device name on the command line.

This optimization is also available on the `zeus-l4t-r32.3.1` branch, if you switch to the compressed-tarball format for the tegraflash package (setting TEGRAFLASH_PACKAGE_FORMAT to `"tar"` in your local.conf), so the filesystem image retains its sparse-file layout (which zip does not preserve).