OE4T/meta-tegra #374: yocto QT5 meta-toolchain-qt5 error

przemyslaw-chwiala, 2020-06-29:
Hey, I'm trying to cross compile my c++ application with yocto. 
I followed this https://variwiki.com/index.php?title=Yocto_Qt5_Toolchain_installation
I got following error
```
NOTE: Running task 5170 of 5289 (/home/fuji/safeway-by-sternkraft/build/../meta-qt5/recipes-qt/meta/meta-toolchain-qt5.bb:do_populate_sdk)
NOTE: recipe meta-toolchain-qt5-1.0-r0: task do_populate_sdk: Started
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(GLIBCXX_3.4.20)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(CXXABI_1.3.9)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(GLIBCXX_3.4.21)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(GLIBCXX_3.4)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libgcc_s.so.1(GCC_4.2.0)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libgcc_s.so.1(GCC_3.3)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto-wrapper contained in package gcc-cross-canadian-aarch64 requires libc.so.6(GLIBC_2.11)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libc.so.6(GLIBC_2.7)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libc.so.6(GLIBC_2.14)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libc.so.6(GLIBC_2.3)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libc.so.6(GLIBC_2.2.5)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libm.so.6()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libgcc_s.so.1()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/plugin/gengtype contained in package gcc-cross-canadian-aarch64 requires libc.so.6()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libgcc_s.so.1(GCC_3.4)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-gcov-tool contained in package gcc-cross-canadian-aarch64 requires libc.so.6(GLIBC_2.3.3)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libm.so.6(GLIBC_2.2.5)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libdl.so.2(GLIBC_2.2.5)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(CXXABI_1.3)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libstdc++.so.6(GLIBCXX_3.4.15)(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libgmp.so.10()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA Issue: /opt/poky/3.0.3/sysroots/x86_64-pokysdk-linux/usr/libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/7.3.0/lto1 contained in package gcc-cross-canadian-aarch64 requires libdl.so.2()(64bit), but no providers found in RDEPENDS_gcc-cross-canadian-aarch64? [file-rdeps]
[... cut ...]
ERROR: gcc-cross-canadian-aarch64-7.3.0-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: Logfile of failure stored in: /home/fuji/safeway-by-sternkraft/build/tmp/work/x86_64-nativesdk-pokysdk-linux/gcc-cross-canadian-aarch64/7.3.0-r0/temp/log.do_package_qa.17117
NOTE: recipe gcc-cross-canadian-aarch64-7.3.0-r0: task do_package_qa: Failed
ERROR: Task (/home/fuji/safeway-by-sternkraft/build/../meta-tegra/contrib/recipes-devtools/gcc/gcc-cross-canadian_7.3.bb:do_package_qa) failed with exit code '1'
```

madisongh, 2020-06-29:
Have you updated to zeus latest recently? [This patch](https://github.com/madisongh/meta-tegra/commit/d6ce4c542c151f9bb9c0c701392794a2f2045770) went in a few days ago, which should fix this issue.

przemyslaw-chwiala, 2020-06-29:
Yeah, it works. Thank you!

closed 2020-06-29 by przemyslaw-chwiala