OE4T/meta-tegra #376: MACHINE=jetson-xavier-nx-devkit is invalid

outbackdingo, 2020-07-04:
Ooops... whats changed ?

MACHINE=jetson-xavier-nx-devkit is invalid. Please set a valid MACHINE in your local.conf

madisongh, 2020-07-04:
Missing meta-tegra from your bblayers.conf, maybe?  Or on the wrong branch?

outbackdingo, 2020-07-04:
cat conf/bblayers.conf 
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/dingo/nvidia/poky/meta \
  /home/dingo/nvidia/poky/meta-poky \
  /home/dingo/nvidia/poky/meta-yocto-bsp \
  /home/dingo/nvidia/meta-openembedded/meta-oe \
  /home/dingo/nvidia/meta-openembedded/meta-multimedia \
  /home/dingo/nvidia/meta-openembedded/meta-networking \
  /home/dingo/nvidia/meta-openembedded/meta-filesystems \
  /home/dingo/nvidia/meta-openembedded/meta-perl \
  /home/dingo/nvidia/meta-openembedded/meta-python \
  /home/dingo/nvidia/meta-openembedded/meta-gnome \
  /home/dingo/nvidia/meta-openembedded/meta-xfce \
  /home/dingo/nvidia/meta-virtualization \
  /home/dingo/nvidia/meta-tegra \
  /home/dingo/nvidia/meta-mortsgna \
  /home/dingo/nvidia/meta-clang \
  /home/dingo/nvidia/meta-qt5 \
  /home/dingo/nvidia/meta-qt5-extra \
  "
dingo@gitlab:~/nvidia/xavier$ git branch
fatal: not a git repository (or any of the parent directories): .git
dingo@gitlab:~/nvidia/xavier$ cd ../meta-tegra
dingo@gitlab:~/nvidia/meta-tegra$ git branch
* dunfell
  master
  warrior
  zeus

madisongh, 2020-07-04:
Xavier NX support is not present on dunfell.  You need to be on dunfell-l4t-r32.4.2.

outbackdingo, 2020-07-04:
dammit... sorry! right wrong branch....

closed 2020-07-04 by outbackdingo