OE4T/meta-tegra #378: Compile darknet for jetson nano

nandra, 2020-07-06:
Hi,

I'm writing recipe to compile [darknet](https://github.com/AlexeyAB/darknet) for jetson nano.
Recipe looks like:
```
SUMMARY = "Darknet: Open Source Neural Networks in C"
DESCRIPTION = "Darknet is an open source neural network framework written in C."
HOMEPAGE="http://pjreddie.com/darknet/"
SECTION = "libs"
PRIORITY= "optional"
LICENSE = "PD"
PR = "r0"

LIC_FILES_CHKSUM = "file://LICENSE;md5=4714f70f7f315d04508e3fd63d9b9232"

SRC_URI = " \
    git://github.com/AlexeyAB/darknet.git;protocol=https \
"

SRCREV = "3af6781de273fa01f4a535c8e40c5056e809e22f"

S = "${WORKDIR}/git"

DEPENDS += "opencv stb cudnn"

inherit cmake cuda

EXTRA_OECMAKE += "-DENABLE_ZED_CAMERA=OFF"
```

I'm hitting some strange issue with cmake configure phase like:
```
| -- Found CUDA: /home/marek/projects/test/build/tmp/work/armv8a_tegra210-poky-linux/darknet/git-r0/recipe-sysroot-native/usr/local/cuda-10.0 (found version "10.0")
| CMake Error: TRY_RUN() invoked in cross-compiling mode, please set the following cache variables appropriately:
|    run_result (advanced)
|    run_result__TRYRUN_OUTPUT (advanced)
| For details see /home/marek/projects/test/build/tmp/work/armv8a_tegra210-poky-linux/darknet/git-r0/build/TryRunResults.cmake
| -- Automatic GPU detection failed. Building for common architectures.
| -- Autodetected CUDA architecture(s): 3.0;3.5;5.0;5.2;6.0;6.1;7.0;7.0+PTX;7.5;7.5+PTX
```

I'm not sure about inherit cuda line in recipe (I'm new to nvidia stuff so pls bear with me ;))
Any ideas what can be checked or from where this error comes (I tried to search but cannot find anything obvious). Thanks.

madisongh, 2020-07-06:
Looks like the cmake files are trying to compile and run a program to automatically detect the CUDA architecture, which isn't a good idea when cross-compiling.

If I'm reading the CMakeLists.txt file right, you should be able to add `-DCUDA_ARCHITECTURES=5.3` (for tegra210) to EXTRA_OECMAKE to bypass the automatic selection.

nandra, 2020-07-06:
@madisongh ok thanks. That did the trick. I'm now struggle with compilation :). Would be maybe good to add this recipe to meta-tegra? IMO could be beneficial for other also. Thanks.

nandra, 2020-07-06:
@madisongh I have another issue that cuda is detected native (not for target) and have some issue with include cuda_runtime. Ideas? Thanks.

nandra, 2020-07-06:
Wel now I understand that native cuda is used for "crosscompilation" for GPU. I have some tweaks in sysroot-native (added to /usr/local/cuda-10/targets/aarch64-linux target with copy include + libs from target) and compilation almost pass just hit linker issue ```libdark.so: undefined reference to `cuCtxGetCurrent'```

madisongh, 2020-07-06:
Yeah, nvcc isn't CPU-architecture dependent, so it always lives in the native-side CUDA tree.  For our cross-builds, headers and libraries are in the target-side sysroot, so the CMake files may need to be tweaked to properly differentiate between CMAKE_CUDA_TOOLKIT_ROOT (where nvcc lives) and CMAKE_CUDA_TOOLKIT_TARGET_DIR (where the target-side headers and libraries live).

madisongh, 2020-12-11:
@nandra Since it's been a while, I'll assume you got this building.

closed 2020-12-11 by madisongh