OE4T/meta-tegra #380: cannonical configuration to building for multiple types of boards in same tree

ericvh, 2020-07-06:
Starting to generalize my build environment for building across Nano, NX, AGX, TX1, TX2 -- while I can set NVIDIA_SDK_DOWNLOAD to the right directory for each of the build types -- is there a better way (ie. establishing a cannonical naming convention like x/nvidia-sdk/<ver>/<target> (ie. 4.4/jetson-xavier).

Is there an established way of doing this?

madisongh, 2020-07-06:
I use a single NVIDIA_DEVNET_MIRROR across all the platforms and across all of the BSP/JetPack versions - I just copy the files out of the SDK Manager's download area into the one directory, with the DLA and NoDLA subdirectories for the recent TensorRT versions... with the exception of TensorRT, package naming is unambiguous for the JetPack downloads.

This has worked OK for me since L4T R32.x, at least.

ericvh, 2020-07-11:
a (loosely) related question -- I also took a stab at building a different archtiecture (x86) in the same build tree (which shouldn't be referencing any of the tegra recipes, but the layer is still in the tree) -- but this seems to run afoul of meta-tegra class definitions.  Is that expected behavior or have I messed something up.

madisongh, 2020-07-11:
> Is that expected behavior

No.  Well, at least not desired behavior.

> or have I messed something up.

Maybe, although it's perfectly possible something is broken in meta-tegra ... which branch are you using?  I don't continuously test builds for non-tegra platforms, but I do so periodically, and will fix errors in the classes/recipes that interfere with such builds.

ericvh, 2020-07-11:
okay, I tried pulling out a clean minimal poky and then adding tegra and the problem doesn't happen - so its something to do with something else in my setup.  Sorry for the noise.

ericvh, 2020-07-12:
I think I narrowed this down, the problem doesn't happen on the dunfell branch, but does happen on the dunfell-l4t-32.4.2 branch

Verified this in my clean tree (which I had originally tested with dunfell, not dunfell-32.4.2)

`ERROR: /work/smarter-yocto/poky/build/conf/../../../meta-tegra/recipes-devtools/cuda/libcublas_10.2.2.89-1.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 2, function: <module>
     0001:__anon_32__work_smarter_yocto_poky_build_conf_______meta_classes_patch_bbclass(d)
 *** 0002:__anon_707__work_smarter_yocto_poky_build_conf_______meta_classes_base_bbclass(d)
     0003:__anon_202__work_smarter_yocto_poky_build_conf_______meta_classes_reproducible_build_bbclass(d)
     0004:__anon_1106__work_smarter_yocto_poky_build_conf_______meta_classes_insane_bbclass(d)
     0005:__anon_1346__work_smarter_yocto_poky_build_conf_______meta_classes_insane_bbclass(d)
     0006:__anon_252__work_smarter_yocto_poky_build_conf_______meta_classes_package_bbclass(d)
File: '/work/smarter-yocto/poky/build/conf/../../meta/classes/base.bbclass', lineno: 590, function: __anon_707__work_smarter_yocto_poky_build_conf_______meta_classes_base_bbclass
     0586:
     0587:    needsrcrev = False
     0588:    srcuri = d.getVar('SRC_URI')
     0589:    for uri in srcuri.split():
 *** 0590:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
     0591:
     0592:        # HTTP/FTP use the wget fetcher
     0593:        if scheme in ("http", "https", "ftp"):
     0594:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
File: '/work/smarter-yocto/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 357, function: decodeurl
     0353:    """
     0354:
     0355:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/;]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0356:    if not m:
 *** 0357:        raise MalformedUrl(url)
     0358:
     0359:    type = m.group('type')
     0360:    location = m.group('location')
     0361:    if not location:
Exception: bb.fetch2.MalformedUrl: The URL: '${L4T_DEB_FEED_BASE}/common/pool/main/c/cuda/cuda-license-10-2_10.2.89-1_amd64.deb;name=lic;subdir=libcublas-10.2.2.89-1' is invalid and cannot be interpreted`

madisongh, 2020-07-13:
Right, thanks. I've cherry-picked b5fbfcfd from master, which should take care of it.

ericvh, 2020-07-13:
that seems to have fixed it.  Thanks.

ericvh, 2020-07-17:
okay, I've fixed all my issues now and am building for multiple platforms cleanly.  Thanks.

closed 2020-07-17 by ericvh