OE4T/meta-tegra #382: How to build cuda nbody sample

bcis93, 2020-07-07:
Hi, I am working with the dunfell-l4t-r32.4.2 branch, and I want to run the CUDA nbody sample on my target (TX2). I found the `cuda-samples` recipe, and I was able to successfully build/run the two samples it builds, so I thought I would try to modify this recipe to build the nbody sample as well. For a quick test, I just added the nbody sample to `CUDA_SAMPLES` like this:
```
CUDA_SAMPLES ?= " \
    0_Simple/UnifiedMemoryStreams \
    1_Utilities/deviceQuery \
    5_Simulations/nbody \
"
```
After running a build like this, the build finished successfully, but the nbody sample didn't get installed. On closer inspection of the `cuda-samples` working directory, it seems that the nbody sample failed to even compile because I don't see the nbody binary file anywhere (but I can see the binaries for `deviceQuery` and `UnifiedMemoryStreams`). This was surprising to me because I didn't get any errors from bitbake - is there something I'm missing? Do you have any suggestions on how to build the nbody sample?

madisongh, 2020-07-07:
I took a quick look at that sample.... it looks like it depends on having libGL, libGLU, and libglut available, so at the very least you'll need to add some dependencies to DEPENDS (probably "virtual/libgl libglu freeglut", freeglut being in meta-oe). You may need to make some additional changes either to the makefile or to the EXTRA_OEMAKE setting in the recipe to override where the makefile looks for those libraries.

I stuck with compiling just those two samples to minimize the number of dependencies and to avoid having to patch the makefiles, which use some unusual techniques to find dependencies and construct the flags passed to the compiler and linker. They're really geared for building directly on the device, and more or less assume you're running one of the mainstream distros. It should be possible to cross-build this one, too, with some effort.

And yes, it looks like the makefiles don't exit with errors when one of the samples fails to build, so bitbake doesn't know about it.

bcis93, 2020-07-07:
Thanks for the response. I'll look into making those modifications, or maybe I'll just end up building them on the target.

bcis93, 2020-07-17:
Thanks for the suggestions. I ended up just building them directly on the device, which worked fine for me.

closed 2020-07-17 by bcis93