OE4T/meta-tegra #383: [jetson-nano] Mounting the root filesystem via NFS (nfsroot)

elPrac, 2020-07-07:
Hi everyone,

I was trying to mount rootfs via NFS setting ${bootargs} following [kernel-doc](https://www.kernel.org/doc/Documentation/filesystems/nfs/nfsroot.txt) instructions so, from U-boot i set the following variables:
```
Tegra210 (P3450-Porg) # setenv serverip 192.168.1.88
Tegra210 (P3450-Porg) # setenv nfsroot /nfsroot
Tegra210 (P3450-Porg) # setenv bootargs root=/dev/nfs rootfstype=nfs ip=dhcp nfsroot=${serverip}:${nfsroot}
```
But after boot system hangs...
```
[   10.038001] IP-Config: Got DHCP answer from 192.168.1.254, my address is 192.168.1.120
[   10.045973] IP-Config: Complete:
[   10.049195]      device=eth0, hwaddr=00:04:4b:e6:61:84, ipaddr=192.168.1.120, mask=255.255.255.0, gw=192.168.1.254
[   10.059535]      host=192.168.1.120, domain=, nis-domain=(none)
[   10.065454]      bootserver=0.0.0.0, rootserver=192.168.1.88, rootpath=     nameserver0=192.168.1.254
[   10.075201] vdd-fan: disabling
[   10.078305] vdd-usb-vbus: disabling
[   10.081786] vdd-usb-vbus2: disabling
[   10.085371] avdd-io-edp-1v05: disabling
[   10.089353] ALSA device list:
[   10.092327]   #0: tegra-hda at 0x70038000 irq 83
[   10.096949]   #1: tegra-snd-t210ref-mobile-rt565x
[   10.104707] Freeing unused kernel memory: 8576K
[   10.190010] usb 2-1: usb_suspend_both: status 0
[   10.194657] usb usb2: usb_suspend_both: status 0
[   56.849998] random: crng init done

Nothing happened after this...
```

I was thinking that maybe i didn't have an available ttyS0 so i tried with...
```
ssh root@192.168.1.120
ssh: connect to host 192.168.1.120 port 22: Connection refused
```
And keep track of my NFS-server logs:
```
$ sudo tail -f  /var/log/syslog
Jul  6 20:25:04 prac-desktop-ubuntu kernel: [1241326.416534] NFSD: laundromat service - starting
Jul  6 20:25:04 prac-desktop-ubuntu kernel: [1241326.416536] NFSD: end of grace period
Jul  6 20:25:04 prac-desktop-ubuntu kernel: [1241326.416538] NFSD: laundromat_main - sleeping for 90 seconds
```
So... i'm kind of sure that kernel just freeze.
I'm new with this so i tried to keep things simple before any attempt to work with PXELINUX so i  manually modified 
extlinux.conf(in my sdcard)  like so...
```
# Generic Distro Configuration file generated by OpenEmbedded
LABEL Poky (Yocto Project Reference Distro)
	KERNEL ../Image
	INITRD ../initrd
	APPEND ${bootargs}
```
and just add a console in my bootargs:
```
Tegra210 (P3450-Porg) # setenv bootargs root=/dev/nfs rootfstype=nfs ip=dhcp nfsroot=${serverip}:${nfsroot} console=ttyS0,115200n8
```
After this i see the following error
```
[   10.142107] IP-Config: Got DHCP answer from 192.168.1.254, my address is 192.168.1.120
[   10.150024] IP-Config: Complete:
[   10.153252]      device=eth0, hwaddr=00:04:4b:e6:61:84, ipaddr=192.168.1.120, mask=255.255.255.0, gw=192.168.1.254
[   10.163588]      host=192.168.1.120, domain=, nis-domain=(none)
[   10.169501]      bootserver=0.0.0.0, rootserver=192.168.1.88, rootpath=     nameserver0=192.168.1.254
[   10.179232] vdd-fan: disabling
[   10.182302] vdd-usb-vbus: disabling
[   10.185782] vdd-usb-vbus2: disabling
[   10.189364] avdd-io-edp-1v05: disabling
[   10.193289] ALSA device list:
[   10.196263]   #0: tegra-hda at 0x70038000 irq 83
[   10.200879]   #1: tegra-snd-t210ref-mobile-rt565x
[   10.208734] Freeing unused kernel memory: 8576K
Mounting /dev/nfs...
mount: mounting /dev/nfs on /mnt failed: Invalid argument
sh: can't access tty; job control turned off
```
So i was wondering why it tried to mount /dev/nfs if doc says that this is just  _a synonym to tell the kernel to use NFS instead of a real device._

After some research i found [this](https://unix.stackexchange.com/a/140183/239749) and finally understand that if my image has an initrd the mount process should be done within the init-script, in meta-tegra named **init-boot.sh** and the message that i'm seeing comes from this line
```
echo "Mounting ${rootdev}..."
mount -t "${fstype}" -o "${opt}" "${rootdev}" /mnt || exec sh
```
So i tried to left the kernel to do the automount and disable initrd, setting the following variable within my-recipe-image
```
INITRAMFS_IMAGE = ""
```
But after cleaning, rebuilding and variable confirmation i still building the initrd 
```
$ bitbake -f -c cleansstate my_recipe_image
$ bitbake -f -c clean my_recipe_image
$ bitbake -e my_recipe_image  grep ^INITRAMFS_IMAGE=
INITRAMFS_IMAGE=""
```
I know that my NFS-server works properly because i mount my rootfs manually via NFS from initrd
```
...

Mounting /dev/nfs...
mount: mounting /dev/nfs on /mnt failed: No such file or directory
sh: can't access tty; job control turned off

/ # mount -o port=2049,nolock,proto=tcp -t nfs 192.168.1.88:/nfsroot /mnt || exec sh
/ # ls -l /mnt/
drwxr-xr-x    2 0        0             4096 Jul  6  2020 bin
drwxr-xr-x    3 0        0             4096 Jul  6  2020 boot
drwxr-xr-x    2 0        0             4096 Jul  5  2020 dev
drwxr-xr-x   34 0        0             4096 Jul  6  2020 etc
drwxr-xr-x    3 0        0             4096 Jul  6  2020 home
drwxr-xr-x    9 0        0             4096 Jul  5  2020 lib
drwxr-xr-x    2 0        0             4096 Jul  5  2020 media
...

/ # mount --move /sys  /mnt/sys
/ # mount --move /proc /mnt/proc
/ # mount --move /dev  /mnt/dev
/ # exec switch_root /mnt /sbin/init
[ 4039.228659] systemd[1]: System time before build time, advancing clock.
[ 4039.240766] cgroup: cgroup2: unknown option "nsdelegate"
[ 4039.261384] systemd[1]: systemd 243+ running in system mode. (+PAM -AUDIT -SELINUX +IMA -APPARMOR +SMACK +SYSVINIT +UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELF)
[ 4039.283614] systemd[1]: Detected architecture arm64.

Welcome to Poky (Yocto Project Reference Distro) 3.0.1 (zeus)!

[ 4039.326952] systemd[1]: Set hostname to <jetson-nano-qspi-sd>.
[ 4039.335358] systemd[1]: Initializing machine ID from random generator.
[ 4039.477179] systemd[1]: /lib/systemd/system/dbus.socket:5: ListenStream= references a path below legacy directory /var/run/, updating /var/run/dbus/system_bus_socket ��→ /run/dbus/system.
[ 4039.522080] systemd[1]: /lib/systemd/system/rpcbind.socket:5: ListenStream= references a path below legacy directory /var/run/, updating /var/run/rpcbind.sock ��→ /run/rpcbind.sock; plea.
[ 4039.607877] systemd[1]: system-getty.slice: unit configures an IP firewall, but the local system does not support BPF/cgroup firewalling.
[ 4039.620239] systemd[1]: (This warning is only shown for the first unit using IP firewalling.)
[ 4039.631078] systemd[1]: Created slice system-getty.slice.
[  OK  ] Created slice system-getty.slice.
[ 4039.650579] systemd[1]: Created slice system-serial\x2dgetty.slice.
[  OK  ] Created slice system-serial\x2dgetty.slice.
[  OK  ] Created slice User and Session Slice.
[  OK  ] Started Dispatch Password ��…ts to Console Directory Watch.
[  OK  ] Started Forward Password R��…uests to Wall Directory Watch.
[  OK  ] Reached target Paths.
[  OK  ] Reached target Slices.
[  OK  ] Reached target Swap.
...
``` 
I don't think this is due because of missing CONFIG_NFS options from kernel, i looked in meta-tegra/recipes-kernel/linux/linux-tegra-4.9/tegra210/defconfig and it comes with everything it supposed to have to enable NFS-client also my build/.config looks fine.

So I'm pretty much stuck on this i'm not sure what else i can do to find out where the problem is... 
why does the Kernel is not doing the automount? probably because i'm building the initrd so why bitbake keeps building it if i already set INITRAMFS_IMAGE="" ?

Do i need to Modify **init-boot.sh** in such a way to make it work?

Any clue on this will be very helpful thank you so much!

madisongh, 2020-07-07:
If you want to disable the initramfs image, you should do that in your `local.conf` with
```
INITRAMFS_IMAGE_jetson-nano-qspi-sd = ""
```
since it needs to be known across multiple recipes.  It's been a while since I've built without one - the original reason for adding it was because the USB controller firmware needed to be loaded early on during boot, but that was some versions back and I'm not sure it's absolutely required any more.

Otherwise, yes, updating the init-boot.sh script to add any extra arguments to the mount command (or additional commands beforehand) that might be needed to successfully NFS-mount the rootfs would be a good approach.  I don't know off-hand what all would be needed, but [this script](https://git.openembedded.org/openembedded-core/tree/meta/recipes-core/initrdscripts/initramfs-framework/nfsrootfs) in the initramfs-framework scripts in OE-Core might be a good place to look for some tips.  At some point, it might make sense to move the tegra-minimal-initramfs recipe to be based off that framework so it's more general.

elPrac, 2020-07-11:
Hi @madisongh thanks for the reply!

I put the INITRAMFS_IMAGE variable as you said and it is working now, thank you very much, I'm still not familiar with the variables that are need to set from local.conf or a distro recipe. 

Also, i don't know if you can help me with the following; I tried to put the hole string to set bootargs in a U-boot environment variable from sourcecode and increase the Bootdelay time(currently 1sec), but i'm failing in both attempts... for the first one i tried to add my variable nfsargs within include/configs/p3450-porg.h
```c
#define BOARD_EXTRA_ENV_SETTINGS \
    "preboot=if test -e mmc " DEFAULT_MMC_DEV ":1 /u-boot-preboot.scr; then " \
        "load mmc " DEFAULT_MMC_DEV ":1 ${scriptaddr} /u-boot-preboot.scr; " \
        "source ${scriptaddr}; " \
    "fi\0" \
    "serverip=192.168.1.88\0" \
    "ipaddr=192.168.1.155\0" \
    "testrfspath=/nfsroot\0" \                                                                     
    "nfsargs=setenv bootargs root=/dev/nfs ip=${ipaddr}:${serverip}::255.255.255.0:off:eth0:off " \
        "nfsroot=${serverip}:${testrfspath},nfsvers=3,tcp rw\0" \

```
But from U-boot shell i only can see that serverip and ipaddr were the only ones set, if i print the hole environment with #printenv or #env print i can't see testrfspath neither nfsargs.

And in order to increase BOOTDELAY time i tried to modify configs/p3450-porg_defconfig with
```bash
CONFIG_BOOTDELAY=3
```
I added these changes as a recipes-bsp/u-boot/u-boot-tegra_2016.07.bbappend recipe
```bash
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI += " file://0001-Feat-add-env-vars-for-nfs.patch"

```

and from tmp/work/jetson_nano_qspi_sd-poky-linux/u-boot-tegra/2016.07+gitAUTOINC+565a18c42c-r0/build
i can see my changes reflected in 

```bash
./.config:233:CONFIG_BOOTDELAY=3
./u-boot.cfg:73:#define CONFIG_BOOTDELAY 3
./include/config/auto.conf:76:CONFIG_BOOTDELAY=3
./include/generated/autoconf.h:78:#define CONFIG_BOOTDELAY 3
```
and in...
```
./u-boot.cfg:272:#define BOARD_EXTRA_ENV_SETTINGS "preboot=if test -e mmc " DEFAULT_MMC_DEV ":1 /u-boot-preboot.scr; then " "load mmc " DEFAULT_MMC_DEV ":1 ${scriptaddr} /u-boot-preboot.scr; " "source ${scriptaddr}; " "fi\0"  "serverip=192.168.1.88\0" "ipaddr=192.168.1.155\0" "testrfspath=/nfsroot\0" "nfsargs=setenv bootargs root=/dev/nfs ip=${ipaddr}:${serverip}::255.255.255.0:off:eth0:off " "nfsroot=${serverip}:${testrfspath},nfsvers=3,tcp rw\0 "
```

So i think the build was successful but why i'm not seeing these env variables in U-boot shell?
Also, why the boot time delay is not changing at all? from shell i keep seeing...
```
Tegra210 (P3450-Porg) # printenv bootdelay 
bootdelay=1
```

Thank you very much for your time!

madisongh, 2020-07-11:
Could be saved environment being picked up... the compiled-in environment is only used if a stored one can't be found.  If you can interrupt U-Boot before it boots the kernel, use the `env default -a` command to reset the environment to the compiled-in default.

If that's not the problem, then maybe your updated u-boot isn't getting onto the SDcard?  Check to make sure your new copy is in `$BUILDDIR/tmp/deploy/images/jetson-nano-qspi-sd` first, then make sure it's getting to whatever your next update step is - tegraflash package, direct SDcard partition writing, BUP payload, etc.

madisongh, 2020-12-11:
No activity on this for several months, so I assume you got this worked out.

closed 2020-12-11 by madisongh