OE4T/meta-tegra #384: Booting with custom extlinux FDT on sd-card nano B01

acostach, 2020-07-07:
Hey Matt,

I have a question, does booting with a custom fdt in extlinux.conf work on your side?

For instance I'm setting the default one in extlinux.conf:
```
Retrieving file: /boot/tegra210-p3448-0000-p3449-0000-b00.dtb
214958 bytes read in 52 ms (3.9 MiB/s)
## Flattened Device Tree blob at 83000000
   Booting using the fdt blob at 0x83000000
ERROR: reserving fdt memory region failed (addr=0 size=0)
ERROR: reserving fdt memory region failed (addr=0 size=0)
   Using Device Tree in place at 0000000083000000, end 00000000830377ad
copying carveout for /host1x@50000000/dc@54200000...
copying carveout for /host1x@50000000/dc@54240000...
```

and get:

```
[    5.934932] Unable to handle kernel read from unreadable memory at virtual address 000001d8
[    5.943303] Mem abort info:
[    5.946102]   ESR = 0x96000005
[    5.949163]   Exception class = DABT (current EL), IL = 32 bits
[    5.955081]   SET = 0, FnV = 0
[    5.958139]   EA = 0, S1PTW = 0
[    5.961288] Data abort info:
[    5.964163]   ISV = 0, ISS = 0x00000005
[    5.968033]   CM = 0, WnR = 0
[    5.971009] user pgtable: 4k pages, 39-bit VAs, pgd = ffffffc0f7c3a000
[    5.977537] [00000000000001d8] *pgd=0000000000000000, *pud=0000000000000000
[    5.984524] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[    5.990090] Modules linked in:
[    5.993154] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.9.140-l4t-r32.4.2 #1
[    6.000193] Hardware name: NVIDIA Jetson Nano Developer Kit (DT)
[    6.006190] task: ffffffc0fa6f0000 task.stack: ffffffc0fa6bc000
[    6.012119] PC is at tegra_channel_populate_dev_info+0xb8/0x1a0
[    6.018031] LR is at tegra_channel_populate_dev_info+0xa8/0x1a0
[    6.023942] pc : [<ffffff8008af0170>] lr : [<ffffff8008af0160>] pstate: 40400045
[    6.031326] sp : ffffffc0fa6bf7c0
[    6.034635] x29: ffffffc0fa6bf7c0 x28: ffffffc0f7ceb1c0 
[    6.039958] x27: 0000000000000038 x26: 00000000009a2065 
[    6.045279] x25: ffffffc0f8b4c038 x24: ffffffc0f7ceb1e8 
[    6.050600] x23: 000000000000000b x22: ffffff800903de90 
[    6.055920] x21: ffffffc0f7ceb018 x20: ffffffc0fa6bf870 
[    6.061240] x19: ffffffc0f948c010 x18: 0000000000000001 
[    6.066561] x17: 0000000000000005 x16: 0000000000000000 
[    6.071882] x15: ffffffffffffffff x14: ffffff800a3f7e10 
[    6.077202] x13: ffffffc0fa6bf7b8 x12: ffffffc0fa6bf6bc 
[    6.082522] x11: 0000000000000028 x10: ffffffc0fa6bf670 
[    6.087843] x9 : 0000000000000002 x8 : 0000000000000002 
[    6.093163] x7 : ffffff8008f96850 x6 : 0000000000000090 
[    6.098483] x5 : 000000000000008d x4 : 0000000000000001 
[    6.103803] x3 : 0000000000000000 x2 : 000000000004098c 
[    6.109124] x1 : 0000000000000000 x0 : 0000000000000000 
[    6.114444] 
[    6.115934] Process swapper/0 (pid: 1, stack limit = 0xffffffc0fa6bc000)
[    6.122625] Call trace:
[    6.125072] [<ffffff8008af0170>] tegra_channel_populate_dev_info+0xb8/0x1a0
[    6.132027] [<ffffff8008af2528>] tegra_channel_init_subdevices+0x590/0x700
[    6.138893] [<ffffff8008af32d4>] tegra_vi_graph_notify_complete+0x30c/0x6d0
[    6.145851] [<ffffff8008ad66d4>] v4l2_async_test_notify+0x104/0x110
[    6.152111] [<ffffff8008ad6bc0>] v4l2_async_notifier_register+0x110/0x1b0
[    6.158893] [<ffffff8008af42e4>] tegra_vi_graph_init+0x29c/0x2d8
[    6.164893] [<ffffff8008aedd88>] tegra_vi_media_controller_init+0x1b8/0x1f8
[    6.171850] [<ffffff8008b0d31c>] vi_probe+0x3b4/0x530
[    6.176901] [<ffffff8008762dc8>] platform_drv_probe+0x60/0xc0
[    6.182641] [<ffffff8008760384>] driver_probe_device+0xd4/0x3f0
[    6.188552] [<ffffff8008760798>] __driver_attach+0xf8/0x120
[    6.194116] [<ffffff800875dd48>] bus_for_each_dev+0x90/0xe0
[    6.199680] [<ffffff800875fb98>] driver_attach+0x30/0x40
[    6.204983] [<ffffff800875f580>] bus_add_driver+0x1a8/0x288
[    6.210548] [<ffffff8008761698>] driver_register+0x68/0x100
[    6.216113] [<ffffff8008762d0c>] __platform_driver_register+0x5c/0x68
[    6.222552] [<ffffff8009907978>] vi_init+0x2c/0x38
[    6.227340] [<ffffff8008083ac0>] do_one_initcall+0x58/0x160
[    6.232909] [<ffffff80098c0d64>] kernel_init_freeable+0x198/0x22c
[    6.238996] [<ffffff8008f2a094>] kernel_init+0x18/0xfc
[    6.244125] [<ffffff80080838a0>] ret_from_fork+0x10/0x30
```

I've overwritten only u-boot with one from a default Nvidia Ubuntu sd-card image and can load the same dtb and run without any kernel crash. 
```
U-Boot 2016.07-g45dfa3dff4 (Apr 08 2020 - 18:04:08 -0700)
...
Retrieving file: /boot/tegra210-p3448-0000-p3449-0000-b00.dtb
214958 bytes read in 365 ms (574.2 KiB/s)
## Flattened Device Tree blob at 80000000
   Booting using the fdt blob at 0x80000000
   reserving fdt memory region: addr=80000000 size=20000
   Using Device Tree in place at 0000000080000000, end 00000000800377ad
ERROR: DT property /psci/nvidia,system-lp0-disable missing in source; can't copy

at /dvs/git/dirty/git-master_linux/3rdparty/u-boot/arch/arm/mach-tegra/dt-edit.c:176/fdt_iter_copy_prop()

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.140-l4t-r32.4.2 (oe-user@oe-host) (gcc version 8.3.0 (GCC) ) #1 SMP PREEMPT Sun Jul 5 15:35:56 UTC 2020

```
Doesn't really seem to be the difference of load address, but more like uboot 2016.07 is modifying some nodes that 2020.04-00002-g914e902b5d-dirty isn't?

madisongh, 2020-07-07:
Yeah, looks like I missed some patches that went into NVIDIA's fork of the 2016.07 sources that they didn't upstream. I've ported the missing patches over that should take care of it. Once the change goes in, give it a try and let me know.

closed 2020-07-07 by madisongh

acostach, 2020-07-08:
Thank you very much Matt, that uboot patch solves the problem on the Nano.

PR #385 (2020-07-07): u-boot-tegra: bump SRCREV
to pick up some additional patches ported from
NVIDIA's u-boot that haven't made it upstream.

Fixes #384