OE4T/meta-tegra #386: [Xavier] Warrior backporting meta: use bundled initramfs instead of initrd on tegra186/194 casue cboot not build

danielwangksu, 2020-07-07:
Hi @madisongh, I'm trying to back port your commit https://github.com/madisongh/meta-tegra/commit/af4492634b6e3c0657973cf20418c95a7d8fe0a4 to warrior branch, so that I can use the bundled initramfs in my Xavier board. 

I have a customized cboot recipe in another layer. and set  `PREFERRED_PROVIDER_virtual/bootloader = "cboot"`,
`PREFERRED_PROVIDER_cboot = "cboot"` in my local.conf. Before the change Yocto build cboot and everything is fine. After made those changes, cboot did build. When I try to flash the board with `doflash.sh` I got error "cannot find cboot.bin". Do you have any idea why this happens? I can add more details.

kekiefer, 2020-07-07:
Is your cboot build not happening and therefore not placing the cboot binary in the `DEPLOY_DIR_IMAGE` directory, or is `image_types_tegra.bbclass` failing to copy the binary from there into the tegraflash package?

danielwangksu, 2020-07-07:
Hi @kekiefer cboot build not happening. I suppose to create a folder in `build/tmp/work/jetson-xavier-poky-linux/cboot` Without the change I saw cboot folder there. After the change, cboot did not even show up.

madisongh, 2020-07-07:
Sounds like you're a missing dependency somewhere, then. The tegraflash package creation needs `cboot:do_deploy`. Looks like I added that as part of 8e69242f9ec8ebbe21f230597be4ec10d75b23a2, although I'm not sure you'd want to back-port that entire change. You'd at least need to add `cboot:do_deploy` to `do_image_tegraflash[depends]` for tegra186 and tegra194 platforms... I think that will be enough.

It was probably working by accident before, due to the dependency chain happening to include cboot:do_deploy for something else getting built (a BUP payload, for instance) before the tegraflash image.

danielwangksu, 2020-07-07:
@madisongh You are absolute right! Thank you so much!!! I will do more test on on Xavier and TX2i and submit a PR.

madisongh, 2020-12-11:
Looks like this is resolved, so closing

closed 2020-12-11 by madisongh