OE4T/meta-tegra #388: [Jetson TX2] Error: Detected Tegra bootloader upgrade failure

ichergui, 2020-07-11:
Hi Matt, 

I'm testing the test-distro build for the zeus-mender-l4t-r32.3.1 branch on Nvidia Jetson TX2. When running the mender commit I got the following error:

```
root@jetson-tx2:~# mender commit 
INFO[0000] Loaded configuration file: /var/lib/mender/mender.conf  module=config
INFO[0000] Loaded configuration file: /etc/mender/mender.conf  module=config
INFO[0000] Mender running on partition: /dev/mmcblk0p1   module=cli
INFO[0000] Update Module path "/usr/share/mender/modules/v3" could not be opened (open /usr/share/mender/modules/v3: no such file or directory). Update modules will not be available  ms
Committing Artifact...
ERRO[0000] stderr collected while running script /var/lib/mender/scripts/ArtifactCommit_Enter_80_bl-check-update [Detected Tegra bootloader upgrade failure
]  module=executor
ERRO[0000] ArtifactCommit_Enter script failed: statescript: error executing 'ArtifactCommit_Enter_80_bl-check-update': 1 : exit status 1  module=standalone
Rolling back Artifact...
INFO[0000] setting partition for rollback: 33            module=dual_rootfs_device
ERRO[0001] statescript: error executing 'ArtifactCommit_Enter_80_bl-check-update': 1 : exit status 1  module=main
root@jetson-tx2:~# 
```

I confirm that the mender install was successfully applied, here is the traces:

```
................................  97% 110592 KiB
................................  98% 111616 KiB
................................  99% 112640 KiB
...................INFO[0088] 0 bytes remaining to be written               module=limited_writer
INFO[0088] The optimized block-device writer wrote a total of 13622 frames, where 9 frames did need to be rewritten  module=block_device
INFO[0090] Wrote 14282760192/14282760192 bytes to the inactive partition  module=dual_rootfs_device
              100% 113271 KiB
INFO[0090] Enabling partition with new image installed to be a boot candidate: 1  module=dual_rootfs_device
[  354.561074] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
Use -commit to update, or -rollback to roll back the update.
At least one payload requested a reboot of the device it updated.
root@jetson-tx2:~# 
```

Do you have any idea about this issue ?
I think that the U-Boot variable **mender_boot_part** was not updated for unknown reason. 
Could we add more checks in **ArtifactInstall_Leave_80_bl-update** after applying bootloader update ?

Best regards,
Ilies.

dwalkes, 2020-07-11:
Hi @ichergui 

```
ERRO[0000] stderr collected while running script /var/lib/mender/scripts/ArtifactCommit_Enter_80_bl-check-update [Detected Tegra bootloader upgrade failure
]  module=executor
```
Did you happen to check the [deployment log](https://docs.mender.io/hosted/troubleshooting/mender-client#deployment-log-files) to see if that has more information about the failure?  I think you can also just run the `/var/lib/mender/scripts/ArtifactCommit_Enter_80_bl-check-update` script ( [redundant-boot-commit-check-script-uboot](https://github.com/madisongh/meta-mender-community/blob/e1226d046b2db44ce1bed5dcbb719e8cb3c7d572/meta-mender-tegra/recipes-mender/tegra-state-scripts/files/redundant-boot-commit-check-script-uboot) ) from the command line to see exactly what is failing where.

> I think that the U-Boot variable mender_boot_part was not updated for unknown reason.

Does `fw_printenv` and `fw_setenv` work from the command line for updating bootloader variables?

madisongh, 2020-07-11:
You did reboot after applying the update and before running `mender -commit`, right?  A common cause of this that I've seen is people assuming (incorrectly) they can just commit immediately after a manual update, without rebooting.  

The point of the Commit-Enter check script is to ensure that Mender's U-Boot partition setting and the NVIDIA bootloader's A/B slot setting match.  This can only be done after reboot, pre-commit, as we need to find out if the settings and bootloader updates that were made during the update actually worked.  If you've updated from A->B, but after reboot U-Boot says "B" but NVIDIA's bootloader says "A" (or vice-versa), it indicates that something went wrong.

ichergui, 2020-07-11:
> Hi @ichergui
>
> ```
> ERRO[0000] stderr collected while running script /var/lib/mender/scripts/ArtifactCommit_Enter_80_bl-check-update [Detected Tegra bootloader upgrade failure
> ]  module=executor
> ``` [...]

Thank you for your response.
I will check the deployment log.
Yes, I confirm the **fw_printenv** and **fw_setenv** works fine for me.

ichergui, 2020-07-11:
> bootloader's

Thanks for your response.
I confirm that I reboot the device before the commit.

ichergui, 2020-07-11:
Hi @dwalkes,

I just checked the deployment log, I can't file it in **/var/lib/mender/** 

Here is what I have in that directory: 
```
root@jetson-tx2:~# ls -la /var/lib/mender/
drwxr-xr-x    3 root     root          1024 Jul 11 18:19 .
drwxr-xr-x    7 root     root          1024 Jun 30 17:06 ..
-r--r--r--    1 root     root            23 Jun 30 17:05 device_type
-rw-------    1 root     root          2455 Jun 30 17:45 mender-agent.pem
-rw-------    1 root     root         32768 Jul 11 18:20 mender-store
-rw-------    1 root     root          8256 Jul 11 18:30 mender-store-lock
-rw-r--r--    1 root     root            77 Jun 30 17:05 mender.conf
drwxr-xr-x    2 root     root          1024 Jul 11 18:19 scripts
root@jetson-tx2:~# 
```

Do you have an idea why I can't find the deployment log file ?

dwalkes, 2020-07-11:
> Do you have an idea why I can't find the deployment log file ?

Are you `mender -install` ing or running a deployment from a mender server?  I think you might only get deployment logs if you deploy from a mender server now that I think of it.

Did you try running the script directly?  I'm guessing it will fail the same way it's failing on commit.

ichergui, 2020-07-11:
Hi @dwalkes, 

I'm using mender CLI for the installation and the commit, I confirm that I reboot the device after **mender -install**.

I confirm the usage of the mender script `/data/mender/scripts/ArtifactCommit_Enter_80_bl-check-update` after rebooting the device failed.

Here is the traces:

```
root@jetson-tx2:~# diff -s /var/lib/mender/scripts/ArtifactCommit_Enter_80_bl-check-update /data/mender/scripts/ArtifactCommit_Enter_80_bl-check-update 
Files /var/lib/mender/scripts/ArtifactCommit_Enter_80_bl-check-update and /data/mender/scripts/ArtifactCommit_Enter_80_bl-check-update are identical
root@jetson-tx2:~# 
root@jetson-tx2:~# bash -x /data/mender/scripts/ArtifactCommit_Enter_80_bl-check-update 
+ LABELCHARS=AB
+ echo 'Verifying Tegra bootloader update'
Verifying Tegra bootloader update
++ nvbootctrl get-current-slot
+ curslot=0
+ cfglbl='"RootfsPartA"'
++ grep -h '"RootfsPartA":' /etc/mender/mender.conf /var/lib/mender/mender.conf
++ cut -d: -f2
++ tr -d '" '
++ cut -d, -f1
+ devnam=/dev/mmcblk0p1
+ '[' -z /dev/mmcblk0p1 ']'
+ '[' 0p1 '!=' /dev/mmcblk0p1 ']'
++ fw_printenv -n mender_boot_part
+ bootpart=33
+ '[' -z 33 ']'
++ expr 1 + 0
+ devnampart=1
+ '[' -z 1 ']'
+ '[' 33 -ne 1 ']'
+ echo 'Detected Tegra bootloader upgrade failure'
Detected Tegra bootloader upgrade failure
+ exit 1
root@jetson-tx2:~# 
```

dwalkes, 2020-07-11:
So it looks like there's a problem [here](https://github.com/mendersoftware/meta-mender-community/blob/zeus-test/meta-mender-tegra/recipes-mender/tegra-state-scripts/files/redundant-boot-commit-check-script-uboot#L30)
```+ '[' 33 -ne 1 ']```
 in that nvbootctrl says you are on slot 0 which should be mmcblk0p1 and your uboot variable says you are in mmcblk0p33.  So most likely something happened during the [install step](https://github.com/mendersoftware/meta-mender-community/blob/zeus-test/meta-mender-tegra/recipes-mender/tegra-state-scripts/files/redundant-boot-install-script-uboot) which means nv_update_engine didn't actually complete an update it attempted there.
Is your rootfs actually on partition 33 or is it on partition 1?
What do you see from `nvbootctrl dump-slots-info` and `nvbootctrl get-number-slots`?  See [this section](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootloader_update_agx_tx2.html) of the Linux Dev guide for relevant information about these tools.
As a troubleshooting step you can also try running the [install script](https://github.com/mendersoftware/meta-mender-community/blob/zeus-test/meta-mender-tegra/recipes-mender/tegra-state-scripts/files/redundant-boot-install-script-uboot) from the command line and see if any of those commands are printing error messages.  I see we aren't checking error result codes on [nv_update_engine](https://github.com/mendersoftware/meta-mender-community/blob/zeus-test/meta-mender-tegra/recipes-mender/tegra-state-scripts/files/redundant-boot-install-script-uboot#L23) today for instance, so if that's failing it might be getting swallowed in the install script.

ichergui, 2020-07-11:
Yes, I confirm that.
The rootfs is on the partition 33.

I have reboot the device, this means that there was a rollback.

```
root@jetson-tx2:~# nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 15,             suffix: _a,             retry_count: 5,             boot_successful: 0
slot: 1,             priority: 14,             suffix: _b,             retry_count: 7,             boot_successful: 1
root@jetson-tx2:~# nvbootctrl get-number-slots
2
root@jetson-tx2:~# 
```
I will retry by using the mender script `/data/mender/scripts/ArtifactInstall_Leave_80_bl-update`

I will keep you informed.

dwalkes, 2020-07-11:
You might need/want to connect the [serial console](https://www.jetsonhacks.com/2017/03/24/serial-console-nvidia-jetson-tx2/) for boot related update failure logs if you haven't already.  My guess is `nv_update_engine --enable-ab` during install worked fine since it looks like your slots are setup correctly.

I'm confused about how your previous log shows that your current slot is 0 from `nvbootctrl get-current-slot` but then `nvbootctrl dump-slots-info` has boot_succcessful: 0 in your dump-slots-info output for slot 0.  Perhaps this is related.

ichergui, 2020-07-13:
> So it looks like there's a problem [...]

Hi @dwalkes @madisongh 

Here is the [patch](https://github.com/madisongh/meta-tegra/files/4913329/0001-fix-check-sucess-ab-update.patch.log) for checking the return code when trying to enable A/B update.

I can not push a PR for that in mender community meta layer ([meta-mender-community](https://github.com/mendersoftware/meta-mender-community/tree/zeus-test)), I don't have permissions .

Please let me know if I can help.

Best regards,

Ilies

dwalkes, 2020-07-13:
Thanks @ichergui 

Were you able to figure out your boot update failure logs?

> I can not push a PR for that in mender community meta layer (meta-mender-community), I don't have permissions .

You could fork https://github.com/BoulderAI/meta-mender-community and add a PR from there if you like.  @kekiefer is using this one with the mender work in progress changes for L4T 32.4.2 in https://github.com/kekiefer/test-distro/blob/master%2Bwip-mender so if you want to test with the latest working changes there that would be appreciated as well.

ichergui, 2020-07-14:
I'm still continuing my investigation.
This issue not happen every time, it difficult to reproduce.
I will let you know.

ichergui, 2020-07-14:
@madisongh 

I have noticed that **nv_update_verifier.service** is masked, in your [commit](https://github.com/madisongh/meta-mender-community/commit/5854ee8bdf286a0e66199af0036dc647eeb5e85f) - it's mentioned that this is masked because the Mender state script is handling the verification. I can see that nv_update_engine --verify is not being used inside the Mender state script. 

Can you confirm why this is the case?

madisongh, 2020-07-14:
It's because `nv_update_engine --verify` does more than just verify that the current boot worked; it also overwrites the bootloader partitions for the other boot slot.  That's OK if you have just one rootfs and are just updating the bootloaders, but it's not really desirable if you're treating the entire bootloaders/kernel/rootfs collection as a single unit and need to fully roll back in the event of an update failure.

ichergui, 2020-07-14:
> It's because `nv_update_engine --verify` does more than just verify that the current boot worked; it also overwrites the bootloader partitions for the other boot slot. That's OK if you have just one [...]

Thanks @madisongh for your response.

ichergui, 2020-07-14:
@madisongh 

> The point of the Commit-Enter check script is to ensure that Mender's U-Boot partition setting and the NVIDIA bootloader's A/B slot setting match. This can only be done after reboot, pre-commit, as [...]

Do you have an idea why I have the mismatch between mender boot partition and NVIDIA slot ? 
I confirm that the **mender -install** was done successfully (return code 0) and **nv_update_engine --install** command in mender state script was also done successfully.   

should I use command **nvbootctrl dump-slots-info** before and after A/B system updates for troubleshooting ?

madisongh, 2020-07-14:
It's hard to tell after the fact how the two got out of sync, if everything has been successful; I've only seen it happen when manually running installs and getting the sequence wrong.

What the scripts aren't doing, and they should, is to recover from the situation... once they're out of sync, you're stuck unless you manually force one or the other bootloaders to use the correct slot (e.g., with `nvbootctrl set-active-boot-slot` for the NV bootloader).  That should probably be fixed so that when the CommitEnter script detects the mismatch, the rollback operation resets things so the two bootloaders are synchronized again. I'm just not sure off the top of my head which of the two should be considered correct in that case.  I don't like the two competing sources of truth about the boot slots and having to coordinate them like this.

For my production project, I did away with U-Boot completely, which eliminated the coordination problem.  But there's probably a better way to do this when U-Boot is still in the mix.  One idea:
* Replace the Mender U-Boot patches for partition selection variables with a patch that maps the slot selection passed in from cboot to the right partition, and
* Use the `u-boot-fw-utils-fake` scripts that are used in the cboot-only case so that Mender is directly interfacing with the NV boot slot information.
I think that could be made to work, and would be more robust than the current setup.

dwalkes, 2020-07-14:
> Do you have an idea why I have the mismatch between mender boot partition and NVIDIA slot ?

I feel like every time I've seen this I've seen some type of error message on boot.  Were you able to find anything interesting with [the serial console](https://github.com/madisongh/meta-tegra/issues/388#issuecomment-657134439)?

> I don't like the two competing sources of truth about the boot slots and having to coordinate them like this.

Agreed, this should be improved.

> For my production project, I did away with U-Boot completely, which eliminated the coordination problem

I like that solution, but for those of us using nano we'll still need to have a u-boot solution I suppose.

> Use the u-boot-fw-utils-fake scripts that are used in the cboot-only case so that Mender is directly interfacing with the NV boot slot information.

I like this solution the best, since it leverages most of the implementation with cboot and this could be the recommended supported way forward for new platforms which support cboot.

ichergui, 2020-07-14:
Thanks @madisongh and @dwalkes very much for your time!

I have logs that I can share with you, please see the traces below:

```
### A/B system update

### Before mender install
jetson-tx2:~$ sudo nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 14,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$

### After mender commit
jetson-tx2:~$ sudo nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 15,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 14,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 

---> reboot the system

jetson-tx2:~$ sudo nvbootctrl dump-slots-info
Password: 
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 14,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 14,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 

---> reboot the system

jetson-tx2:~$ sudo nvbootctrl dump-slots-info
Password: 
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 14,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 
```
After rebooting the system several times, the slot priority change for unknown reason.
Then, I have rebooted the system one more time, I got a strange result, please see below:

```
jetson-tx2:~$ sudo nvbootctrl dump-slots-info
Password: 
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 13,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 
jetson-tx2:~$ 
jetson-tx2:~$ sudo nvbootctrl get-current-slot
1
jetson-tx2:~$
```
Normally, I should see slot 0 instead of 1, because I didn't apply a system update `mender -install`  and I didn't use the `nv_update_engine` or other command.
I checked the mender boot part via the command `fw_printenv`
```
jetson-tx2:~$ sudo fw_printenv -n mender_boot_part
Password: 
1
jetson-tx2:~$ 
```

There is a mismatch here.

madisongh, 2020-07-14:
> those of us using nano we'll still need to have a u-boot solution I suppose.

Nano/TX1 are different in that they don't support A/B redundancy in the NVIDIA boot ROM/bootloaders. Sticking with the normal Mender/U-Boot integration should be OK for those platforms.

madisongh, 2020-07-15:
@ichergui Am I understanding the sequence correctly? I've added (??) to the points where I'm not sure, based on the info provided:
1. You started out in a known good state (NV slot=1, MENDER_BOOT_PART=33, and rootfs actually mounted on partition 33). (??)
2. You did a `mender install`.
3. You did a `mender commit` without rebooting first (??)
4. You rebooted multiple times.
End result: NV slot and MENDER_BOOT_PART desynchronized.

If so, the step 3 was the problem.

ichergui, 2020-07-15:
> @ichergui Am I understanding the sequence correctly? I've added (??) to the points where I'm not sure, based on the info provided:
> 
> 1. You started out in a known good state (NV slot=1, MENDER_BOOT_PART=33, and rootfs actually mounted on partition 33). (??)

Yes

```
jetson-tx2:~$ sudo nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 14,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$
```

> 2. You did a `mender install`.

Yes then reboot the device

> 3. You did a `mender commit` without rebooting first (??)

Yes
```
jetson-tx2:~$ sudo nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 15,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 14,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 
```

> 4. You rebooted multiple times.
>    End result: NV slot and MENDER_BOOT_PART desynchronized.
> 

```
---> reboot the system

jetson-tx2:~$ sudo nvbootctrl dump-slots-info
Password: 
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 14,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 14,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 

---> reboot the system

jetson-tx2:~$ sudo nvbootctrl dump-slots-info
Password: 
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 14,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 

---> reboot the system

jetson-tx2:~$ sudo nvbootctrl dump-slots-info
Password: 
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 13,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1
jetson-tx2:~$ 
jetson-tx2:~$ 
jetson-tx2:~$ sudo nvbootctrl get-current-slot
1
jetson-tx2:~$
jetson-tx2:~$ sudo fw_printenv -n mender_boot_part
Password: 
1
jetson-tx2:~$ 
```

> If so, the step 3 was the problem.
Agree, I think we should not have two slots active with `boot_successful equal to 1`

dwalkes, 2020-07-15:
So the solution here is just to avoid `mender -commit` until after reboot.  However, to avoid this confusion going forward, we should make any Mender u-boot support for platforms which support A/B redundancy (any but Nano since we don't support TX1 on Mender today) use u-boot-fw-utils-fake scripts logic shared with the cboot build and let nvidia completely handle slot switching and corresponding rootfs selection.

Do I have this right?

dwalkes, 2020-07-15:
Actually nevermind, I think I misunderstood the comment above.  The reboot did happen between `mender -install` and `mender -commit`, so this issue still appears to be something else.

I'd still suggest capturing the serial boot logs for the reboot after step 2 above in case there's a clue there.

ichergui, 2020-07-16:
> Thanks @ichergui
>
> Were you able to figure out your boot update failure logs?
>
>
> You could fork https://github.com/BoulderAI/meta-mender-community and add a PR from there if you like. @kekiefer is [...]

Hi @dwalkes @kekiefer 

I have sent a PR [here](https://github.com/BoulderAI/meta-mender-community/pulls) 

Best regards
Ilies

ichergui, 2020-09-01:
Hi @madisongh 

I will close this issue.
I have managed to fix it.

closed 2020-09-01 by ichergui

dwalkes, 2020-11-10:
This relates to https://github.com/BoulderAI/meta-mender-community/pull/1