OE4T/meta-tegra #389: [Jetson TX2] Mender option --log-file is not working

ichergui, 2020-07-11:
Hi Matt,

I'm testing the test-distro build for the zeus-mender-l4t-r32.3.1 branch on Nvidia Jetson TX2, mender version **eef7471**.

I tried to use the option **--log-file** but is not working for me.

Here is the version of mender CLI I used.

```
root@jetson-tx2:~# mender --help
NAME:
   mender - manage and start the Mender client.

USAGE:
   [global options] command [command options] [arguments...]

VERSION:
   eef7471  runtime: go1.12.9

DESCRIPTION:
   mender integrates both the mender daemon and commands for manually performing tasks performed by the daemon (see list of COMMANDS below).
```

Could you please confirm that ?
BTW, Can I help on that ?

Best regards, 
Ilies

madisongh, 2020-07-11:
Can you be a bit more specific about what "not working" means?  Is the option being rejected, the log file not getting created, or is empty, or.... ?  Is this something you've added to the systemd service file to direct the daemon's output to a different location?

ichergui, 2020-07-11:
Hi @madisongh 
Sorry, I was not specific in my comment.

Actually, when I perform an OS update via mender CLI, I use the option **--log-file** to print logs in a file.
Here is the command I used:

```
root@jetson-tx2:~# mender install /mnt/ssd/core-image-base-jetson-tx2-20200710172258.mender --log-level debug --log-file /tmp/my_mender_install.log                                      
INFO[0000] Loaded configuration file: /var/lib/mender/mender.conf  module=config
INFO[0000] Loaded configuration file: /etc/mender/mender.conf  module=config
INFO[0000] Mender running on partition: /dev/mmcblk0p1   module=cli
```

The file was not created, that's why I said "not working" in my previous comment.

I can investigate here, if want.

I wish help you guys.

dwalkes, 2020-07-11:
Maybe try one `-` instead of two `--`, same as you do with `-install`?

ichergui, 2020-07-11:
I just try it, unfortunately, the log file is not created.

dwalkes, 2020-07-11:
Might be a better question for https://hub.mender.io/, I doubt this is tegra specific.

madisongh, 2020-07-13:
I think I noticed this myself some time ago... IIRC, `--log-file` is listed as a global option, but it really only applies to running the agent daemon, and has no effect on locally-run operations.  But as @dwalkes mentioned, it's a question better directed to the Mender folks.

madisongh, 2020-07-13:
Well, I just tried it myself, with version 2.2 of mender-client, and I do get some logging to the file when I use `--log-file` with `mender install`. Some output still goes to stdio, though.

ichergui, 2020-07-13:
Thanks @madisongh 

I already create a topic in [mender hub](https://hub.mender.io/) [here](https://hub.mender.io/t/mender-option-log-file-is-not-working/2191)

ichergui, 2020-07-15:
Hi @dwalkes @madisongh 

I got a response concerning the log file when when using the `mender -install`   and `mender -commit`.

The log file was created and contains the logs.

Here the commands I used:

- Mender install 

`root@jetson-tx2:~#  sudo mender -log-level debug -log-file /mnt/ssd/my_mender_install.log -install /mnt/ssd/core-image-base-jetson-tx2-20200713105842.mender`

- Mender commit

`root@jetson-tx2:~# mender -log-level debug -log-file /mnt/ssd/my_mender_commit.log -commit  `

closed 2020-07-15 by ichergui