OE4T/meta-tegra #390: Jetson Nano NMI watchdog: BUG: soft lockup - CPU#0 stuck for 22s! [kworker/0:4:4202]

przemyslaw-chwiala, 2020-07-15:
Hey,
I'm working with yocto zeus and meta-tegra to build image for jetson-nano. Today I had issue with booting nano. It didn't boot at all and UART log told me just 
```
Starting kernel ...

<hit enter to activate fiq debugger>
```
and then boots again. 

I tried this SD card in another nano and it works fine. I put new sd card in this 'broken' nano and I also boots but after ~80s ow working it reboots with error
```
[   80.221608] NMI watchdog: BUG: soft lockup - CPU#0 stuck for 22s! [kworker/0:4:4202]
```
I'm attaching full booting log below. 

I'm wondering what could gone wrong and how to fix this nano (if it is possible)

Do you have any idea what could cause it?

[original_sd_card.log](https://github.com/madisongh/meta-tegra/files/4925406/original_sd_card.log)
[good_sd_card_but_rebooting.log](https://github.com/madisongh/meta-tegra/files/4925408/good_sd_card_but_rebooting.log)

Thank you

mozcelikors, 2020-07-15:
I have the same issue trying to use enc28j60 driver with jetson nano/yocto distro.
Thread is here: https://forums.developer.nvidia.com/t/enc28j60-ethernet-module-interfacing-with-jetson-nano/129254
No solutions are available yet.
I suspect this is caused by faulty interrupt config, in my case. The problem happens on yocto distro for me, so if there is any suggestions from this thread too, Id very much want to try them out.
Thanks.

przemyslaw-chwiala, 2020-07-15:
@mozcelikors Will DT update fix my nano?

madisongh, 2020-07-15:
@mozcelikors This isn't exactly the same situation as yours.  Based on your forum post, I'm not sure the falling-edge irq flag is getting propagated to the irq subsystem, so it could be expecting level trigger, which could cause the interrupt storm... I don't remember off-hand, but a call to of_irq_get() might be needed in the driver.

@przemyslaw-chwiala From the log it look like you have some kind of NVME device connected via PCI.  Does the Nano boot OK without that device connected? What carrier board are you using?

madisongh, 2020-12-11:
No activity on this for several months, so I assume you got this figured out.  Closing.

closed 2020-12-11 by madisongh