OE4T/meta-tegra #394: Ethernet tuning to achieve Gigabyte speed: GigE Cameras

anishmonachan7, 2020-07-21:
Hello,
I am using GigE cameras with my jetson Tx2. The camera is connected via the eth port.
I have a problem that the capture fails at high fps. This is because the RX Ethernet buffers are not able to handle this incoming traffic on time. The following network parameters are set at maximum. It has improved the capacity but not yet to the required level.
 ```
   net.core.netdev_max_backlog
   net.core.rmem_max
   net.core.rmem_default

```

anishmonachan7, 2020-07-21:
> 

> 

-

madisongh, 2020-07-21:
You'll likely get better results by posting this on the NVIDIA Developer Forum. Some info about what the incoming network traffic is, how much there is, and if it's, for example, video that's being encoded/decoded in hardware, would probably help with getting some pointers on where to look.

madisongh, 2020-11-27:
Looks like this is being discussed here: https://forums.developer.nvidia.com/t/tx2-eth-port-can-t-handle-high-incoming-traffic-gige-stream-issue/142979

closed 2020-11-27 by madisongh