OE4T/meta-tegra #395: [Question] Write-protect TX2i eMMC

matanliber, 2020-07-21:
I'm trying to enable the eMMC user partition write protect using  `mmc writeprotect user set`. I saw the the `mmc_utils` code requires that `ext_csd[EXT_CSD_ERASE_GROUP_DEF] = 1` for this command, yet when using `mmc extcsd read /dev/mmcblk0` I get that this value in the eMMC is set to 0.
What is the right way to use enabled user partition write protection?

madisongh, 2020-07-21:
From [this NVIDA dev form post](https://forums.developer.nvidia.com/t/use-emmc-hardware-write-protect-in-jetson-tx2i/139415) it's supposed to be possible, but IIRC it's a bit more involved than just using that command - you have to set the device up in a way that enables that functionality (enhanced user area, maybe?).  Looking the JEDEC spec for eMMC might help in figuring out exactly what needs to be set up.

It's not something I've ever done, and I'm not sure what impact making such changes will have on the functionality of the bootloaders or the Linux kernel, so be careful.

matanliber, 2020-07-21:
Thank you for your answer, do you know if it is possible to add another user partition to the eMMC (maybe through the `flash.xml`?)

madisongh, 2020-07-21:
Depends on what kind of partition you're referring to.  There are "hard" partitions in an eMMC, which are handled by the controller in the eMMC itself, while the `flash.xml` file controls the layout of the "soft" partitions that are understood by the bootloaders and operating system.

Soft partitions are easy enough to create and work with - just start with the stock XML file that NVIDIA provides in the BSP and make changes as needed.  To use it in a bitbake build with meta-tegra, there are a few steps involved.  There's an example in the test-distro repository for a custom MACHINE I have set up there which:
1. Adds a [recipe to install the custom flash layout file](https://github.com/madisongh/test-distro/blob/master/layers/meta-testdistro/recipes-bsp/tegra-binaries/custom-flash-layout_1.0.bb)
2. Adds a [bbappend to have `tegra-bootfiles` DEPEND on the custom flash layout](https://github.com/madisongh/test-distro/blob/master/layers/meta-testdistro/recipes-bsp/tegra-binaries/tegra-bootfiles_32.4.3.bbappend)
3. Sets the PARTITION_LAYOUT_TEMPLATE variable to point to the custom layout file [in the machine configuration file](https://github.com/madisongh/test-distro/blob/master/layers/meta-testdistro/conf/machine/jetson-tx2-cboot.conf).

matanliber, 2020-07-21:
Oh, sorry for not making myself clear :), I meant what you call "hard" partitions, I am interested in creating another `sdmmc_user` partition

madisongh, 2020-07-21:
It's probably doable with mmc-utils, but again it's not something I've done.  And I'd again caution that messing with the hard partitioning could have impact on the bootloader and possibly the flashing tools, which may or may not be able to handle it... this is another question you might want to ask about it on the NVIDIA dev forum.

closed 2020-07-21 by matanliber