OE4T/meta-tegra #396: Add pre-switchroot script to init-boot.sh

bcis93, 2020-07-22:
Hi,

I ran into a situation where I needed to mount a `/data` partition early in the boot process, so I tried modifying [init-boot.sh](https://github.com/madisongh/meta-tegra/blob/master/recipes-core/initrdscripts/tegra-minimal-init/init-boot.sh) to mount `/data` there. I added the extra commands as shown below, and this worked well for me.
```
...
echo "Mounting ${rootdev}..."
mount -t ext4 -o "$opt" "${rootdev}" /mnt || exec sh

#These 2 lines are new
mount /data
mount --move /data /mnt/data

echo "Switching to rootfs on ${rootdev}..."
...
```

I could easily create a .bbappend that replaces init-boot.sh with my customized init-boot.sh, but I don't necessarily want to do that (in case updates are made to init-boot.sh in the future, for example). I thought perhaps a better and more generic approach would be to add my changes into a separate script, and then conditionally run that script from within init-boot.sh (similar to what was done with the `platform-preboot` script). I committed a simple change [here](https://github.com/bcis93/meta-tegra/commit/af732259b55490b3429bcfecf43ea279c305eaa5), which will run a `platform-pre-switchroot` script (if it exists) right before switching the rootfs.

What are your thoughts on this? Would you be open to a PR to add this change here? Or do you have any other suggestions on how to run some extra commands at this point without completely overwriting init-boot.sh?

madisongh, 2020-07-22:
Looks familiar... in my test distro I have something very similar; a replacement init-boot.sh that, among other things, calls out to a platform-pre-switchroot script if present.  I'm fine with making that hook more generally available, so go ahead and submit a PR with your change.

bcis93, 2020-07-22:
Ah, so it did come from your test distro! We have a custom init-boot.sh that we've been using, and a coworker mentioned it was from your test distro, but I couldn't find it anywhere. But that sounds good, I'll submit the PR shortly. Thanks!

bcis93, 2020-07-28:
PR merged and everything works great. Thanks for your help!

closed 2020-07-28 by bcis93