OE4T/meta-tegra #399: No recipes available for: vulkan-loader_1.2.%.bbappend & vulkan-tools_1.2.%.bbappend

fre-wolf, 2020-07-27:
Hi,
I am trying to build a rpm-based-distribution on the Jetson Xavier NX Development Kit.
I'm new to Yocto. 

I downloaded Yocto Dunfell 3.1.1, cloned meta-tegra into the folder and added 
`/poky/poky-dunfell-23.0.1/meta-tegra \`
to bblayers.conf
in local.conf, I added following lines:
`MACHINE ??= "jetson-xavier-nx-devkit" #replacing previous machine
NVIDIA_DEVNET_MIRROR = "file:///path/to/sdkm_downloads"`
in meta-tegra/classes/nvidia_devnet_downloads.bbclass, I changed NVIDIA_DEVNET_MIRROR to the same path as well.

While trying to build a minimal build, I get following error after Bitbake is parsing recipes:
`ERROR: No recipes available for:
  /home/wof0101/poky/poky/meta-tegra/recipes-graphics/vulkan/vulkan-loader_1.2.%.bbappend
  /home/wof0101/poky/poky/meta-tegra/recipes-graphics/vulkan/vulkan-tools_1.2.%.bbappend`

do I have to edit local.conf further?
Thanks for the help,

madisongh, 2020-07-27:
If you're using the dunfell branch in poky/OE-Core, you need to check out a dunfell-based branch in meta-tegra to match.  Since you have a Xavier NX dev kit, you should check out `dunfell-l4t-r32.4.3` in meta-tegra.

See [this wiki page](https://github.com/madisongh/meta-tegra/wiki/Which-branch-should-I-use%3F) for more info on branching.

fre-wolf, 2020-07-27:
thank you for the quick response, that solved the issue

closed 2020-07-27 by fre-wolf