OE4T/meta-tegra #400: Xavier NX based custom machine

atharvanan1, 2020-07-28:
Hi,

I am working on a device that's based off of `jetson-xavier-nx-devkit-emmc` MACHINE. This machine refers to `xavier-nx.inc`, which adds MACHINEOVERRIDES - https://github.com/madisongh/meta-tegra/blob/faa128edd46d784f42e7e232b528ce9f1a77c9a3/conf/machine/include/xavier-nx.inc#L12

This includes a problem when we are trying to build a machine which inherits `jetson-xavier-nx-devkit-emmc` by using `require conf/machine/jetson-xavier-nx-devkit-emmc.conf` as mentioned here - [Creating a custom MACHINE](https://github.com/madisongh/meta-tegra/wiki/Creating-a-custom-MACHINE#custom-machine-definitions-for-existing-hardware).

Example for custom-machine:
`MACHINEOVERRIDES = "jetson-xavier-nx-devkit-emmc:${MACHINE}"
require conf/machine/jetson-xavier-nx-devkit-emmc.conf"
`
Results in MACHINEOVERRIDES as
MACHINEOVERRIDES="cuda:tegra:aarch64:armv8a:tegra194:xavier-nx:custom-machine"

To overcome this problem, I have changed custom-machine to
`require conf/machine/jetson-xavier-nx-devkit-emmc.conf"
MACHINEOVERRIDES_append = "jetson-xavier-nx-devkit-emmc"
`
Results in MACHINEOVERRIDES as
MACHINEOVERRIDES="cuda:tegra:aarch64:armv8a:tegra194:xavier-nx:custom-machine:jetson-xavier-nx-devkit-emmc"

This poses a problem with ordering of MACHINEOVERRIDES, and `jetson-xavier-nx-devkit-emmc` would override any `custom-machine` based settings, if added.

Other solution would be to add a python function that replaces `${MACHINE}` with `jetson-xavier-nx-devkit-emmc:${MACHINE}`

madisongh, 2020-07-28:
The simple way to do this would be to just set MACHINEOVERRIDES yourself, either via the replace that you mention, or just hard-coding the overrides:
```
require conf/machine/jetson-xavier-nx-devkit-emmc.conf
MACHINEOVERRIDES = "cuda:tegra:tegra194:xavier-nx:jetson-xavier-nx-devkit-emmc:${MACHINE}"
```
The hard-coding may be a bit ugly, but it at least avoids any issues that might crop up due to the use of the `:=` immediate assignment operator.

The other approach I've taken with custom MACHINEs is to start by copy-pasting the machine config (in this case, probably the combined contents of the `.conf` and the `xavier-nx.inc` file) and editing as needed.  This is probably the cleanest approach if your custom carrier is substantially different from the development kit carrier board.

Either way, you'll have to watch for changes to the machine configs when pulling in updates to meta-tegra, which might require corresponding changes in your customized config.

The machine config files in the layer could probably stand another round of refactoring to more cleanly separate out module-related settings from carrier-related settings. NVIDIA doesn't always do a very good job of that in the BSP, though, either - or at least they don't make it very clear where the dividing line is.

atharvanan1, 2020-08-02:
Thank you Matt for your comment.

> The machine config files in the layer could probably stand another round of refactoring to more cleanly separate out module-related settings from carrier-related settings.

I was wondering if we should add your suggestions to the wiki, until  we do the refactoring.

atharvanan1, 2020-08-02:
I found a way to cleanly add `xavier-nx` into MACHINEOVERRIDES. It's similar to how SOC_FAMILY is added into the MACHINEOVERRIDES.

atharvanan1, 2020-10-21:
Updated the [wiki](https://github.com/OE4T/meta-tegra/wiki/Creating-a-custom-MACHINE/8659faefedc3cde042fbd47082e0a0b8c9658689) with hardcode solution -

closed 2020-10-21 by atharvanan1