OE4T/meta-tegra #401: Writing to a specific partition on Jetson AGX Xavier's flash

levioctl, 2020-07-30:
Hi,
I would like to write a modified cbo.dtb file to the 'CPUBL-CFG' partition in my Jetson AGX Xavier's flash, in order to boot from network, however i'm not sure how to do that from the artifacts when building with meta-tegra (my IMAGE_FSTYPES is set to "tegraflash");
According to the Nvidia documentation ([BSP cusomization](https://docs.nvidia.com/jetson/archives/l4t-archived/l4t-3231/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fgetting_started.html%23wwpID0E06C0HA)), you have to run the 'flash.sh', however it's not one of the scripts in the tegraflash zip file that is produced by the build.
I tried copying this script to the directory where i extracted the tegraflash ZIP, but it did not find a jetson-xavier.conf file.

How can i write a file to a specific partition in the flash, using the meta-tegra artifacts? Is there a different way, or am I missing some extra configuration that's needed in my local.conf file?

Thanks.
Eliran

kekiefer, 2020-07-30:
The meta-tegra `doflash.sh` will flash the `CPUBL-CFG` partition with the contents of cbo.dtb. This file is generated by the `tegra-bootfiles` package, where the default source for it is set as `CBOOTOPTION_FILE ?= "${S}/bootloader/cbo.dts"`. In a bbappend to `tegra-bootfiles`, you can ship your own dts, and change the value for `CBOOTOPTION_FILE`.

levioctl, 2020-07-30:
Thanks @kekiefer !

If i understand correctly, the doflash.sh writes to all the partitions. So does this mean that there's no direct way to write to a specific partition (that is, i need to write to all the partitions)?
(i'm guessing that it could be done by figuring out the usage of tegraflash.py, but that would probably, but i was just wondering if there's a more 'high level' approach).

kekiefer, 2020-07-30:
You've got a few options. As far as writing a single partition from the flash helper, you should be able to use the `-c` switch to specify custom flash commands. So you can do something like this:

```
./doflash.sh -c "write CPUBL-CFG cbo.dtb; reboot"
```

Please note that it looks like there's a bug in the tegra194-flash-helper.sh that it doesn't accept this argument, though the code is in there to handle it. I just sent #405 to fix that. You can fix this script up manually by making these changes to the copy in your flash directory.

Since this partition doesn't need any special signing, you could also do this from runtime. One simple way (since this partition doesn't need any special signing) would be use `dd` to write the file directly to the partition:

```
dd if=cbo.dtb of=`blkid | grep CPUBL-CFG | cut -d: -f 1`
```

The last way would be to have a look at the `BUP_generator.py` and generate a custom update payload to apply with `nv_update_engine` from runtime.

levioctl, 2020-07-30:
I will give that a try. Thanks so much!

closed 2020-11-02 by madisongh