OE4T/meta-tegra #404: date starts at Fri Feb  7 15:50:XX UTC 2020 (dunfell)

DAMEK86, 2020-08-06:
after building a new image, the DateTime starts at **Fri Feb  7 15:50:XX UTC 2020**
without an internet connection for time sync over ntp, the date will be reset to **Fri Feb  7 15:50:XX UTC 2020** after restart.

I build the image with branch dunfell and dunfell-l4t-r32.4.3 but both results in the same issue.

Is there a workaround to set the default time to the build time?
I hope you can help me and thank you a lot!

madisongh, 2020-08-06:
This isn't specific to meta-tegra or the BSP version (other than the Jetson hardware doesn't come with a battery-backed RTC like some embedded dev kits do).

How this works depends on what you're including in your image.  For instance, are you using systemd and systemd-timesyncd, or are you using initscripts? With initscripts as startup, there are scripts installed that will read a timestamp out of `/etc/timestamp` at  boot time, and set the system time based on that; there's a rootfs post-processing hook that writes that file into the rootfs at build time.

I don't think OE-Core has a similar mechanism in place for systemd/systemd-timesyncd, however. systemd-timesyncd uses `/var/lib/systemd/timesync/clock`, and I don't see anything in the core layer to put a timestamp in that file. I think systemd-timesyncd will try and update that file at shutdown with the current timestamp so it can be used at the next boot; if it's not doing that, you probably need to go through the logs to determine why.

If you're using systemd and not systemd-timesyncd, you'll probably have to roll your own solution.

Another alternative would be to add a battery for the RTC so time is kept without having to create timestamp files.

DAMEK86, 2020-08-06:
I using systemd with no special settings or recipes for systemd-timesyncd and `cat/etc/timestamp` returns 20180309123456 which is as expected but `date` cmd returns the wrong date.
I use this distro also for an intel-nuc and an imx8 and see no problems there (rtc available).

u-boot says `U-Boot 2020.04 (Jul 30 2020 - 20:23:11 +0000)` and starts the kernel `Linux version 4.9.140-l4t-r32.4+g0be1a5744801 (oe-user@oe-host) (gcc version 9.3.0 (GCC) ) #1 SMP PREEMPT Mon Jul 13 20:46:03 UTC 2020`

the logs of systemd-timesyncd:
```
-- Logs begin at Fri 2020-02-07 15:50:52 UTC, end at Fri 2020-02-07 16:02:40 UTC. --
Feb 07 15:50:52 jetson-tx2 systemd[1]: Starting Network Time Synchronization...
Feb 07 15:50:53 jetson-tx2 systemd[1]: Started Network Time Synchronization.
Feb 07 15:50:55 jetson-tx2 systemd-timesyncd[3069]: Network configuration changed, trying to establish connection.
Feb 07 15:50:56 jetson-tx2 systemd-timesyncd[3069]: Network configuration changed, trying to establish connection.
Feb 07 15:51:06 jetson-tx2 systemd-timesyncd[3069]: Network configuration changed, trying to establish connection.
```

dmesg otherwise says:
```
[    0.442229] DTB Build time: Jul 13 2020 20:46:03
[    3.853770] rtc rtc1: alarm rtc device
[    3.853783] tegra_rtc c2a0000.rtc: rtc core: registered c2a0000.rtc as rtc1
[    3.853797] tegra_rtc c2a0000.rtc: Tegra internal Real Time Clock
[    4.304643] tegra_rtc c2a0000.rtc: setting system clock to 2000-01-01 12:13:32 UTC (946728812)
[    5.404715] systemd[1]: System time before build time, advancing clock.
```

but i have no idea, where the timestamp of **Feb 07 15:50:52** comes from

closed 2020-08-06 by DAMEK86

cg3dland, 2020-09-07:
@DAMEK86 @madisongh 

I am new to Yocto and meta-tegra, and I get same problem reported by this issue, I try to add systemd and systemd-timesyncd into my build with following 2 lines in local.conf:

INIT_MANAGER = "systemd"
CORE_IMAGE_EXTRA_INSTALL += " systemd-timesyncd"

unfortunately, I get following errors when I run command "bitbake core-image-sato":
ERROR: Nothing RPROVIDES 'systemd-timesyncd' (but /data/yocto1/tegra/meta/recipes-sato/images/core-image-sato.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'systemd-timesyncd' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['systemd-timesyncd']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'systemd-timesyncd']

the build process could run to end and the core image could be built while the line 'CORE_IMAGE_EXTRA_INSTALL += " systemd-timesyncd"' is removed from local.conf, but there is no systemd-timesyncd in the core image.
Anything wrong do I make in the local.conf?

BTW, @DAMEK86, do you resolve the date and time issue after the system restarts?

thanks,
George

madisongh, 2020-09-07:
@cg3dland  systemd-timesyncd isn't a separate package, it's a feature of the systemd build that is set via PACKAGECONFIG for the systemd recipe.  If you are using a recent version of OE-Core, it should be enabled by default.  So just setting INIT_MANAGER should be enough to enable it.

What problem are you trying to solve?

cg3dland, 2020-09-08:
The problem I am facing is, the system time is 2020-02-07 15:50:00 when the system boot up or reboot, I learn from your comment in the thread of this issue, the systemd/systemd-timesyncd could sync the date time from Internet, but after I enable systemd in the build, I could find following systemd packages when I run command ‘dpkg -l | grep systemd’:

ii  libsystemd0                                                          1:244.3-r0                                                arm64        Systemd systemd library
ii  pulseaudio-module-systemd-login                                      13.0-r0                                                   arm64        PulseAudio module for systemd-login
ii  systemd                                                              1:244.3-r0                                                arm64        A System and service manager
ii  systemd-compat-units                                                 1.0-r29                                                   arm64        Enhances systemd compatilibity with existing SysVinit scripts
ii  systemd-conf                                                         244.3-r0                                                  arm64        Systemd system configuration
ii  systemd-dev                                                          1:244.3-r0                                                arm64        A System and service manager - Development files
ii  systemd-extra-utils                                                  1:244.3-r0                                                arm64        A System and service manager
ii  systemd-serialgetty                                                  1.0-r5                                                    arm64        Serial terminal support for systemd
ii  systemd-serialgetty-dev                                              1.0-r5                                                    arm64        Serial terminal support for systemd - Development files
ii  systemd-vconsole-setup                                               1:244.3-r0                                                arm64        A System and service manager

I could not find systemd-timesyncd in my Linux image which built with Yocto and meta-tegra.

I am using Yocto (tags: yocto-3.1.2) and meta-tegra dunfell-l4t-r32.4.3.

madisongh, 2020-09-08:
As I said, it isn't a separate package, so you won't find it that way. Just use  `systemctl status systemd-timesyncd` to see if the service is running, and the `timedatectl` command to see the status of the system clock.

cg3dland, 2020-09-08:
@madisongh thanks for your help, the commands "systemctl status systemd-timesyncd" and "timwhich edatectl" work, it seems the NTP servers which timesyncd accessed are blocked by my ISP, I get following messages when the command  "systemctl status systemd-timesyncd" is run:

Sep 08 19:31:07 jetson-nano-qspi-sd systemd-timesyncd[3556]: Network configuration changed, trying to establish connection.
Sep 08 19:31:17 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.12:123 (time4.google.com).
Sep 08 19:31:27 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.0:123 (time1.google.com).
Sep 08 19:31:37 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.4:123 (time2.google.com).
Sep 08 19:31:48 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.8:123 (time3.google.com).
Sep 08 19:31:58 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.12:123 (time4.google.com).
Sep 08 20:06:16 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.0:123 (time1.google.com).
Sep 08 20:06:27 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.4:123 (time2.google.com).
Sep 08 20:06:37 jetson-nano-qspi-sd systemd-timesyncd[3556]: Timed out waiting for reply from 216.239.35.8:123 (time3.google.com).

so, the problem I am facing is a network issue, not the issue of Linux image.

lfdmn, 2020-09-08:
Stupid question maybe: are you able to ping these addresses?

I've had issues once on an image having both resolved and connman enabled
at the same time. I could not ping anything before disabling one.

cg3dland, 2020-09-09:
@lfdmn thanks for your help, I try to ping time4.google.com and time1.google.com, no response from these 2 servers, all packets are loss.

I try to update the NTP server configuration by updating the file /etc/systemd/timesyncd.conf, the config line "FallbackNTP" is changed from "#FallbackNTP=time1.google.com time2.google.com time3.google.com time4.google.com" to "#FallbackNTP=ntp.ubuntu.com", the ntp.ubuntu.com could be pinged.

After the system reboot, I try command "systemctl status systemd-timesyncd", the NTP servers time*.google.com are still be used by timesyncd, it is a bit strange to me,