OE4T/meta-tegra #405: u-boot device tree bad magic error on do_image_tegraflash

xerdink, 2020-08-12:
Using branch `dunfell-l4t-r32.4.3` with `MACHINE=jetson-nano-qspi-sd` building image `core-image-base` failed with:
```
| DEBUG: Executing shell function do_image_tegraflash
| Error at '/chosen': FDT_ERR_BADMAGIC
| WARNING: exit code 1 from a shell command.
```

It seems that u-boot cannot find FDT address right?

madisongh, 2020-08-12:
It's not U-Boot, it's the `fdtput` command being run during the image packaging that is trying to insert kernel args into the device tree under `/chosen`. Sounds like something's not right with the DTB file that got built, or possibly something to do with the tool so it's not recognizing the device tree blob in that file.

Can you provide more info on your build setup?

xerdink, 2020-08-13:
yes, it was about .dtb, I solved it. thank you.

closed 2020-08-13 by xerdink