OE4T/meta-tegra #406: cuda binaries issues with dunfell

nandra, 2020-08-17:
Hi,

I'm using downloaded cuda packages + other using SDK manager and everything was working fine. I bumped to latest dunfell branch and seeing issues like:
```
WARNING: cuda-binaries-ubuntu1804-native-10.2.89-1-r0 do_fetch: Failed to fetch URL file:///home/marek/projects/test/repo/downloads/cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb, attempting MIRRORS if available
```

so I fetch latest cuda stuff and I'm still getting:
```
WARNING: cuda-binaries-ubuntu1804-native-10.2.89-1-r0 do_fetch: Failed to fetch URL file:///home/marek/projects/test/repo/downloads/cuda-repo-ubuntu1804-10-2-local-10.2.89-440.40_1.0-1_amd64.deb, attempting MIRRORS if available
```

I tried to set ```CUDA_BINARIES = "cuda-binaries-ubuntu1604"``` but then I'm getting:
```
ERROR: Nothing PROVIDES 'cuda-binaries-ubuntu1604'
cuda-binaries-ubuntu1604 was skipped: incompatible with host aarch64-poky-linux (not in COMPATIBLE_HOST)
ERROR: Required build target 'test-app' has no buildable providers.
Missing or unbuildable dependency chain was: ['test-app', 'cuda-libraries', 'cuda-cusolver', 'cuda-binaries-ubuntu1604']

```

List of latest fetched files:
```
cuda-repo-cross-aarch64-10-2-local-10.2.89_1.0-1_all.deb
cuda-repo-ubuntu1604-10-2-local-10.2.89-440.40_1.0-1_amd64.deb
libvisionworks-repo_1.6.0.501_amd64.deb
libvisionworks-sfm-repo_0.90.4.501_amd64.deb
libvisionworks-tracking-repo_0.88.2.501_amd64.deb
NsightSystems-linux-public-2020.2.3.4-444e9f2.deb
NVIDIA_Nsight_Graphics_L4T_2020.1.20059.deb
NVIDIA_VisionWorks_References.zip
sdkml3_jetpack_l4t_44_ga.json
vpi-dev-0.3.7-cuda10-x86_64-linux.deb
vpi-lib-0.3.7-cuda10-x86_64-linux.deb
vpi-samples-0.3.7-cuda10-x86_64-linux.deb
```

Any idea what I'm doing wrong? :) Thanks.

madisongh, 2020-08-17:
Is this `dunfell` or `dunfell-l4t-r32.4.3`?  From the first message it looks like the latter.

Try
```
CUDA_BINARIES_x86-64 = "cuda-binaries-ubuntu1604"
```
to deal with this for now.  I'm not sure whether I just goofed when writing the instructions, or if I changed something that made it stop working.  I'll track that down, but the above should work.

nandra, 2020-08-18:
Yes it is ```dunfell-l4t-r32.4.3```

I've added :
```
CUDA_BINARIES_x86-64 = "cuda-binaries-ubuntu1604"
```

but then I still got:

```
WARNING: cuda-driver-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-driver-dev-10-0_10.2.89-1_arm64.deb;name=dev;subdir=cuda-driver-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-misc-headers-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-misc-headers-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-misc-headers-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-cusolver-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-cusolver-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-cusolver-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-cudart-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-cudart-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-cudart-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-npp-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-npp-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-npp-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-cusparse-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-cusparse-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-cusparse-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-cufft-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-cufft-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-cufft-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-curand-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-curand-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-curand-10.2.89-1, attempting MIRRORS if available
WARNING: cuda-nvcc-headers-10.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-nvcc-10-0_10.2.89-1_arm64.deb;name=main;subdir=cuda-nvcc-headers-10.2.89-1, attempting MIRRORS if available
WARNING: libcublas-10.2.2.89-1-r0 do_fetch: Failed to fetch URL https://repo.download.nvidia.com/jetson/common/pool/main/c/cuda/cuda-license-10-0_10.2.89-1_arm64.deb;name=lic;subdir=libcublas-10.2.2.89-1, attempting MIRRORS if available

```

madisongh, 2020-08-18:
It looks like you have `CUDA_VERSION` set to `10.0` instead of `10.2` somewhere - maybe a your local.conf, or a custom MACHINE config file?

nandra, 2020-08-18:
@madisongh thanks that was it. Sorry for bothering with this obvious thing ;). Thanks again

PR #408 (2020-08-19): cuda-shared-binaries: fix use of CUDA_BINARIES
Add an override for CUDA_BINARIES to nullify it for
that aarch64 builds, and reconfigure the _DEFAULT
variable settings accordingly.

Fixes #406
closed 2020-08-19 by madisongh