OE4T/meta-tegra #409: meson build system + cuda

nandra, 2020-08-24:
Hello,

I'm trying to add support for one application which use meson build system and it require cuda.
I inherited :```inherit meson cuda``` but I'm getting strange error in configure phase:
```
add_languages_for
|     comp = self.environment.detect_compiler_for(lang, for_machine)
|   File "/home/marek/projects/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app/git-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/mesonbuild/environment.py", line 1585, in detect_compiler_for
|     comp = self.compiler_from_language(lang, for_machine)
|   File "/home/marek/projects/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app/git-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/mesonbuild/environment.py", line 1563, in compiler_from_language
|     comp = self.detect_cuda_compiler(for_machine)
|   File "/home/marek/projects/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app/git-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/mesonbuild/environment.py", line 1129, in detect_cuda_compiler
|     linker = CudaLinker(compiler, for_machine, 'nvlink', CudaCompiler.LINKER_PREFIX, [], version=CudaLinker.parse_version())
|   File "/home/marek/projects/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app/git-r0/recipe-sysroot-native/usr/lib/python3.8/site-packages/mesonbuild/linkers.py", line 963, in __init__
|     super().__init__('nvlink', *args, **kwargs)
| TypeError: __init__() takes 6 positional arguments but 7 positional arguments (and 1 keyword-only argument) were given
| The Meson build system
| Version: 0.53.2
```

I'm using ```dunfell-l4t-r32.4.3``` meta-tagra branch. Thanks

madisongh, 2020-08-27:
I think you're breaking new ground here... from what I can see, meson CUDA support is experimental, and probably hasn't been tried with cross-compiling, let alone the somewhat non-standard (from a CUDA toolkit perspective) setup we have with OE, where the tools are in one tree and the target libraries are in another.

If this is an open-source app, point me at it and I can try and help.

nandra, 2020-08-27:
@madisongh I've asked same question on meson github and seems issue is resolved by bumping meson to 0.55. When bump meson to 0.55 I'm getting other error:
```
The Meson build system
Version: 0.55.0
Source dir: /home/marek/projects/test_cuda/test/build/tmp/work/armv8a_tegra194-poky-linux/app/git-r0/git
Build dir: /home/marek/projects/test_cuda/test/build/tmp/work/armv8a_tegra194-poky-linux/app/git-r0/build
Build type: cross build
Using 'PKG_CONFIG_PATH' from environment with value: '/home/marek/projects/test_cuda/test/build/tmp/work/armv8a_tegra194-poky-linux/app/git-r0/recipe-sysroot/usr/lib/pkgconfig:/home/marek/projects/test_cuda/test/build/tmp/work/armv8a_tegra194-poky-linux/app/git-r0/recipe-sysroot/usr/share/pkgconfig'
Project name: app
Project version: 1.00.0.1
C++ compiler for the build machine: g++ (gcc 5.4.0 "g++ (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609")
C++ linker for the build machine: g++ ld.bfd 2.26.1
Cuda compiler for the build machine: nvcc (nvcc 10.2.89)
Cuda linker for the build machine: nvcc nvlink 10.2.89
C++ compiler for the host machine: aarch64-poky-linux-g++ -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/marek/projects/test_cuda/test/build/tmp/work/armv8a_tegra194-poky-linux/app/git-r0/recipe-sysroot (gcc 7.3.0 "aarch64-poky-linux-g++ (GCC) 7.3.0")
C++ linker for the host machine: aarch64-poky-linux-g++ -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/marek/projects/test_cuda/test/build/tmp/work/armv8a_tegra194-poky-linux/app/git-r0/recipe-sysroot ld.bfd 2.34.0.20200220

../git/meson.build:1:0: ERROR: 'cuda' compiler binary not defined in cross or native file
``` 

in recipe I just inherited ```inherit meson cuda``` and also add to depends ```DEPENDS = "cuda-nvcc-native"``` but when looking to cuda.bblacck there is some setup for cmake but I'm not sure how to do it for meson (never used meson build system though). Unfortunately app is not opensource but I think cuda-samples can be tried :). Thanks.

nandra, 2020-08-28:
@madisongh would be an option to rewrite build system using cmake? cmake should have better support for cuda cross-compiling (I saw some changes in cuda.bbclass related to cmake). Thanks.

madisongh, 2020-08-28:
@nandra If cmake is an option for you, that might be a better choice for now.  I did manage to get meson to to cross-compile all of the cuda test cases in its repository, after patching meson itself to fix what looks like a bug in how it handles the `cuda =` setting in a cross-compile configuration file.  I've attached the patch here, and PR #412 has the changes to `cuda.bbclass` that worked for my tests, in case you'd like to try it out yourself.

[0001-Fix-environment-detect_cuda_compiler-method.patch.txt](https://github.com/OE4T/meta-tegra/files/5142806/0001-Fix-environment-detect_cuda_compiler-method.patch.txt)

nandra, 2020-08-31:
@madisongh thanks a lot for fix. Unfortunately it still doesn't work for my use case. I also tried to use this simple cuda program (https://github.com/mesonbuild/meson/issues/5791) with ```meson.build``` file but I'm getting:
```
Log data follows:
| DEBUG: Executing shell function do_configure
| NOTE: Executing meson ...
| The Meson build system
| Version: 0.55.1
| Source dir: /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0
| Build dir: /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/build
| Build type: cross build
| Using 'PKG_CONFIG_PATH' from environment with value: '/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot/usr/lib/pkgconfig:/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot/usr/share/pkgconfig'
| Project name: hello-cuda
| Project version: undefined
| C++ compiler for the build machine: g++ (gcc 5.4.0 "g++ (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609")
| C++ linker for the build machine: g++ ld.bfd 2.26.1
| Cuda compiler for the build machine: nvcc (nvcc 10.2.89)
| Cuda linker for the build machine: nvcc nvlink 10.2.89
| C++ compiler for the host machine: aarch64-poky-linux-g++ -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot (gcc 7.3.0 "aarch64-poky-linux-g++ (GCC) 7.3.0")
| C++ linker for the host machine: aarch64-poky-linux-g++ -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot ld.bfd 2.34.0.20200220
| 
| ../meson.build:1:0: ERROR:
| 
| A full log can be found at /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/build/meson-logs/meson-log.txt
| ERROR: meson failed
| WARNING: /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/temp/run.do_configure.5025:1 exit 1 from 'exit 1'
| ERROR: Execution of '/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/temp/run.do_configure.5025' failed with exit code 1:
| The Meson build system
| Version: 0.55.1
| Source dir: /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0
| Build dir: /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/build
| Build type: cross build
| Using 'PKG_CONFIG_PATH' from environment with value: '/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot/usr/lib/pkgconfig:/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot/usr/share/pkgconfig'
| Project name: hello-cuda
| Project version: undefined
| C++ compiler for the build machine: g++ (gcc 5.4.0 "g++ (Ubuntu 5.4.0-6ubuntu1~16.04.12) 5.4.0 20160609")
| C++ linker for the build machine: g++ ld.bfd 2.26.1
| Cuda compiler for the build machine: nvcc (nvcc 10.2.89)
| Cuda linker for the build machine: nvcc nvlink 10.2.89
| C++ compiler for the host machine: aarch64-poky-linux-g++ -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot (gcc 7.3.0 "aarch64-poky-linux-g++ (GCC) 7.3.0")
| C++ linker for the host machine: aarch64-poky-linux-g++ -march=armv8-a+crc -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/recipe-sysroot ld.bfd 2.34.0.20200220
| 
| ../meson.build:1:0: ERROR:
| 
| A full log can be found at /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/build/meson-logs/meson-log.txt
| WARNING: /home/marek/projects/testproj/test/build/tmp/work/armv8a_tegra194-poky-linux/test-app-test/git-r0/temp/run.do_configure.5025:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/marek/projects/testproj/test/meta-veo/recipes-veo/test-app-test/test-app-test_git.bb:do_configure) failed with exit code '1'
```

Simple test file:
```
#include <iostream>
#include <cuda_runtime.h>

__global__
void hello(int *b)
{
    *b = 42;
}

int main(){
    int *pd;
    cudaMalloc(&pd, sizeof(int));
    hello<<<1, 1>>>(pd);
    cudaDeviceSynchronize();

    int res = 0;
    cudaMemcpy(&res, pd, sizeof(int), cudaMemcpyDeviceToHost );
    cudaFree(pd);

    std::cout << "hello world: " << res << std::endl;
    return 0;
}
```

meson.build file:
```
project('hello-cuda', 'cpp', 'cuda')

cuda_comp = meson.get_compiler('cuda')
cuda = import('unstable-cuda')
nvcc_flags = cuda.nvcc_arch_flags(cuda_comp, 'Common')
executable('hello-cuda', 'hello.cu', cuda_args : nvcc_flags)
```

and recipe:
```
SUMMARY = "test application"
SECTION = "base"
LICENSE = "CLOSED"

inherit meson cuda

DEPENDS += "cuda-nvcc-native"

SRC_URI = "file://hello.cu file://meson.build"

S="${WORKDIR}"

```

Thanks a lot

madisongh, 2020-08-31:
Did you add a bbappend to apply that patch I mentioned to meson?  From the empty ERROR: message in the log, it doesn't look like it.

nandra, 2020-08-31:
@madisongh thanks for pointer. That was the issue that I somehow not correctly setup meson patch. Now everything works as expected!. Thanks again.

madisongh, 2020-09-02:
I've submitted a PR for the meson change upstream.  Since the bbclass change is somewhat dependent on whether or not that change is accepted there, I'll hold off until I hear something about that PR.

nandra, 2020-09-02:
@madisongh ok thanks a lot.