OE4T/meta-tegra #410: cannot reboot xavier devkit

lfdmn, 2020-08-24:
Hi,

Does anyone have issues with rebooting xavier devkits?

Mine, zeus-l4t-r32.3.1, doesn't and I see some i2c errors in terminal

```
[  202.536205] tegra-i2c c250000.i2c: pio timed out addr: 0x1a tlen:28 rlen:4
[  202.536345] tegra-i2c c250000.i2c: --- register dump for debugging ----
[  202.536453] tegra-i2c c250000.i2c: I2C_CNFG - 0x22c00
[  202.536540] tegra-i2c c250000.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001
[  202.536645] tegra-i2c c250000.i2c: I2C_FIFO_CONTROL - 0x0
[  202.536752] tegra-i2c c250000.i2c: I2C_FIFO_STATUS - 0x800080
[  202.536856] tegra-i2c c250000.i2c: I2C_MST_FIFO_CONTROL - 0x70000
[  202.536954] tegra-i2c c250000.i2c: I2C_MST_FIFO_STATUS - 0x7c0000
[  202.537058] tegra-i2c c250000.i2c: I2C_MST_PACKET_TRANSFER_CNT - 0x0
[  202.537160] tegra-i2c c250000.i2c: I2C_INT_MASK - 0x7d
[  202.537243] tegra-i2c c250000.i2c: I2C_INT_STATUS - 0x2
[  202.537328] tegra-i2c c250000.i2c: i2c transfer timed out addr: 0x1a
```

it might eventually reboot after some minutes.

I use force reboot which should be ~instant.

Thanks
-Damien

madisongh, 2020-08-27:
That looks like I2C address 7-001A.   The default device tree configures an RT5659 audio codec at that address, but I don't think it actually comes with the dev kit.  Do you have something connected to I2C bus 7, or have you maybe configured in some ALSA audio that's trying to use that codec?  By default the layer only configures in HDMI audio, so it shouldn't normally be a problem.

lfdmn, 2020-09-01:
I have to look more into it. 

I am testing a bare devkit and no additional device on bus 7. 

I'm not implicitly including alsa but maybe some Qt package or someone else does. 

Ultimately I don't need audio at all so I'll disable it completely.

przemyslaw-chwiala, 2020-10-12:
@madisongh Is there a way to disable those unused I2C addressed? I have the same issue but with some other port
```
[   93.049308] tegra-i2c 7000c700.i2c: --- register dump for debugging ----
[   93.056026] tegra-i2c 7000c700.i2c: I2C_CNFG - 0x22c00
[   93.061184] tegra-i2c 7000c700.i2c: I2C_PACKET_TRANSFER_STATUS - 0x10001
[   93.068378] tegra-i2c 7000c700.i2c: I2C_FIFO_CONTROL - 0x1c
[   93.073989] tegra-i2c 7000c700.i2c: I2C_FIFO_STATUS - 0x800040
[   93.080616] tegra-i2c 7000c700.i2c: I2C_INT_MASK - 0x6c
[   93.085927] tegra-i2c 7000c700.i2c: I2C_INT_STATUS - 0x2
[   93.091370] tegra-i2c 7000c700.i2c: i2c transfer timed out addr: 0x50
[  103.282783] tegra-i2c 7000c700.i2c: rx dma timeout txlen:28 rxlen:128
```

madisongh, 2020-10-12:
@przemyslaw-chwiala You'd need to modify the device tree and disable the node so the driver doesn't try probing the device.

madisongh, 2020-12-11:
@lfdmn I assume you got this straightened out, so am closing.

closed 2020-12-11 by madisongh