OE4T/meta-tegra #413: Dunfell: libcublas error in bitbake

fre-wolf, 2020-08-31:
I'm trying to build for the Nvidia Jetson Xavier NX devkit. I use the dunfell branch. While trying to build a Core-image-minimal with opencv, it threw the following error message:
`poky-dunfell-23.0.1/build/tmp/work/armv8a_tegra194-poky-linux/opencv/4.1.0-r0/recipe-sysroot/usr/local/cuda-10.2/include/crt/host_config.h:138:2: error: #error -- unsupported GNU version! gcc versions later than 8 are not supported!
  138 | #error -- unsupported GNU version! gcc versions later than 8 are not supported!
      |  ^~~~~`
As dunfell doesn't have gcc 8, I copied the recipe from the zeus branch (poky/meta/recipes-devtools/gcc) and put the contents into the existing folder in dunfell. After changing to `GCCVERSION = "8.%"`, I get following new error:
`ERROR: libcublas-10.2.2.89-1-r0 do_compile: Execution of '/home/wof0101/poky-dunfell-23.0.1/build/tmp/work/armv8a_tegra-poky-linux/libcublas/10.2.2.89-1-r0/temp/run.do_compile.12721' failed with exit code 1:
mv: cannot stat '/home/wof0101/poky-dunfell-23.0.1/build/tmp/work/armv8a_tegra-poky-linux/libcublas/10.2.2.89-1-r0/libcublas-10.2.2.89-1/usr/lib/aarch64-linux-gnu/lib*': No such file or directory
WARNING: exit code 1 from a shell command.`

Does this come from changing the gcc wrongly? Do I have to adjust a *.bbappend?

I use following layers:
meta-tegra (branch dunfell-l4t-r32.4.3)
meta-openembedded (branch dunfell)
meta-qt5 (branch dunfell)

Thanks

madisongh, 2020-08-31:
We do have gcc 8 recipes available, so you don't have to copy them over yourself - see [this wiki page](https://github.com/OE4T/meta-tegra/wiki/Using-gcc8-from-the-contrib-layer) for more info.

To clear up the libcublas error, try `bitbake -c clean libcublas` and then re-run your build.  If that doesn't clear up the issue, yotry removing everything under `$BUILDDIR/tmp` just to make sure there isn't some cruft leftover from previous builds that might be interfering.  As long as you still have your shared-state cache, the rebuild shouldn't take too long.

madisongh, 2020-12-11:
I assume you resolved this issue, since it's been a few months.  Closing.

closed 2020-12-11 by madisongh