OE4T/meta-tegra #414: Layer home page

lfdmn, 2020-09-01:
@madisongh , i've made a script to generate an opensource acknowledgments html page for all the packages in our image.
[image]

Packages from meta-tegra are missing the HOMEPAGE keywords I use to generate the links.

How would you like meta-tegra to be referred to?

madisongh, 2020-09-01:
I suppose we should fix that in the recipes.  HOMEPAGE isn't strictly required, but it's useful fro this sort of thing.

Recipes for NVIDIA packages should probably point to the NVIDIA developer website.

The kernel recipe should, I suppose, point to kernel.org.

Recipes for code originating from this project should, for now point to https://github.com/OE4T . One of my to-do list items is to make a start on creating a Github pages site for the project so we'll have an actual home page and a (hopefully) an easier way to locate documentation.  I just haven't got round to it yet.

ichergui, 2020-09-01:
Hi @madisongh 
Please let me know if I can help on this one.
Have a nice day.

closed 2020-11-02 by madisongh