OE4T/meta-tegra #415: Kernel build timestamp

msalvinik, 2020-09-01:
Hi,
I'm using with success your meta-layer (thank you for your great work!) to build an image for Xavier NX platform using dunfell branch. The build completes without errors, and the board boots. But when I type `uname -a` I see that the Linux build timestamp is "Jun 11 13:28:30 UTC 2020" instead of today. This date is the same contained in `nvidia,dtbbuildtime` property in the devicetree.

Is it normal? If yes, how (and why) this date comes into the `uname` data?

Thanks in advance.

Mauro

madisongh, 2020-09-01:
That field in the device tree comes from this line in the source:
```
nvidia,dtbbuildtime = __DATE__, __TIME__;
```
which is filled in by `cpp` when the dts file is compiled. If your image build used kernel artifacts that came from your shared-state cache, you would see the date time those artifacts were built, rather than the current date and time.  That's normal.  The same situation applies to the information reported by `uname` - the date/time stamp is generated when the kernel is compiled.

msalvinik, 2020-09-01:
Thanks @madisongh for the quick reply. I cleaned the sstate cache, but the result was the same.
Meanwhile I found the answer: this behavior depends on the `reproducible_build` class, inherited by default by `poky` distro starting from `dunfell` release.
This have an impact in `kernel.bbclass`, that retrieves the build date from the last git commit instead of using the current date (see first lines of `kernel_do_compile` function).

Thanks again
Mauro

closed 2020-09-01 by msalvinik

madisongh, 2020-09-01:
Ah, right.  I forgot about the reproducible build setting...