OE4T/meta-tegra #416: Recommendations about Jetson AGX Xavier

ichergui, 2020-09-01:
Hi @madisongh @dwalkes 

I was working on the Nvidia Jetson TX2 platform, I used the **meta-tegra**
I'm planning to buy the Jetson AGX Xavier.

I have done a quick comparison between Xavier NX and AGX Xavier, I chose the AGX xavier because it is simply more powerful than the other one.

My goal is to work on video/image processing and security (Secure boot, Tegra fuses, Trusted Zone,...).

Could you please guys give your feedback about the AGX Xavier platform ? 
Is anyone use the Jtag [Lauterbach](https://www.lauterbach.com/frames.html?home.html) with this platform ? 

Please let me know if I can help you.

Have a nice day.

koansoftware, 2020-09-01:
hi @ichergui 
we tested the Nvidia Jetson AGX Xavier using this great meta layer and it worked like a charm!
[Koan's NVIDIA Jetson AGX Xavier with Yocto Project](https://koansoftware.com/yocto-project-on-nvidia-jetson-agx-xavier/)
P.S. Never used the expensive Lauterbach.
-- Marco Cavallini

ichergui, 2020-09-01:
> hi @ichergui
> we tested the Nvidia Jetson AGX Xavier using this great meta layer and it worked like a charm!
> [Koan's NVIDIA Jetson AGX Xavier with Yocto Project](https://koansoftware.com/yocto-project-on-nvidia-jetson-agx-xavier/)
> P.S. Never used the expensive Lauterbach.
> -- Marco Cavallini

@koansoftware 
Thanks for your response.

dwalkes, 2020-09-01:
@ichergui I haven't used AGX Xavier myself

madisongh, 2020-09-01:
I've only used the AGX Xavier for testing the layer - got one of the early 16GiB dev kits at a discount, fortunately, since it's relatively pricey.  It's worked fine for me, but due to cost and power constraints, I'm unlikely to be using it in a project.

I haven't used a Lauterbach JTAG unit for many years, so I can't comment on their Xavier support, either.  With the SOMs, there's less of a need for investing in a JTAG debugger.

ichergui, 2020-09-01:
Thanks guys for your feedback.

> I haven't used a Lauterbach JTAG unit for many years, so I can't comment on their Xavier support, either. With the SOMs, there's less of a need for investing in a JTAG debugger.

@madisongh I wish to use the JTAG Lauterbach, because Nvidia propose some samples  [here](https://developer.nvidia.com/embedded/linux-tegra), also it is very useful when trying to perform low level debugging (e.g. TAs, Exception Levels (ELx), ...).

lfdmn, 2020-09-02:
@ichergui we use both TX2 and AGX. 

We started with TX1, then rapidly switched to TX2 before the project was out. 

For another project, we needed more performance and PCIe lanes so we settled for AGX.

We do video/image near real time analysis. With TX2 we are able to reach  ~500+fps. With AGX we reached ~1200fps with the same software. There is probably still some room for optimization.

From experience, the move from TX2 to AGX was smooth as meta-tegra is being improved all the time. The only hiccups I encounter was bumping our distro from pyro to zeus (and my little knowledge of yocto at the start of the project :) ), and fixing build dependencies like Qt libraries; which is now business as usual as I maintain the distros.

ichergui, 2020-09-07:
Thanks @lfdmn for your response it very helpful

closed 2020-09-21 by ichergui