OE4T/meta-tegra #418: Wiki documentation device tree / pinmux

jlucke86, 2020-09-05:
I've been trying for a few hours to setup the pinmux without success. I taught myself how to get the register values from the nvidia pinmux spreadsheet tool by generating the .dtsi files and then convert the output to cfg and then manually diff the files so I can see the cpu registers that require changing and then use use devmem2 to set the settings on a live system.

Could I please have some guidance what is the accepted way to slice in the .dtsi files or .cfg into my system images.

kekiefer, 2020-09-06:
You can do this by adding a bbappend to tegra-flashvars that deploys the customized pinmux .cfg files (you don't need the dtsi files except to generate those cfgs). You also need to provide a copy of the `flashvars` file that refers to your cfg files. You'll end up with something like this:

```
meta-mylayer/recipes-bsp/tegra-binaries/tegra-flashvars_%.bbappend
meta-mylayer/recipes-bsp/tegra-binaries/tegra-flashvars/mymachine/
meta-mylayer/recipes-bsp/tegra-binaries/tegra-flashvars/mymachine/flashvars
meta-mylayer/recipes-bsp/tegra-binaries/tegra-flashvars/mymachine/tegra19x-mb1-pinmux-mymachine.cfg
meta-mylayer/recipes-bsp/tegra-binaries/tegra-flashvars/mymachine/tegra19x-mb1-padvoltage-mymachine.cfg
```

```
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

SRC_URI_append_mymachine = " file://tegra19x-mb1-pinmux-mymachine.cfg \
    file://tegra19x-mb1-padvoltage-mymachine.cfg \
"

do_install_append_mymachine() {
    install -d ${D}${datadir}/tegraflash
    install -m 0644 ${S}/*.cfg ${D}${datadir}/tegraflash/
}
```

jlucke86, 2020-09-06:
Thanks, that information is really helpful.

I'm reading the existing `recipes-bsp/tegra-binaries/tegra-flashvars/xavier-nx/flashvars` file and its referring to file
```
PINMUX_CONFIG="tegra19x-mb1-pinmux-p3668-a01.cfg"
```

Where do I find this pre existing tegra19x-mb1-pinmux-p3668-a01.cfg file? It doesn't seem to be in tegra-linux_4.9 or meta-tegra repos? I usually like to read the existing code / settings before changing anything

Contents of tegra-flashvars/xavier-nx/flashvars
```
FLASHVARS="BPFDTB_FILE PMIC_CONFIG SCR_COLD_BOOT_CONFIG MISC_COLD_BOOT_CONFIG PINMUX_CONFIG PMC_CONFIG UPHY_CONFIG DEVICE_CONFIG GPIOINT_CONFIG PROD_CONFIG SCR_CONFIG BR_CMD_CONFIG DEV_PARAMS"
BPFDTB_FILE="tegra194-a02-bpmp-p3668-a00.dtb"
PMIC_CONFIG="tegra194-mb1-bct-pmic-p3668-0001-a00.cfg"
SCR_COLD_BOOT_CONFIG="tegra194-mb1-bct-scr-cbb-mini-p3668.cfg"
MISC_COLD_BOOT_CONFIG="tegra194-mb1-bct-misc-l4t.cfg"
PINMUX_CONFIG="tegra19x-mb1-pinmux-p3668-a01.cfg"
PMC_CONFIG="tegra19x-mb1-padvoltage-p3668-a01.cfg"
UPHY_CONFIG=""
DEVICE_CONFIG="tegra19x-mb1-bct-device-qspi-p3668.cfg"
GPIOINT_CONFIG="tegra194-mb1-bct-gpioint-p3668-0001-a00.cfg"
PROD_CONFIG="tegra19x-mb1-prod-p3668-0001-a00.cfg"
SCR_CONFIG="tegra194-mb1-bct-scr-cbb-mini-p3668.cfg"
BR_CMD_CONFIG="tegra194-mb1-bct-reset-p3668-0001-a00.cfg"
DEV_PARAMS="tegra194-br-bct-qspi.cfg"
```

kekiefer, 2020-09-06:
This comes from the `tegra-binaries` package which is unpacked into: `${TMPDIR}/work-shared/L4T-${L4T_BSP_ARCH}-${PV}-${PR}/Linux_for_Tegra`. So, for master you'd find that here: `${BUILDDIR}/tmp/work-shared/L4T-tegra-32.4.3-r0/Linux_for_Tegra/bootloader/t186ref/BCT/tegra19x-mb1-pinmux-p3668-a01.cfg`

jlucke86, 2020-09-06:
OK that is super helpful. I'll give it another go now.

jlucke86, 2020-09-06:
At this stage it looks like you've helped me to get the cfg files into my file system. I'm not sure the cfg files are correct now

Is there a special version of the pinmux 2 cfg python script suited for the xavier-nx, or I can use any of them located under the t19x / t194 folder?

madisongh, 2020-09-06:
It should be the one at `Linux_for_Tegra/kernel/pinmux/t19x`.  See the README.txt file there for instructions. The  copy in `t186` is for the TX2 only.

I've only used the TX2 cfg generator myself, but I did end up manually adjusting the resulting cfg files, since the script doesn't always generate 100% correct results.  You may have to do the same with the t19x output... generating diffs from the stock cfg files and consulting the documentation on the pins to make sure the differences really reflect what you entered into the spreadsheet.

closed 2021-06-10 by madisongh