OE4T/meta-tegra #419: Fail to start docker due to no cuda-capable device on core-image-minimal

cg3dland, 2020-09-09:
I build a Nano minimal image (core-image-minimal, jetson-nano-qspi-sd) with cuda, cudnn, tensorrt and nVidia container, the following lines are added into local.conf for building cuda, cudnn, tensorrt and nVidia container:

IMAGE_INSTALL_append += " cuda-toolkit cuda-compiler cuda-nvcc cuda-libraries cuda-driver cuda-cudart cuda-command-line-tools"
IMAGE_INSTALL_append += " cudnn tensorrt gstreamer1.0-plugins-nvvideo4linux2"
IMAGE_INSTALL_append += " l4t-graphics-demos-egldevice tegra-udrm-probeconf tegra-libraries libdrm"

DISTRO_FEATURES_append = " virtualization"
IMAGE_INSTALL_append += " nvidia-docker"

but when I start docker with command "docker run -it --rm --net=host --runtime nvidia --ipc=host --gpus all  nvcr.io/nvidia/deepstream-l4t:5.0-dp-20.04-samples", I get following error message:

docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"process_linux.go:413: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: cuda error: no cuda-capable device is detected\\\\n\\\"\"": unknown.

with same local.conf, I build a sato image (core-image-sato), same command  "docker run -it --rm --net=host --runtime nvidia --ipc=host --gpus all  nvcr.io/nvidia/deepstream-l4t:5.0-dp-20.04-samples" works normally on this sato image without above error.

What config do I missed in the local.conf?

BTW, the error might relates to a comment from Madison (https://github.com/OE4T/meta-tegra/issues/411#issuecomment-686055176), I presume some cuda packages are missed in minimal image, what packages are missed by me?

madisongh, 2020-09-09:
> what packages are missed by me?

The GPU driver is the main one causing the error.  Try using `core-image-base` as your starting point, rather than `core-image-minimal`.

cg3dland, 2020-09-09:
Is it possible to add the the GPU driver into core-image-minimal?

madisongh, 2020-09-09:
Yes, but you may end up with other packages missing.  Since you're new to all this, try something simple first.  When you learn more about how the build system works, you can try something more advanced.

cg3dland, 2020-09-10:
The error reported by this issue goes away with core-image-base.

closed 2020-10-20 by cg3dland