OE4T/meta-tegra #420: [XAVIER NX] When echo 3550000.xudc> UDC， got error INVALIDARGUMENT

VCBE123, 2020-09-10:
I suppose to expose the OTG Micro-B USB port as an HID device such as a keyboard or gamepad.

So I stop the USB Device Mode by
```bash
service nv-l4t-usb-device-mode stop
systemctl disable nv-l4t-usb-device-mode.service
```

Then I modify the **/opt/nvidia/l4t-usb-device-mode/nv-l4t-usb-device-mode-start.sh** as :
```bash
#!/bin/bash

# Copyright (c) 2017-2020, NVIDIA CORPORATION.  All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions
# are met:
#  * Redistributions of source code must retain the above copyright
#    notice, this list of conditions and the following disclaimer.
#  * Redistributions in binary form must reproduce the above copyright
#    notice, this list of conditions and the following disclaimer in the
#    documentation and/or other materials provided with the distribution.
#  * Neither the name of NVIDIA CORPORATION nor the names of its
#    contributors may be used to endorse or promote products derived
#    from this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS ``AS IS'' AND ANY
# EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR
# PURPOSE ARE DISCLAIMED.  IN NO EVENT SHALL THE COPYRIGHT OWNER OR
# CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL,
# EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO,
# PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR
# PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY
# OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT
# (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
# OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.

set -e

script_dir="$(cd "$(dirname "$0")" && pwd)"
. "${script_dir}/nv-l4t-usb-device-mode-config.sh"

# Wait for any modules to load and initialize
for attempt in $(seq 60); do
    udc_dev_t210=700d0000.xudc
    if [ -e "/sys/class/udc/${udc_dev_t210}" ]; then
        udc_dev="${udc_dev_t210}"
        break
    fi
    udc_dev_t186=3550000.xudc
    if [ -e "/sys/class/udc/${udc_dev_t186}" ]; then
        udc_dev="${udc_dev_t186}"
        break
    fi
    sleep 1
done
if [ "${udc_dev}" == "" ]; then
    echo No known UDC device found
    exit 1
fi

#(macs_file="${script_dir}/mac-addresses"
#if [ -f "${macs_file}" ]; then
#    . "${macs_file}"
#else
#    # Generate unique data
#    if [ -f /proc/device-tree/serial-number ]; then
#        random="$(md5sum /proc/device-tree/serial-number|cut -c1-12)"
#    else
[... 156 lines cut ...]
# bridge, and assigns the static IP to that bridge. This allows the script to
# more easily handle the potentially variable set of Ethernet devices.
#
# If your custom use-case requires separate IP addresses per interface, or
# only ever has one interface active, you may modify this script to skip
# bridge creation, and assign IP address(es) directly to the interface(s).

if [ ${enable_rndis} -eq 1 ]; then
    /sbin/brctl addif l4tbr0 "$(cat functions/rndis.usb0/ifname)"
    /sbin/ifconfig "$(cat functions/rndis.usb0/ifname)" up
fi

if [ ${enable_ecm} -eq 1 ]; then
    /sbin/brctl addif l4tbr0 "$(cat functions/${ecm_ncm}.usb0/ifname)"
    /sbin/ifconfig "$(cat functions/${ecm_ncm}.usb0/ifname)" up
fi

cd - # Out of /sys/kernel/config/usb_gadget

# Create a local disk device that exposes the same filesystem image that's
# exported over USB. This will allow local users to see the files too.
#/sbin/losetup -f -r "${fs_img}"

exit 0
```

madisongh, 2020-09-10:
That script isn't used in the implementation of USB device mode support with this layer - see [this wiki page](https://github.com/OE4T/meta-tegra/wiki/USB-device-mode-support). If you're using the stock L4T/JetPack, you should post this question on the NVIDIA developer forum, if you haven't already.

Just off-hand, though, the EINVAL you're getting is likely due to an unrecognized configuration setup. I don't really know enough about using the configfs interface to know what might be wrong.  Also, I don't see USB HID gadget support in the default kernel configuration, so you may need to add CONFIG_USB_CONFIGFS_F_HID to your kernel config.

VCBE123, 2020-09-10:
Yes, I rebuild the kernel and set CONFIG_USB_CONFIGFS_F_HID to support HID. And I also have posted on the NVIDIA forum.
Thank you very much.

closed 2020-09-10 by VCBE123