OE4T/meta-tegra #423: OTA (Over the air updates) possible?

mrspirytus, 2020-09-15:
Hello,

I am waiting for my unit to arrive (AGX). So I am planning to do some development. Part of my project is remote system update. Did anyone attempted doing this? How do you go about it?
It could be partial or fill upgrade. Preferably I would like dial partition so when upgrade fails the unit can revert to previous version. 
Thank you

dwalkes, 2020-09-15:
Hi @mrspirytus 
There's a project to support remote OTA update using mender, see https://github.com/OE4T/meta-mender-community/, https://hub.mender.io/t/nvidia-tegra-jetson-tx2/123 and https://hub.mender.io/t/nvidia-tegra-jetson-nano/1360.  Currently only Nano, TX2, and Xavier NX are tested/supported.  I don't know enough about the differences between Xavier NX and Xavier AGX to comment on the amount of work it would take to add support for Xavier AGX.

See the related thread at https://github.com/madisongh/test-distro/issues/5 as well - there's a plan to move mender support going forward into a repository at https://github.com/OE4T/tegra-demo-distro.   You can watch this PR for more information: https://github.com/OE4T/tegra-demo-distro/pull/1

There have also been some threads about swupdate support if you search here, but I'm not sure if others have shared anything related to swupdate for tegra yet.

kekiefer, 2020-09-15:
AGX isn't tested, but should be supported by meta-mender-tegra, as it's almost identical to the NX and TX2 cboot.

Swupdate is going to be more DIY than mender, but straightforward (for the cboot machines) since it just needs to leverage the nv_update_engine for A/B rootfs selection. It would be nice to have an example in tegra-demo-distro, but we don't have one.

madisongh, 2020-09-18:
I ran a test build for an AGX Xavier using the mender integration in [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro), and it worked fine.

closed 2020-12-11 by madisongh