OE4T/meta-tegra #424: rootfs on external (SATA) device

eshcheglov, 2020-09-17:
Hello!

I'm wondering if there is a way to use SATA as default rootfs location. 

As I know u-boot should be located at eMMC partition anyway, but maybe it's possible to keep u-boot and rootfs separately on different devices?

My goal is mender integration + rootfs on SATA, so I can't just copy rootfs to SATA from eMMC after flashing

Thank you in advance

madisongh, 2020-09-17:
Which Jetson are you planning on using?  You mention U-Boot, so I'm guessing either a Nano or a TX2.  I think U-Boot for Nano includes SATA support, but I'm not sure if it's there on the TX2.  That would be the first thing to check.

A few other things off the top of my head:
* You would need to develop an installer/recovery image to help with getting the SATA drive partitioned, formatted, and loaded up with the rootfs as part of initial flashing.
* You would need to adjust the Mender variables that indicate where to boot from to configure the right name and partition numbers on the SATA drive in U-Boot and in Linux.  The default setup assumes, of course, booting from the eMMC.
* Double-check that the SATA drive you're looking to use works properly with the driver/controller on the Jetson.  I haven't used SATA for booting, but I have used it for data storage in a couple of products in the past, and ran into some compatibility issues.

An alternative to directly booting from the SATA drive from U-Boot, you could put a kernel/initramfs into a partition on the eMMC, point U-Boot at that, and then have the init script in the initramfs mount the SATA drive for you.

eshcheglov, 2020-09-18:
Thank you for reply! 
Forgot to mention, I'm using TX2. I'm sure that U-boot is able to boot SATA-based rootfs, already tested with Ubuntu 16.04.

As I understand, partitioning for SATA should be added to flash.xml file, right? 
But how can I specify the root partition for flasher? Original nVidia "flash.sh" script can read special file called "l4t-rootfs-uuid.txt" to determine rootfs drive, but I was unable to find any mention of this file in meta-tegra

Your concern regarding SATA drive is quite unexpected, can you please tell what kind of compatibility issues you faced with?

madisongh, 2020-09-18:
> As I understand, partitioning for SATA should be added to flash.xml file, right?

I don't think the tegraflash tool supports "flashing" to a SATA drive, which is why I suggested the two-step process - flash an installer/recovery image to the MMC using tegraflash.  Have the tools in that image to format and install your final rootfs onto the SATA drive.

> But how can I specify the root partition for flasher? Original nVidia "flash.sh" script can read special file called "l4t-rootfs-uuid.txt" to determine rootfs drive, but I was unable to find any mention of this file in meta-tegra

All that file does is tell tegraflash to assign a UUID to the APP flash partition, and to update the kernel command line with a `root=` parameter to specify that UUID.  You could do something similar, although if you're using Mender, you need to specify device names and partition numbers.  U-Boot also may need customization to get it to look at the SATA drive to find the rootfs.

> Your concern regarding SATA drive is quite unexpected, can you please tell what kind of compatibility issues you faced with?

It's been a few years, but IIRC it had to do with DEVSLP (device sleep) support.  The controller or driver didn't fully support that feature, and I think I had to hack the driver to just ignore it when it was advertised by the drive.  That may not be an issue any more, or may have been something unique to the drive I was testing at the time.

closed 2020-12-11 by madisongh