OE4T/meta-tegra #426: OE4T and meta-tegra communication survey

dwalkes, 2020-09-22:
Hi Everyone!

We've created a survey at https://forms.gle/utSwVTCYYr3XQQjM9 to gather some input about communication preferences related to OE4T and meta-tegra.  If there's enough interest we may setup monthly status meetings and/or some type of communication channel for discussing OE4T related issues.

I'll post updates to this issue in github, so if you are interested in the outcome please add yourself as a watcher here.

Dan

dwalkes, 2020-10-03:
Hi Everyone,

Thanks for those who filled out the survey! I'm going to keep it open to track any new responses.

Based on the limited responses we have so far, Slack is the clear winner with all but two saying they could use this. I've setup a slack channel at [this link](https://join.slack.com/t/trellislogic/shared_invite/zt-i8dymacg-XcLzNdsJBZuGDw7~EGNzxA).

Regarding meeting times - we were considering a morning time for the Pacific Time zone, possibly around 8AM PT UTC-7 (3PM or 1500 UTC).  We were tentatively thinking about Thursday or Friday next week, the 8th or 9th of October for our first meeting.  

Do any of the people who responded here have preferences on time/day for the first meeting?

ichergui, 2020-10-03:
Thursday or Friday next week is fine for me.
Thanks @dwalkes and @madisongh
Have a good weekend.

nandra, 2020-10-05:
Works for me. Thanks.

dwalkes, 2020-10-07:
I setup a meeting for Friday morning at 8AM PT - let me know if this link works for you: https://teamup.com/event/show/id/JVhkcgPMnZ2u9UawP76n4jEiFLc1rX

You can find a link to the hangouts dial-in and phone numbers there as well as some thoughts about agenda.

ichergui, 2020-10-07:
Hi @dwalkes 

I think you should also share this in the slack, what do you think ?

dwalkes, 2020-10-07:
@ichergui done, thanks!

dwalkes, 2020-11-05:
Hi folks: Here's the info for our next scheduled meeting: https://teamup.com/event/show/id/BvBR4Xyi5xYBBvzeJsnoPBmVn8G7JC#

om26er, 2020-11-24:
> Based on the limited responses we have so far, Slack is the clear winner with all but two saying they could use this. I've setup a slack channel at [this link](https://join.slack.com/t/trellislogic/shared_invite/zt-i8dymacg-XcLzNdsJBZuGDw7~EGNzxA).

Hey it seems that link does not work. Can you share an updated/working link again ?

dwalkes, 2020-11-24:
>  Can you share an updated/working link again ?

The one on the homepage at https://github.com/OE4T/meta-tegra/wiki should work, and I'll try to keep that one up to date (it expires monthly).

dwalkes, 2020-11-27:
Hi everyone, here's a link to the December 10th meeting invite:

https://teamup.com/event/show/id/4d59QuScLvcSpWF5ki8HuoGGFE78ML

I'm proposing we keep a 2nd thursday of the month schedule going forward until/unless we run into conflicts with other events.

dwalkes, 2020-12-09:
Moving monthly meeting discussions into https://github.com/OE4T/meta-tegra/discussions/515 so I think we can close this issue.

closed 2020-12-09 by dwalkes