OE4T/meta-tegra #427: Stopping and Starting weston on Jetson Nano Developer Kit

mozcelikors, 2020-09-24:
I would like to find commands to stop and start weston on Jetson Nano Developer Kit. The reason I want to do this is that I want to create a systemd service that  stops and starts weston as the system demands it, Additionaly, I would like to create Qt apps in weston with the systemd service. In other boards (e.g. IMX8 or RPi), I used to use `weston-start` or `weston-launch` which would work. However, with Jetson Nano image that is created with `"core-image-weston"`, I am unable to do this.

Stopping weston properly;
`systemctl stop weston@root.service`

This works. I can see with `ps -A | grep weston` that there is no weston process any longer.

However, I can't restart weston.

With `weston-start`, I get:
`Internal warning:debug scope 'drm-backend' has not been destroyed.`

With `weston-launch`, I get:
`weston: failed to get current keyboard mode: Inappropriate ioctl for device.`

I enter the commands via UART serial debug console.

Did anyone managed to stop/start weston properly with Jetson Nano image created with meta-tegra?

Thank you for guidance.

madisongh, 2020-09-24:
Does `systemctl start weston@root.service` work?  That's usually what I use.

Take a look at the command that the service file uses, which you should be able to see with `systemctl status weston@root` while the service is running)... Weston startup on the Jetsons requires an additional command line argument, `--use-egldevice`, to have it use the EGLstreams support.

mozcelikors, 2020-09-24:
Thank you for very helpful pointers. I had been using `systemctl start weston.service` apparently for starting instead of `weston@root.service`.

madisongh, 2020-09-25:
> I had been using `systemctl start weston.service`

Oh, right. That was the service in older OE-Core/YP releases, and it confused me, too, when it switched over to using the template unit approach.

madisongh, 2020-12-11:
Closing as this looks like it's been answered.

closed 2020-12-11 by madisongh