OE4T/meta-tegra #428: updating just kernel

arrow53, 2020-09-24:
Instead of using `doflash.sh` to update the whole image is there a way to just update the kernel `Image`? I'm doing driver development and looking for a way to more quickly iterate.

kekiefer, 2020-09-24:
You should have two ways to do this. The flash helper supports a `-c` command to send arbitrary commands to the tegra flash tools, and these should be passed through doflash.sh:

To program the kernel partition:

```
./doflash.sh -c "signwrite kernel ${DEPLOY_DIR}/tegra-minimal-initramfs-jetson-xavier.cpio.gz.cboot;reboot"
```

Or the device tree partition:

```
./doflash.sh -c "signwrite kernel ${DEPLOY_DIR}/tegra194-p2888-0001-p2822-0000.dtb;reboot"
```

The other way you can do this is by setting up the the extlinux.conf file in the rootfs. You can search the L4T documentation for extlinux and look for the instructions specific to the platform you're on.

arrow53, 2020-09-24:
thanks! @kekiefer

arrow53, 2020-10-05:
@kekiefer i'm using mender now so I don't think the direct partition programming will work (didn't work for me at least) on xavier nx.

The extlinux.conf is great though. I'm new to secure boot and that seems to be enabled as it's failing on not have corresponding .sig files. 

There's very little info on the web about these files. Have you heard of them? Can they be generated as part of yocto?

All I can really find is a mention of it but not how to create it
https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%2520Linux%2520Driver%2520Package%2520Development%2520Guide%2Fbootflow_jetson_xavier.html%23wwpID0E0FB0HA

kekiefer, 2020-10-05:
Have you burned the fuses on your board? cboot will complain about the lack of the signature, but as long as the fuses aren't burned, it will ignore the signature errors.

Also, the `doflash.sh` approach should work fine with mender. Just mind if you're in the A/B root (`nvbootctrl get-current-slot`, 0=A 1=B), and select e.g. `kernel`/`kernel_b` or `kernel-dtb`/`kernel-dtb_b`.

arrow53, 2020-10-06:
I don't think I've burned the fuses. I'll try to figure that out then close the issue.

Thanks for the tip about the slot. That's what I needed. FWIW, in case anyone comes across this issue for the xavier nx, this was the kernel I used
```
sudo ./doflash.sh -c "signwrite kernel ${DEPLOY_DIR}/Image-initramfs-jetson-xavier-nx-devkit.cboot;reboot"
```

arrow53, 2020-10-06:
@kekiefer I see these .sig errors
```
[0007.238] I> Loading kernel sig file from rootfs ...
[0007.239] I> rootfs path: /sd/boot/Image.sig
[0011.369] I> lookup_linear_dir:441: Invalid file block num
[0011.369] I> ext2_walk:142: 'Image.sig' lookup failed
[0011.370] I> ext4_open_file:647: '/boot/Image.sig' lookup failed
[0011.370] E> file /sd/boot/Image.sig open failed!!
```
and then it runs for a bit then hangs
```
[0018.999] panic (caller 0xa0601238): die
[0019.003] HALT: spinning forever...
```

From here https://forums.developer.nvidia.com/t/is-there-a-way-to-disable-image-sig-authentication/119091/4
it says if I do this 
```
This can be achieved by disabling CONFIG_ENABLE_SECURE_BOOT feature.
Remove following line from bootloader/partner/t18x/cboot/platform/t194/l4t.mk
CONFIG_ENABLE_SECURE_BOOT=1
```
it will ignore the signatures.

is cboot even built within meta-tegra? As far as I can tell NVIDIA just publishes a binary? Maybe my best path is just to figure out how to make the signature files.

madisongh, 2020-10-06:
We use the prebuilt cboot by default, but there is a recipe in place to build it from source.

Setting
```
PREFERRED_PROVIDER_cboot = "cboot"
```
will switch to using the built-from-source copy.

arrow53, 2020-10-06:
@madisongh hmm, I should be able to add to my `local.conf` correct? I put it at the end and it just seems to build `cboot-preuilt` with `bitbake cboot -C compile`. Any idea what I'm doing wrong?

madisongh, 2020-10-06:
Oops.  Try also adding
```
PREFERRED_PROVIDER_virtual/bootloader = "cboot"
```
I forgot that that also gets set to `cboot-prebuilt` by default.

arrow53, 2020-10-06:
yahtzee

arrow53, 2020-10-06:
i'm going to close. I still can't figure out for the life of me how to generate the .sig signature files but the solutions here get me to a working state. Thanks!

closed 2020-10-06 by arrow53